dresden-elektronik/deconz-rest-plugin #2977: "Not Connected" since FW Update

Paragrimm, 2020-06-21:
## Describe the bug
I'm Using the stick with my Synology NAS and installed Deconz via Docker. It all worked "quite well" until I installed the FW, now it seems like it can't connect anymore.

I already tried to restart everything, plugged the stick out and in again and so on but nothing seems to work.

## Expected behavior
Established Connection as it was before.

## Environment
 - Host system: Synology NAS
 - Running method: Home Assistent deCONZ Add-on, Marthoc Docker container
 - Firmware version: (26xxyy00)
 - deCONZ version: 2.05.78
 - Device: ConBee II
 - Do you use an USB extension cable: no

Smanar, 2020-06-22:
Hello, can you make a try on another machine ? A windows machine if possible (with installing deconz on it).
And during this test, check the usb connexion.

Paragrimm, 2020-06-22:
> 
> 
> Hello, can you make a try on another machine ? A windows machine if possible (with installing deconz on it).
> And during this test, check the usb connexion.

Hey, just did it and I can't connect to the stick aswell.

DeConz recognizes the stick and I can click on "Connect" and the button get's grey/disabled for some seconds and thereafter it "aborts" and I can click on "Connect" again with the same behaviour.

Smanar, 2020-06-22:
ok
So just by curiosity is not an USB disconnexion loop ?

It s easy to see on windows because the "USB sound", but you can too on unix journal or using "GCFFlasher_internal -l", one time the command is working, 5 s it don't work, 5s after working again.
Or in Phoscon, at gateway, you can a firmware numer, 5s after "not connected", 5s the firmware number ect ...

Paragrimm, 2020-06-22:
Nope, not a loop, I don't hear the Connected/Disconnected Sounds from windows. And it's always "not connected" in Phoscon.

Smanar, 2020-06-22:
And what is the result of "GCFFlasher_internal -l" ?
It's not software because it don't work too on windows (and there is realy few situation for the conbee don't work on windows)

I think the better thing is trying to "play again" with firmware update.

Paragrimm, 2020-06-22:
Wait, thanks for your help, I'll try.
I'm trying to avoid to reset the Gateway completely, because it's annoying to pair the Spirit Zigbee Thermostats :/.

Paragrimm, 2020-06-22:
```
root@deconz:/# GCFFlasher_internal -l                                                                     
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh                                             
Path             | Vendor | Product | Serial     | Type                                                   
-----------------+--------+---------+------------+-------                                                 
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II
```

This is the result.

Smanar, 2020-06-23:
Don't worry, you can update the firwmare, you will not loose your config.
But make a backup before, you will not loose your config if you don't have worse issue ^^.

Like I have said on windows, there is realy few problem, no power alimentation problem, no perturbations (I think even with USB 3.0, there is no issues), no software problems.
So I realy think the problem is on the device.
Your key is detected, it' a good thing, but I don't see better thing to try that a new firmware update.

Try to make it on windows (but there is no problem with other system) https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Paragrimm, 2020-06-23:
Do I have to flash it back to an older version before? I can't connect to the device since I did the firmware update to the most recent version.

I did the update on Linux (directly on my NAS), maybe it's better if I do the update on my windows pc again.

Thanks so far.

Smanar, 2020-06-23:
Nope, you can use the last one, you will have same problem with all versions, there isn't version easier to update.

closed 2020-07-25 by stale[bot]