dresden-elektronik/deconz-rest-plugin #2978: Hive Active Heating Receiver SLR2

peterguy04, 2020-06-22:
## Device

- Product name: Hive Active Heating
- Manufacturer: Computime
- Model identifier: SLR2
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [x] Thermostat
  - [ ] Other: 

## Screenshots

  - Endpoints and clusters of the node
[image]

  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]

 ## 05
[image]
[image]
[image]

## 06
[image]
[image]

## 07
[image]

## 08
[image]

Smanar, 2020-06-22:
Lol, but you have the complete collection ?

peterguy04, 2020-06-22:
@Smanar haha yes i have a few Hive devices.

Smanar, 2020-06-22:
I will take it, like that I will make a package

Smanar, 2020-06-24:
We are searchaing how to act on the second relay.
Is someone have information ?

Smanar, 2020-07-12:
@peterguy04 have sniffed the original gateway, for info

> ---------------------------------------------------------
> 
> Endpoint 6
> cluster 0201
> 
> for "on - manual"
> 
> attribute 0x001C > 0x04 = "heat"
> attribute 0x0023 > 0x01 = "Setpoint Hold On"
> 
> For "off"
> 
> attribute 0x001C > 0x00 = "off"
> attribute 0x0023 > 0x00 = "Setpoint Hold Off"
> 
> For "boost"
> 
> attribute 0x001C > 0x05 = "Emergency heating"
> attribute 0x0023 > 0x01 = "Setpoint Hold On"
> attribute 0x0024 > 0x3C = 60
> 
> ------------------------------------------------------------
> 
> Endpoint 5
> cluster 0201
> 
> for "on - manual"
> 
> attribute 0x001C > 0x04 = "heat"
> attribute 0x0023 > 0x01 = "Setpoint Hold On"
> attribute 0x0012 > 1150
> 
> For "off"
> 
> attribute 0x001C > 0x00 = "off"
> attribute 0x0023 > 0x00 = "Setpoint Hold Off"
> attribute 0x0012 > 0
> 
> For "boost"
> 
> attribute 0x001C > 0x05 = "Emergency heating"
> attribute 0x0023 > 0x01 = "Setpoint Hold On"
> attribute 0x0024 > 0x3C = 60
> attribute 0x0012 > 2400

closed 2020-07-17 by peterguy04