dresden-elektronik/deconz-rest-plugin #2980: Receiving automatic message every 50min 24sec

pandschai, 2020-06-22:
Hi,
I use the following setup:
Multiple Aqara Door Sensors are connected to conbeeII-stick in a Raspberry Pi. On the Raspberry there is Deconz connected to node-red.
All works out fine (which is really great!) the only thing is that **I receive an automated generated message after 50min 24sec if a sensor is inactive**. Aqara sayed its not them generating the message that's why i presume its deconz.
Unluckily the automatic-generated message is identical to the message which is produced when the sensor is registering an event, so i cannot just block it.
Does anybody know:
a) a way to stop this automatic generated messages?
b) delay the time of automated message creation from 50min 24 sec to once in 24h?
In the last case I could use it as ping. 
Thanks, that would be very helpfull!
Peter

SwoopX, 2020-06-22:
Never seen anything like that in the code. Sensors usually turn unavailable after 24h or if their report intervall is significantly exceeded afaik.

ebaauw, 2020-06-22:
> I receive an automated generated message after 50min 24sec if a sensor is inactive. Aqara sayed its not them generating the message that's why i presume its deconz.

The device sends a _Report Attributes_ command for the Xiaomi special attribute, which contains the device state, battery level, temperature, signal strength of the last message, and some other stuff we haven’t been able to decode.  deCONZ uses this to update `state.open`, `state.lastupdated`, `config.battery`, and `config.temperature`, sending out corresponding notifications.

> Does anybody know:
a) a way to stop this automatic generated messages?
b) delay the time of automated message creation from 50min 24 sec to once in 24h?

No.  The frequency of the special attribute report seems hard-coded in the sensor firmware.

> Unluckily the automatic-generated message is identical to the message which is produced when the sensor is registering an event, so i cannot just block it.

Your node red automation will need to take into account if the value of `state.open`has actually changed.

pandschai, 2020-06-22:
Thanks for the quick answers. I am going to solve it than with the value of state.open in node-red. Best Regards and thank you for all the support!

closed 2020-06-22 by pandschai