dresden-elektronik/deconz-rest-plugin #2982: Q: How to turn off Hue bulbs' bluetooth?

gvdhoven, 2020-06-23:
I've connected about 10 new Hue bulbs to my Hue bridge to firmware update them before connecting them to Deconz. After updating, i thought i had to reset them before i could pair them to Deconz; so i did and they are succesfully connected to Deconz now.

However, they keep on emitting a Bluetooth signal which i would like to have turned off (which is normally the case after pairing them with your Hue bridge)

So the question is how to turn the Bluetooth off?
1) Is there a command i can send (or)
2) Can i still (re)connect the lamps to my Hue bridge retaining also the connection to Deconz?

ebaauw, 2020-06-23:
> Is there a command i can send (or)

I don’t know.  Afaik, the Hue API doesn’t provide anything here.  There might be a manufacturer-specific attribute somewhere which takes some magic value to turn off bluetooth.  You’d need to sniff the Zigbee traffic between the Hue bridge and the bulb to find out.

I don’t have any of the latest Hue lights supporting bluetooth, and I won’t be in the market for any anytime soon, as my entire house has been fitted with smart lights already.  Happy to take a look if some-one can lend or donate a bulb, though.

> Can i still (re)connect the lamps to my Hue bridge retaining also the connection to Deconz?

No.

gvdhoven, 2020-06-23:
I have a spare bulbs which i can send to you, you're Dutch right?

bsheepdiy, 2020-06-25:
@gvdhoven 

> However, they keep on emitting a Bluetooth signal which i would like to have turned off (which is normally the case after pairing them with your Hue bridge)

Are you sure of this feature, is it documented ? how long after pairing the device ?

I have philips hue / bluetooth device and sniffing key, I can try to sniff the network

ebaauw, 2020-06-25:
> However, they keep on emitting a Bluetooth signal which i would like to have turned off (which is normally the case after pairing them with your Hue bridge)

Where did you get the info that this would normally be the case with the Hue bridge?  From what I can find on the Internet, you can**not** disable bluetooth on the Hue lights.  They can only be paired over Bluetooth within 30 minutes after powering on, but already paired devices can continue to control the lights over Bluetooth, even when they're also connected to the Hue bridge.  The range for pairing has been reduced to ~1m; after that, the normal Bluetooth range (~10m) applies for controlling the lights.

See e.g. https://www.reddit.com/r/Hue/comments/djd4lf/is_it_possible_to_disable_bluetooth/.

bsheepdiy, 2020-06-27:
I've made some test with Philips HUE network. I ugraded my philips hue hub (BSB002) to 1939070020 version (released on June 8, 2020).
I ve upgraded my A60 E27 filament bulb LWA004 (bluetooth) to latest version 1.65.9_hB3217DF4
Using bluetooth scanner, one hour after the upgrade, Hue Bulb still emit bluetooth advertisment...

At this time,I can confirm unfortunately, there is no upgrade that disable bluetooth advertisment.

Hypfer, 2020-07-08:
While the bulbs do still advertise, it's not possible to connect to them after the time has passed. At least iirc.

Works for me

closed 2020-08-06 by stale[bot]

Hubro, 2020-11-20:
No update on this I suppose? I have about 20 bulbs blasting Bluetooth advertisements into my apartment, and I suspect it's affecting the signal quality of my other Bluetooth devices :disappointed:

Hypfer, 2020-11-20:
@Hubro Only Signify could change that but they won't.
Anyways, those advertisements shouldn't interfer with anything.

Ishidres, 2022-09-18:
I've had the same issue a couple months (or maybe over a year ago) where I wanted to disable Bluetooth on my Hues which were already connected with the Hue bridge. I just checked again on my phone and scanned for nearby Bluetooth devices but I couldn't find a single Philips Hue device listed. Does this mean that Philips Hue devices now automatically disable Bluetooth if it's not necessary, e.g. if you control it via the Hue bridge? Can anyone confirm this?

ebaauw, 2022-09-18:
Hm, the Hue Bluetooth app now says it’s been retired and you should use the regular app.  Don’t see any Bluetooth options, but probably because it’s connect to my Hue bridge.  I do still see Hue devices in nRF Connect though.  Apple applies all sorts of filters, so I wouldn’t trust the iOS native Bluetooth overview.

PuscasSebi, 2025-01-06:
Any way to turn off bluetooth with a v1 bridge?