dresden-elektronik/deconz-rest-plugin #2983: Danfoss Ally

JSteunou, 2020-06-23:
## Device

- Product name: Danfoss Ally
- Manufacturer: Danfoss
- Model identifier: eTVR0100
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [X] Thermostat
  - [ ] Other: 

## Screenshots

I do not know have screenshots yet because I'm planning on buying those devices, but not owning one ATM.

**BUT** Danfoss was kind enough, through a Chief Salesman, to provide clusters specification and documentation. This is rare enough to be noted and I appreciate that from them. A good brand for the people and the workers, with good stuff.

[Draft - Zigbee Cluster Specifications Danfoss eTRV.xlsx](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4818269/Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV.xlsx)
[Danfoss Ally™ Datasheet color.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4818270/Danfoss.Ally.Datasheet.color.pdf)

Mimiix, 2020-06-23:
[Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV (1).pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4818393/Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV.1.pdf)
Sheet 1 (Commands)

Mimiix, 2020-06-23:
Sheet 2: Attributes 
[Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV (1)_attributes.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4818501/Draft.-.Zigbee.Cluster.Specifications.Danfoss.eTRV.1._attributes.pdf)

Mimiix, 2020-07-08:
If anyone actually has this device, please let us know. We lack the MAC prefix.

zozogb, 2020-07-13:
Hello, if it helps, I have a new Danfoss Ally ZigBee. And I'm French so sorry for my English.

Smanar, 2020-07-13:
Hello, I m french too if you have problem.
We miss some information for this device, you can find information here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

For information, someone in the dev team have buy one too, but need some time for livraison.

_Pour résumer, on aurait besoin des mêmes capture que sur le lien precedent, capture faites avec deconz._

JSteunou, 2020-07-15:
@zozogb come on to the discord channel to talk about it if you want, I'm french too ;)

zozogb, 2020-07-15:
Ok, @JSteunou but you seem to have the screenshot which you miss now and I do not see how I can be useful to you.

JSteunou, 2020-07-16:
just if have some question or wanna chitchat about it ;)

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

JSteunou, 2020-07-20:
Victory Dance :tada:

Mimiix, 2020-07-20:
@JSteunou I'd like a video of that ;D

ComPeter33, 2020-07-29:
i got a Danfoss Ally radiator thermostat (prototype)
Without any manual.

How can reset device and pair it in Phoscon App (Conbee 2)

SwoopX, 2020-07-29:
Should be something like pushing the button 10 secs or so to set it into pairing mode

ComPeter33, 2020-08-20:
Thnx, i got my Ally thermostat connceted in DeConz and working in Domoticz
When i set the temp on the Thermostat the value changes in Domoticz

But when i set the temp in Domoticz i get this error: 

2020-08-20 15:28:12.425 (ConBee 2) onCommand called for Unit 11: Parameter 'Set Level', Level: 26.5, Hue:
2020-08-20 15:28:12.414 Status: User: competer initiated a SetPoint command
2020-08-20 15:28:12.448 Error: (ConBee 2) Connexion problem (1) with Gateway : 400

My Deconz is on a serperate PI than my Domoticz installation

What can i do to resolve this issue?

SwoopX, 2020-08-20:
This is no deconz issue, the REST API is working here.

dowhiletrue, 2020-12-12:
I wonder if `0x4015 External Measured Room Sensor` might be set as writable attribute in `general.xml` since [Danfoss trv.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/5493076/Danfoss.trv.xlsx) states the attribute to be writable. I changed this for testing purposes in and wrote a value. The TRV does not honor the value. Does anybody know what else has to be configured in order to get an external room sensor to integrate with the TRV?

JSteunou, 2020-12-14:
I wondered the same thing. Maybe it is just to give external measure in the
sense of outside the house temperature so the trv can adapt its logic like
the external sensor for boilers.