dresden-elektronik/deconz-rest-plugin #2984: Niko 170-33505

Sennevds, 2020-06-24:
## Device

- Product name: 170-33505 
- Manufacturer: Niko
- Model identifier: Connected socket outlet
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [x] Other: 

## Screenshots
[image]
[image]
[image]
[image]
[image]

Is recognized as light but doens't expose the measurements as sensors (checked via postman and the deconz api)
So the request is to expose the measurements as a sensor so I can use these in Home Assistant.
If there's any information missing let me know!

SwoopX, 2020-06-24:
@Sennevds Yeah, you're my hero! So they're finally out?! Wanted to get my hands on them for months now. Where did you buy and at what price?

pergolafabio, 2020-06-24:
yeah, really interesting!! gonna buy those too
here for example : 

https://www.zelektro.be/slim-stopcontact-zigbee-geconnecteerd-schakelbaar.html

Sennevds, 2020-06-24:
Ha now they have them at zelektro(normally I buy my gear there but 3 days ago they didn't had them in stock) I've bought them at gigatek.

pergolafabio, 2020-06-24:
btw, is there also an hardware button on the Niko itself? not quite clear on the pictures... i only see a led , no button
or can you only turn them off/on with the deconz?

Sennevds, 2020-06-24:
The LED is also a button

pergolafabio, 2020-06-24:
Cool

SwoopX, 2020-06-24:
But the size and the frame is a crime... I'd need 55mm instead of 45mm and I'm not sure if adapters may do the trick.

Sennevds, 2020-06-24:
I used a standard niko frame

SwoopX, 2020-07-03:
@Sennevds How do you pair those things? Seems I cannot get it to work...

SwoopX, 2020-07-03:
Nvm, got a reply from their support. It's not even mentioned in the manual...

**The sequence to put the socket in pairing mode is: 3x short button press**

pergolafabio, 2020-07-03:
Where did you buy now?

SwoopX, 2020-07-03:
@pergolafabio electro-colli.be

Quite some extensive manufacturer specific cluster

[image]

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

pergolafabio, 2020-09-15:
@SwoopX  @Sennevds 
look at this one : https://www.niko.eu/nl-be/artikel/122-91004
that switchs seems without a battery, seems the power is mechical with a press of a button
do you guys also tried that one?

Sennevds, 2020-09-15:
I have it at home and it works with deconz

pergolafabio, 2020-09-15:
seems only 4 buttons , is there no 2 button variant? 
and is there indeed no battery ? thats nice!

Sennevds, 2020-09-15:
Only 4 buttons and no battery. The buttons are harder to push then a standard niko push switch because when you push you generate the energy needed to send out the signal(piezo effect)

pergolafabio, 2020-09-15:
ok, but its indeed a nice switch, but maybe not interesting for kids then

camarillla, 2021-01-25:
> ## Device
> * Product name: 170-33505
> * Manufacturer: Niko
> * Model identifier: Connected socket outlet
> * Device type:
>   
>   * [ ]  Light
>   * [ ]  Lock
>   * [ ]  Remote
>   * [ ]  Sensor
>   * [ ]  Siren
>   * [ ]  Thermostat
>   * [x]  Other: Connected socket(recognized as light)

What's the status of this request? Is it possible to get the power measurements with deconz (without using ZigbeeM2qtt)?

Please let me know.  Tanks in advance!

swannes, 2021-09-29:
Did anyone try the Niko connected switch/double switch/dimmer yet ?
https://www.niko.eu/en/article/552-72122 https://www.niko.eu/en/article/552-72201 https://www.niko.eu/en/article/552-72101
Should be Zigbee
(yes, these are expensive devices)

swannes, 2021-09-29:
My bad, not yet available, although I see them in some webshops