dresden-elektronik/deconz-rest-plugin #2985: deCONZ crash...?

Morphy-8, 2020-06-24:
I have deCONZ running on ubuntu by a PROXMOX VM. USB is set to version 2 and I do have a 40cm USB cable in between the stic and the PC. The PC is an intel NUC, the VM has 5G of RAM the VM is vital and there is no issues with it.

On random occations the communication of zigbee does not work. I am currently trying to find out the time of the crash...I think it is random - every 1-2- days. :-)

Looking into deconz, I see the connect screen in disconnect state. Pressing the connect button is non responsive. Re-plug the stick does not do anything. 
If I stop deconz (x on the window) and start again, I can get a connect on the start screen and the system is running fine for another 1-2 days.

I am not able to further debug this problem without any further help. I am willing to offer logs, but do not know what to provide, as I am not having much experience with Linux. Please bear with me.
Sharing a pic of the connect screen with a not working buton is not helpful I guess... :-)

deCONZ 2.05.77
Firmware 0x264a0700
ubuntu 18.04.4

Cheers!
Andy

SwoopX, 2020-06-24:
That feels like the serial communication passes out from time to time. If I recall correctly, a watchdog has been implemented into the firmware to autonomously restart it.

In that sense, I'd recommen to upgrade the firmware. The required steps are described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Mimiix, 2020-06-25:
Don't mind me editing your title, just making it more conclusive.

However, i've read about this earlier. People saying stuff stops working after 1 or 2 days. Always hapening on ProxMox too. I have no clue what causes it, but it just seems to be related.

A workaround is rebooting it every day.

Morphy-8, 2020-06-25:
Hi SwoopX and Dennis, Thank you for your inputs. I will go ahead and do the FW update, then check the VM reboot workaround. Thanks again! 

  after a few days. (#2985) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2985#issuecomment-649437714",
"url": "https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2985#issuecomment-649437714",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Morphy-8, 2020-06-25:
Updated to:
- deConz 2.05.78
- ConBeeII 26580700

Now monitoring...

Mimiix, 2020-06-29:
@Morphy-8 Hows things going :D?

Morphy-8, 2020-07-03:
Update: Still running after the update... Looks good. Dont do reboot of the VM

Morphy-8, 2020-07-06:
Yesterday it crashed... Needed to close deConz and then restart. I like to find out, when it crashed... any idea how I can find this out?

Mimiix, 2020-07-06:
Proxmox has shown this behavior to me a few times. Can you run it on a native environment?

Morphy-8, 2020-07-06:
Not much I have around - a Raspi2 recomended for a reliable operation? Thats what I have at this time. 
Maybe I dump my proxmox setup and install Windows 10 on the NUC...

closed 2020-08-06 by stale[bot]

smqrt, 2021-01-24:
I have this problem to, same setup. Ubuntu 20.04 in VM. ModemManager disabled.

Someone found a fix? With half of my setup (around 63 devices) I never had this problem (2 months without a crash) The moment I added the rest of my devices up to 126 the connection to the serial disconnects every 2/3 days.. very annoying when you come out of your bed whilst it's stil dark and none of the lights respond..

Mimiix, 2021-01-26:
@smqrt I recommend opening a new user question issue. Please add some logs there too :)