dresden-elektronik/deconz-rest-plugin #2986: Router connecting to nodes with really bad LQI instead of picking better ones

fi-sch, 2020-06-25:
Hello!

At first sorry for reporting it as a bug, as I assume that it may not a bug regarding deCONZ itself. I've tried searching for the answer, I've read the Wiki and deCONZ manual – didn't find anything helpful.
So here's my problem.

I have multiple devices in my network (around 40), 12 of them are acting as routers (Ikea and Hue bulbs, CC2530+CC2591 modules, CC2530 modules, Sonoff BASICZBR3, etc.). The problem is with those devices that are based on the CC2530 module. They're just weak and they play tricks on the whole mesh network.

In this particular case, I have an issue with two Sonoffs (CC2530) struggling to route the signal. The outcome is that another CC2530 (DYI light switch) connects to both of them instead of picking "stronger" routers (that it was connecting before). Finally, it often loses the signal from those two Sonoffs and does not pick the better routers.  

I would like to be able to switch off the routing feature for those wicked CC2530 modules (Sonoffs) and/or force this problematic CC2530 (DYI light switch) to connect with other nodes (Hue or CC2530+CC2591) which would offer definitely better and more stable LQI. So here we come to two questions:

1. Is it possible to disable the routing feature for selected nodes in deCONZ?
2. Is it possible to force a link between two nodes?

Thanks for your assistance in advance!

Smanar, 2020-06-25:
Not realy, but it will be useless.
The mesh is something "alive", even you had the mesh you want (using tricks) , one hour later it can move (except for xiaomi)

But we generaly have this problem with end device, not router ...
And yes I have already see another issue about the short range with sonoff relay.

fi-sch, 2020-06-25:
That is really annoying to me as I did setup a perfect coverage using ConBee II and about 5 CC2530+2591, not mentioning the Philips Hue, Ikea Tradfri and Aqara bulb. So having a 150% of coverage needed those kind of issues are quite surprising and upsetting.
I can't understand any logic between that. This CC2530 (DIY switch) has got nearest router (CC2530+CC2591) as close as ~1-1.5 meters away, right behind the 30 cm wall.

Smanar, 2020-06-25:
Do you have a capture of your mesh ?

Your problem concern only router ? So I don't understand, routers have more than one connexion, so they can have a bad connexion and an other good.

What is the devive that have connexion problem ?

morfei1, 2020-06-26:
I don't know, but some days ago. I'd tested my Aqara Water Leak Sensor, and I found out it lost connection to deCONZ after a month working. Maybe after updating to .78.... I'm not quite sure... I dont blame anything. Anyway, I re-paired it, and at first it conected to the coordinator it's normal.Then, I putted it on it's place, and the line to coordinator was not green, but orange, it was working.  I rebooted my phoscon GW several times, and it always connect to coordinator. 2-3 days passed. Today, I checked the deconz map. What I found out was the Aqara water leak is not conected to the coordinator anymore, but to a light which works a repeater in my mesh and is closer and at better position to the Aqara Sensor, than the coordinator. The conection line now is green.

In my case, the Aqara sensor end device did switch to a better router...

Smanar, 2020-06-26:
So you don't have problem ?

And you are lucky, Aquara never (usually at least) change their parent, so one time they choose it, if you move it and if they can't see the parent, they leave the network.

The pairing is made at the final location.

morfei1, 2020-06-27:
As I said, I don't know. Maybe, that's the way Aqara sensors works, otherwise, it probably wouldn't lost the connection to the parent after 3-4 weaks working. What I've noticed after I had to re-prair it again, at first, it was conected to the coordinator for 2 - 3 days, then it moved it's connection to the repeater which is actually closer.

Maybe, it needs some days to finalize it's connection and after that, if it loses connection, the sensor is not able to re-connect again to the mesh.

Smanar, 2020-06-27:
Xiaomi have too compatiblities problme with somes router, but if your repeter was not compatible he would have left the network faster than after 2/3 days.

fi-sch, 2020-06-27:
> Do you have a capture of your mesh ?
> 
> Your problem concern only router ? So I don't understand, routers have more than one connexion, so they can have a bad connexion and an other good.
> 
> What is the devive that have connexion problem ?

You can find a capture below. However, I'm not sure how it can be helpful (it may even cause a confusion) as I described my situation extensively in the first post. Please not that I got rid of one Sonoff (named 'Sonoff 1') yesterday. 

My problem concern a device (in this case CC2530 router – named 'Klatka') connecting to the other weak CC2530 routers (Sonoff Basic Zigbee – named 'Sonoff 1' and 'Sonoff 2') with LQI 0/0 instead of powerful ones that are much closer, offer better LQI and are for sure available to this router. This particular device establishes only one or two connections at maximum.

[image]

morfei1, 2020-06-27:
My repeaters a 2x e27 Tradfri 1000lu ; 3x Tradfri Plugs ; 1x Sonoff Basiczbr3. 

I have an Aqara T&H Sensor very close to the Basiczbr3, but the Aqara preferes to connect to one of the e27 Tradfri 1000lu, which is actually farther. 

Anyway, Basiczbr3 is not a good repeater, IMO. My remotes (Tradfri) and sensors (Aqara) don't like it. Lights seams to be OK with it.

fi-sch, 2020-06-27:
Man, you have completely different issue than mine. Sensor and remotes are end-devices. I'm writing only about devices acting as routers. And I wrote at the beginning that I don't use them as routers as I have a perfect ZigBee coverage at my place.

morfei1, 2020-06-27:
Actually I don't have issue. If a switch, plug or light is powered ON, even it's switch OFF, it still acts as a repeater.

We both use basiczbr3, as I've mentioned already it's a bad repeater, but the are great for build-in. 

Power them off from your mesh, reboot , see how your routing gonna be, replace them with something different for example tradfri plug if you can.

fi-sch, 2020-06-27:
Honestly, you didn't write anything new and helpful.

morfei1, 2020-06-27:
Allow me to tell you how not smart (dumb) the basiczbr3 can be. Maybe you already know that...

I power OFF all my lights from the wall switch excep the 2 of them, which are working as repeaters, and I reboot the deCONZ.

When GW is online and it starts to form the mash again, the Tradfri lights that works as repeaters and the Tradfri Plugs are smart enough to not try to connect to lights that are powered OFF from the wall switch and are unavailable at the moment . They start to build conections to the end points and thats it, then they wait for unavailable nods to be powerd ON. 

The Basiczbr3 is not smart enough and it does not try to build connections with the end point devices, but instead he always try to connect to all the light in its range, even though the are currently unavailable and spams connections with red lines.

During the winner, it was on my terrace switching on/off my cristmas lights, all the lines to the other nodes were always red at best case orange, and it's not so far away. I know it's range is limited. Now, it's in my bathroom controling my exhaust fan and things are little bit better, because it's closer to the coordinator and the other devices. 

Anyway, I don't blame deCONZ, it can't teach a dumb router how to route, because of it's HW limits or maybe FW problems. I get what I paid for at the end.

Sorry if my posts spams your issue and were not helpfull.

Good luck! Regards!

ebaauw, 2020-06-27:
> When GW is online and it starts to form the mesh again,

The gateway doesn't form the mesh.  All routers do collectively.  When you power down the coordinator, the network continues to function, just like when powering down any other router.

The building of the map in the GUI has nothing to do with the mesh.  It just displays the neighbour tables of the nodes, as the gateway queries these nodes.  It's like a city map showing roads (possible routes); it's **not** showing actual traffic (connections).

> I would like to be able to switch off the routing feature for those wicked CC2530 modules (Sonoffs)
> And I wrote at the beginning that I don't use them as routers as I have a perfect ZigBee coverage at my place.

If you join a Zigbee router (full-feature device, yellow node in the GUI) to your network, it acts as router.  Period.  You don't get to choose whether or not to "use them as routers".

The only way not to "use them as routers" is to configure the device firmware as end device.  I don't know the Sonoff, let alone whether their firmware supports such configuration.  The RaspBee/ConBee does **not** (you can only configure the device as coordinator or router).  The XBee firmware does.  I haven't seen any other device that supports this.

fi-sch, 2020-06-28:
> If you join a Zigbee router (full-feature device, yellow node in the GUI) to your network, it acts as router. Period. You don't get to choose whether or not to "use them as routers".
>
> The only way [...]

Thanks for extensive reply. Is it possible to somehow initiate the connection between two nodes (force, promote or just make them try to connect)?
I never observed any connection between the "problematic" CC2530 module (DIY switch) and its nearest router (CC2530+CC2591). I even tried repairing that node. If initiating of connection would be possible, then it would maybe solve my issue. Routers based on CC2530+CC2591 provide really good and stable coverage. For some reason, I just can't make those two particular devices talk with each other.

Sorry about asking for such basic things. I'm not so familiar with how the ZigBee networks work and how the mesh is supposed to form. From what I can see I'm not the only one, though. Some behavior for me is totally counter-intuitive and inconsequential (I mean, for example, weak router picking another weak router instead of much better "neighbor") – which happen really often.

Smanar, 2020-06-28:
For, first no it s not implemented, but it s possible to make only one device in permit join instead of the whole network, so you can force a device to pair with another, but it s useless, because some minuts after it can change.

You say the router is weak, but perhaps for the device it s not the weaker in reality. If the zigbee algorytm choose it, there is a reason.

fi-sch, 2020-06-28:
Can you please tell me how to do so?

Im 100% sure that this router that I want it to join to is strong enough to handle the connection and it's the best router available. I'm not sure how zigbee algorithm works but it's calculations are definitely not right here. The evidence is established link between two weakest nodes (they must be weak in reality as they are CC2530 without any amplifier or antenna...) that I have in home, while there are 2-3 other and better available – with antena and/or amplifier, and in closer distance.

Smanar, 2020-06-28:
You can use that https://github.com/ma-ca/deconz-cli-plugin
You already have a command that look that the correct one in it.

fi-sch, 2020-06-28:
> You can use that https://github.com/ma-ca/deconz-cli-plugin
> You already have a command that look that the correct one in it.
Thanks, I will look into that.

> Or you can just turn off all your unwanted router, and just power on the router you want and make the inclusion. You will have the connexion you want, ..., up you power on disabled routers. If the algorithm have decided it s a bad router, even you force the connection, the zigbee network will be reformed and without it, some minut after.
Unfortunately this CC2530 DIY switch does not want to communicate with said CC2530+CC2591 router which it **should** connect to. There's no logical explanation for that. Believe me, I know my setup and that is really strange.

> I have never see router with 2 connexion maximum, for me this device just have bad range.
Sure it has bad range, because for some unknown reason it won't connect to CC2530+CC2591 router which is **just 1 meter away**.

fi-sch, 2020-07-20:
I've removed this Sonoff BASICZBR3 completely from the network and but this CC2530 router (for me it's "end-device") still won't connect to CC2530+CC2591 router that it should connect to. Like I said before, I did not observe it connecting with that node. Never ever.
Few times a week it is even loosing losing the link completely and I need to force scanning (adding new light via Phoscon UI) to make it re-connect.

Thoky81, 2020-08-15:
@fi-sch I have the same issue like you! Very annoying. Today I removed both Sonoff BASICZBR3 from my zigbee network and everything is back to normal. I dont know where is the problem. I tried everything whats in my possibilities so I hope somebody will find out. Thanks

mamrai1, 2021-02-20:
sorry to spam this but diy routers (CC2530+CC2591) don't show connection between them or end devices...they only show connection to conbee and other routers like lights, plugs, etc

fi-sch, 2021-02-22:
> sorry to spam this but diy routers (CC2530+CC2591) don't show connection between them or end devices...they only show connection to conbee and other routers like lights, plugs, etc

This is true.