dresden-elektronik/deconz-rest-plugin #2988: Blitzwolf BW-SHP13

SkyThes, 2020-06-27:
Device

- Product name: Blitzwolf BW-SHP13
- Manufacturer: _TZ3000_3ooaz3ng
- Model identifier: TS0121
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [*] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Screenshots
[image]
[image]
[image]
[image]

If you need more info let me know!

SwoopX, 2020-06-27:
Thanks for raising the request. Could you please apply a load to the plug and let it run for some time? Unfortunately, I cannot deduce the measurement units undoubtly.

So I need the last 2 screenshots once again after some time a load has been applied and please take then while the load is still applied.

SkyThes, 2020-06-27:
[image]
[image]

Just took those screenshots with a load (laptop charger) connected for almost a hour.

SwoopX, 2020-06-27:
Thanks. Ok, so for power, we use the raw values as I thought.

However, for instantaneous demand the value of 1 feels just wrong, considering the applied load. If it would be 1kWh, that would be way too high. 1Wh can also not be. Based on that, I assume we need to apply a multiplier.

If it jumps to 2 any time soon, I'd assume the measurement unit is dWh...

SkyThes, 2020-06-27:
You mean the Current Summation Delivered (id 0x0000)? This value actually jumped to 2 in the last couple of minutes.
If you want to be sure I will be able to connect a known load (stable current drawing device) tomorrow for a specific amount of time so we can calculate how the values should be interpreted.

SwoopX, 2020-06-27:
That would be awesome!

SwoopX, 2020-06-27:
Hm, interestingly, seems to be the exact same device as here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2677

SkyThes, 2020-06-28:
> That would be awesome!

I just connected a 200W load (4x 50W halogen bulbs) for exactly 15min to the plug. This are the values when I turned on the load:
[image]
[image]

This are the values after 15min:
[image]
[image]

This load is using 0,2kWh so I used 0,05kW in this 15min. The Current Summation Delivered went from 3 to 8 so I think it is safe to say that this value represents 0,01kWh or 10Wh.

SwoopX, 2020-06-28:
Cool, thanks for checking out. Already took that assumption with my commit.

tekgator, 2020-07-04:
Is there anything we can contribute to get it working? The SHP13 connected successfully but only on/off is possible in Home Assistant. Load etc. is not available.

SwoopX, 2020-07-09:
Wait for the next release

J05HI, 2020-07-18:
How to pair it?
Is it a Switch -> Other?

jvyoralek, 2020-07-18:
I added it as light and it works. But only on/off functionality is available for now so waiting for the next release.

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

jvyoralek, 2020-07-21:
I've updated deCONZ version to 2.05.79 and consumption in kWh + power in W are available in HA now what is amazing! 

The only minor issue remains -  Manufacturer name is still _TZ3000_3ooazng and version contains a crazy value.
[image]

Thanks for the great work already!

Webserve, 2020-07-24:
@jvyoralek i removed the old entity and now the right info is showed

[image]

darkhaek87, 2020-09-18:
Hey  @SwoopX  I have one question: Rest-API support is not provided for this Smart plug, right? Because the smart plug is not listed neither as lights nor as sensors.

How can I help to get it into Rest-API?

My cconfiguration is:
RaspBee: 0x26370500
deCONZ: V2_05_82 (i also tried V2_05_79)

Mimiix, 2020-09-18:
@darkhaek87 Hey!

I've got this in use with HA. So it does show with the rest-api in some way.

darkhaek87, 2020-09-18:
@Mimiix I wonder why it works with HA, but reading the Rest-API for example with Firefox REST Client I did not find the smart plug.

FYI: For my automation I use Node Red.

Mimiix, 2020-09-18:
@morfei1 could help

morfei1, 2020-09-18:
If the plug is paired correctly to deCONZ it should be visible in the rest-api. I have 4 of them.

Plug itself is visable in ..../lights
It's 2 senors are visible in .../sensors
in the rest-api

If your NodeRed is paired correctly to deCONZ, the plug will be visable in the deCONZ nodes with it's phoscon name.
Note that the plug sensors are visible with their default names TS....something. You can't change their names nigher in phoscon nor in deCONZ. You can only change them in the rest-api.

If you are able to control your other deCONZ devices in NodeRed your connection is ok, and it seems the plug is not paired or not paired correctly. You can check that in deCONZ. The simple way to resolve this is to reset the plug and hit the search button in phoscon for another try.

morfei1, 2020-09-18:
> The only minor issue remains -  Manufacturer name is still _TZ3000_3ooazng and version contains a crazy value.
> [image]
> 
> Thanks for the great work already!
> 

This is actually the correct name of the plug IMHO. At first it's always Heiman, but if you read the basic cluster of it's node in deCONZ it will change to TZ3000 something.

CarJu, 2020-10-17:
Hello!
I use 2.05.83 and I also can only connect to the "Light". There is no sensor in the list.
What am I doing wrong?
Vendor in my case is Heiman,

Thank you!

SwoopX, 2020-10-17:
Which list? You wont see anything else in Phoscon. The REST API tells you what is available.

If you have no corresponding sensors in the API, enter sensor search in Phoscon, apply a load to the plug and switch the plug on. That should do the trick.

bcutter, 2021-02-07:
Can someone confirm this device is working
a) without any issues,
b) giving at least switch, current consumption, total consumption values (perfect would be overpower warning and temperature in addition) and
c) there´s no need to push the button once the connection to deCONZ has been lost.

I´m looking for a stable alternative to the Xiaomi Mi Smart Plug which has too many problems to be used with deCONZ (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4238).

Thanks for any feedback, maybe especially from Home Assistant users.

morfei1, 2021-02-07:
Yes, it's OK. I have 5-6 of them. They can do all you want, but they can't
monitor the temp.

bcutter, 2021-02-07:
> 
> Yes, it's OK. I have 5-6 of them. They can do all you want, but they can't monitor the temp.
> […](#)

Thanks

> 
> Yes, it's OK. I have 5-6 of them. They can do all you want, but they can't monitor the temp.
> […](#)

Thanks. Xiaomi has an internal temperature sensor to detect overtemp (automatic shutdown at 65 °C) next to overpower detection. It´s optional for me, much more important is a stressles pairing with deCONZ and painless integration into Home Assistant. I think I´ll give the BW-SHP13 a try.

jantielens, 2021-02-15:
I'm trying to get the BW-SHP13 working with my Conbee II stick plugged into a Raspberry Pi (default installation). So far I can only control the plug, but it's not reporting power consumption in Home Assistant. There are the steps I did:

Manually updated the firmware to the latest version:
[image]

Removed the BW-SHP13 light/plug from the device list, and paired it again. So the plug is added as a light: 
[image]

I can now control the plug through the Phoscon App UI and Home Assistant, but I don't see the power consumption showing up anywhere. Can you give some tips where to look? Should I find a sensor in the Phoscon App?

morfei1, 2021-02-15:
You can only see these 2 parameters in Home Assistant for example or others
third party software. You can't see them in the phoscon app.

jantielens, 2021-02-15:
Thanks, where should I look in Home Assistant? My device only shows: 
[image]

jantielens, 2021-02-15:
Ok, strange thing happened: I added another plug (which was never paired before the upgrade), and this one does report power consumption in HA! So I guess I need to figure out how to really delete the first plug. :)

morfei1, 2021-02-15:
Just read the basic descriptor of the old one or repair it again and you
gonna be fine.

userMak, 2021-02-25:
> Ok, strange thing happened: I added another plug (which was never paired before the upgrade), and this one does report power consumption in HA! So I guess I need to figure out how to really delete the first plug. :)

Hi, did you manage to add the particular plug? If yes can you explain how please? I am trying but I always got only the on off function in HA and not the sensors. Thanks

morfei1, 2021-02-25:
Update to the latest drconz and read the basic descriptor of the plug/s or
just hard rest and re-pair them without delete them from phoscon.

userMak, 2021-02-25:
> Update to the latest deconz and read the basic descriptor of the plug/s or just hard rest and re-pair them without delete them from phoscon.
> […](#)

Please (I am new to this) can you explain what do you mean to "read the basic descriptor" or give me a link to read how to hard reset the plug.

morfei1, 2021-02-25:
Check here - https://github.com/dresden-elektronik/deconz-rest-plugin/wiki

bcutter, 2021-03-06:
> 
> ...or just hard rest and re-pair them without delete them from phoscon.

I guess that somehow did the trick for me. Ended up with only switch entity after first pairing, missing current and total consumption entities. Tried everything (reading basic cluster attributes, reading simple descriptors, deleting and re-pairing), wild guess: in the end resetting the device (initializing pairing sequence by holding power button for more than 5 seconds) while having pairing mode/sensor search active in Phoscon some kind of "re-joined" the plug to the ConBee/ZigBee mesh. In Home Assistant both missing entities were immediately available.

I´ll test this with another plug of the same model, hope to finally see some plug and play (getting all entities in the first place).

SAOPP, 2021-03-08:
Guys, I plan to order a similar outlet, as far as I see there is no support of power outage memory?

Mimiix, 2021-03-09:
@SAOPP I haven't checked that tbh. I'll test later if you want?

SAOPP, 2021-03-09:
Yes sure, I'm wondering if the socket is able to adjust the state after a power loss? After power on - turn on or turn off? Like in z2m... Thanks in advance homie!

SAOPP, 2021-03-09:
And also... power and consumption is present via deconz?

SAOPP, 2021-03-09:
For now, I have full confirmation when working through z2m:

[image]

morfei1, 2021-03-09:
By default, the plug do remember the state before the power cut. If it was
previous ON, it will continue to to be ON, if Off it will stay Off. I've
checked the On/Off cluster in deCONz, but was not able to find a way to
configure it another way.

SAOPP, 2021-03-09:
This is of course good, this is similar behavior to the heiman sockets, but at the same time it is not quite what is needed, since there are sockets that must be turned off by default, even if they were turned on, this is a very big drawback of working with sockets on the side deconz

SAOPP, 2021-03-09:
Actually, this is why I keep my sockets on the z2m.

cobch, 2022-09-30:
> This is of course good, this is similar behavior to the heiman sockets, but at the same time it is not quite what is needed, since there are sockets that must be turned off by default, even if they were turned on, this is a very big drawback of working with sockets on the deconz side.

Do you know if the following as memory state on z2m?
- Aqara single module T1 (no neutral)
- Aqara smart plug

I am struggling on Deconz with that and that is really enjoying.