dresden-elektronik/deconz-rest-plugin #2989: Expected a string but was NULL at path $.config.mac - when connecting via Hue Essentials

airbone42, 2020-06-27:
My Hue Essentials connection was already working without a problem, but then it from one day to another couldn't connect anymore.

When trying to manually connect using the app I get the error "Expected a string but was NULL at path $.config.mac".

deConz is running on port 8080 instead of 80, but this is also set in the app, afaik 80 is only required for the official Hue app.

What I also checked is that the webserver is running as expected, e.g. the response of http://192.168.1.5:8080/description.xml

Restart of deConz also didn't change anything in the behavior. Does anyone have a hint what else I could try?

Thomas-Vos, 2020-06-28:
It looks like Hue Essentials has some trouble reading your bridge config. From the error it looks like the mac address could be null, which would be strange.

Could you do a GET request to `192.168.1.5:8080/api/<username>/config`? (Replace `<username>` with a valid whitelist username). Please paste the response either here or by sending an email to support@hueessentials.com.

MrMooed, 2020-06-29:
Same issue. The mac, however, seems to be null:
`{
    "apiversion":"1.16.0",
    "bridgeid":"00212EFFFF040C5E",
    "datastoreversion":"93",
    "devicename":"RaspBee",
    "factorynew":false,
    "mac":null,
    "modelid":"deCONZ",
    "name":"Phoscon-GW",
    "replacesbridgeid":null,
    "starterkitid":"",
    "swversion":"2.5.78"
}`

Thomas-Vos, 2020-07-12:
Thanks for the details. The latest version of Hue Essentials contains a fix. Could you try again and let me know how it goes?

MrMooed, 2020-08-03:
> Thanks for the details. The latest version of Hue Essentials contains a fix. Could you try again and let me know how it goes?

It works again. The mac, however, is still null.

Mimiix, 2020-08-03:
@Thomas-Vos

closed 2020-08-31 by stale[bot]