dresden-elektronik/deconz-rest-plugin #2990: Device name

madman2143, 2020-06-28:
## Device

- Product name: Wireless Switch
- Manufacturer: Sonoff
- Model identifier: SNZB-01
- Device type:
  - [ ] Light
  - [ ] Lock
  - [x] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [x] Other: Wireless Switch

[image]
[image]

madman2143, 2020-06-29:
Product Page:
https://sonoff.tech/product/smart-home-security/snzb-01

SwoopX, 2020-06-29:
Please add the missing information as mentioned in the device request, thanks!

madman2143, 2020-06-30:
Sorry, I'm new to this!

Are you talking about this?
[image]

NovaGL, 2020-07-24:
Any update on this. Do you need more information?

lbouriez, 2020-07-30:
Very interested on this switch

closed 2020-09-11 by stale[bot]

SAOPP, 2020-09-19:
Hi! Guys, I just get this device, but one question, I have only two actions: double and long, and missing single click.

[image]

[image]

[image]

[image]

Any suggestion for this situation or at current moment device is not fully supported yet?

SAOPP, 2020-09-19:
[image]

SAOPP, 2020-09-19:
[image]

Smanar, 2020-09-19:
Can you try to run deconz with log enabled, and make a short press , pls ?

SAOPP, 2020-09-19:
If I correctly understood your question 'bout values of buttons... double 1002, long 2002 (at Home Assistant events)

Smanar, 2020-09-19:
Ok so it seem this devices have other return value.
I have used the @SAOPP log to make the table, but will be nice if someone can test it ?

Modifications > https://github.com/Smanar/deconz-rest-plugin/commit/1b4540a35a1a089ce4a6af39eba2454d0a3fdb3c

To test the code

```
git clone --branch sonoff https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

bonelessc, 2020-09-19:
I would be very happy to test this, but I am running DeConz in docker. Is that possible?

Smanar, 2020-09-19:
Nope, need a classic OS to compile the code.
But not in hurry.

NovaGL, 2020-09-20:
I would like to test to but don't know how on RPI2

pippo73, 2020-09-20:
Does it works only for snzb-01 or maybe also with 03?

Smanar, 2020-09-20:
@pippo73 The 03 is a motion sensor ? not a switch ?

@NovaGL If you have a Rpi with a classic OS (not HA for exemple), you have the procedure on homepage https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

Just change a little for step (1- 4)
```
git clone --branch sonoff https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

NovaGL, 2020-09-20:
How do I install qmake?

Smanar, 2020-09-20:
You have tested 
`sudo apt install deconz-dev`

NovaGL, 2020-09-21:
Doesn't work for me, can you make a package for RPI2 and I'll test it.

NovaGL, 2020-09-21:
I tried I built it on ubuntu but get login failed

pippo73, 2020-09-21:
Yes @Smanar I'm talking about the motion sensor

SAOPP, 2020-09-21:
> I tried I built it on ubuntu but get login failed

Login failed - what u r mean? U can't login into phoscon with ur creds or?

NovaGL, 2020-09-21:
The error was misleading, it actually wasn't compiled for the right arch i believe.

Don't know what I'm doing. Tried in an ubuntu vm i can build it but don't know what arch its building for.

Any tips for compiling would be appreciated.

Smanar, 2020-09-21:
@pippo73 this code is for the switch what is the problem with the motion sensor ? It seem work since the version 79 > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2991

@NovaGL You just need to use `sudo apt install deconz-dev` it will make the correct "environement" according to your system. If this command work, the rest will be fine

And after use the same commands than me, what is your error displayed, it can come from me, I don't test all the code.

NovaGL, 2020-09-22:
Hi @Smanar I struggle at this stage
``sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins ``

Before I copy the file everything works fine, the moment I copy the file I can't access anything. No error

If you can tell me how to debug it I'll send you what error I can find but the logs I have dont show errors.

SAOPP, 2020-09-22:
See what are the access rights for the new library that you compiled?

NovaGL, 2020-09-22:
I think I was getting confused and rebooting the computer shows it not as accessible.
So it looks like it was compiled for the wrong arch when used on the PI and it shouldn't have been accessible on Ubuntu as it wasn't physically connected to it.

Is there anyway I can compile for the Raspberry PI when using Ubuntu VM.

SAOPP, 2020-09-22:
as far as I remember, it's a universal library, I can of course be wrong, the last time I compiled a few months ago

Smanar, 2020-09-22:
Compilation are machine specific, you can't compile on a machine and use the file on a different one.

>Before I copy the file everything works fine, the moment I copy the file I can't access anything. No error

The system freeze ? Better to close deconz before replacing the file.

NovaGL, 2020-09-23:
I don't have a proper computer to test it on as Hyper-V doesn't do USB Passthrough and can't compile on a RPI2 so I'll have to leave this for others to test.

pippo73, 2020-09-23:
@Smanar I've bought 5 of them, and I can connect only 2 of them. anyway I will try to ask in the 3d yo indicated me. Thank you anyway

NovaGL, 2020-09-23:
I moved to zigbee2mqtt and these devices are working flawlessly

Mimiix, 2020-09-23:
@NovaGL This is not a proper solution 😉

Mimiix, 2020-09-23:
@Smanar Do we need to re-open this?

NovaGL, 2020-09-23:
I agree this issue needs to be re-opened as it is not implemented correctly. Missing single click.

Happy to test but I just can't build it.

SwoopX, 2020-09-23:
The current code base allows you to "fix" that on your own with a little guidance.

NovaGL, 2020-09-23:
Awesome, can you point me in the right direction and I'll try and report back.

SAOPP, 2020-09-23:
I can put my raspbee at official image, so maybe I can tested it...

SwoopX, 2020-09-23:
You need to compile the current code of the offical repo. The instructions are mentioned there, in the readme.
Additionally, you need to copy file button_maps.json to the location where your database (zll.db) resides. In the json file, you cann add the missing entry which is either known or can be taken from deconz debug output.

NovaGL, 2020-09-23:
Thanks I gave up on that cause I couldn't compile it on my RPI2

Smanar, 2020-09-23:
@SwoopX It s an old code make before your change so for this one they don't need yet to use the button_maps.json

Jimmy-SafeCash, 2020-09-26:
Hey guys, has this now been tested and confirmed working and will be merged into the next release?

Smanar, 2020-09-26:
It s not tested yet ^^

Jimmy-SafeCash, 2020-09-26:
> It s not tested yet ^^

Will it be pushed into beta branch for testing? Seems odd to have half of the sonoff range working and the other half not.

Smanar, 2020-09-26:
PR done, but still untested.

Jimmy-SafeCash, 2020-09-26:
> PR done, untested, but you will be able to edit the file yourself on next version, will be json file.

Excellent, I'll wait for it to be released. Good job and thanks.

jdonofrio728, 2020-09-30:
@Smanar I just received this exact switch in the mail today and stumbled upon this thread when it didn't work. I cloned down your branch, compiled, and copied over the libde_rest_plugin.so and button_maps.json file. It didn't work for me. I still see the following logged when I do any of the button presses:

```
# Short
[INFO] - No button map for: WB01 endpoint: 0x01 cluster: 0x0006 command: 0x02 payload[0]: 000
# Double
[INFO] - No button map for: WB01 endpoint: 0x01 cluster: 0x0006 command: 0x01 payload[0]: 000
# Long
[INFO] - No button map for: WB01 endpoint: 0x01 cluster: 0x0006 command: 0x00 payload[0]: 000
```

NovaGL, 2020-09-30:
What device are you using with the combee? Maybe you have to update button maps json

jdonofrio728, 2020-09-30:
I'm using the ConBee 2. I copied the button maps file from Smanar's branch to the `./.local/share/dresden-elektronik/deCONZ/button_maps.json`. I guess I'm not sure if that is the right place, but its the same directory as the zll.db file.

NovaGL, 2020-09-30:
I mean what device is the combee connected too? A raspberry Pi?

jdonofrio728, 2020-09-30:
Oh sorry. Deconz is installed on an LXD container.

SwoopX, 2020-09-30:
Try creating a folder called devices where your zll.db resides and copy the json file there.

jdonofrio728, 2020-09-30:
@SwoopX  Yeah that made a difference. However the double press and the long press might be mixed up.

```
# Short
[INFO] - Button 1002 short
# Double
[INFO] - Button 1003 Long
# Long
Button 1004 double
```

if I edit the button_maps.json to switch the too like this then the log output is as expected:

```json
"sonoffOnOffMap": [
                        [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "double"],
                        [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Long"],
                        [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "short"]
                ]
```

Smanar, 2020-09-30:
Thx, corrected on my side.

SwoopX, 2020-09-30:
So, now with some more freedom and keyboard in front of me ;)

Manu changed the file path and obviously forgot to mention it. I found this out just yesterday myself. The path must be

`~/.local/share/dresden-elektronik/deCONZ/devices/button_maps.json`

@jdonofrio728 Glad it worked out, so the new approach already paid off.

jdonofrio728, 2020-09-30:
I will say that even though the log messages look better, something still doesn't seem right. Home assistant (out of scope here) doesn't see this switch as a "switch", rather only detects the battery level. Deconz also only shows the same.

I expected the cluster object for On/Off or something else to be available. Perhaps I'm just misunderstanding something?

SwoopX, 2020-09-30:
You got your ZHASwitch resource, don't you?

jdonofrio728, 2020-09-30:
The only entity that home assistant found was the battery:

[image]

jdonofrio728, 2020-09-30:
I'm sorry, I re-read the deconz integration documentation and found out that this is working correctly. Button pushes only fire events which I am fully able to capture. Sorry about that!

NovaGL, 2020-10-03:
Seems to be working now with the new button map which is good.

[47 more replies]