dresden-elektronik/deconz-rest-plugin #2991: Sonoff SNZB-03

madman2143, 2020-06-28:
## Device

- Product name: Sonoff SNZB-03
- Manufacturer: Sonoff
- Model identifier: SNZN-03
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [x] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Screenshots

[image]
[image]

madman2143, 2020-06-29:
Product Page:
https://sonoff.tech/product/smart-home-security/snzb-03

SwoopX, 2020-06-29:
Please add the missing information as mentioned in the device request, thanks!

madman2143, 2020-06-30:
[image]

SwoopX, 2020-07-07:
The IAS Zone cluster is still missing.

GeospatialDaryl, 2020-07-09:
Hi devs - thanks so much.  I'm also looking to use this device.  I went to supplement this request but it looks like the screenshots above include the (1) Cluster Info panel, (2) Node Info panel, and the supplemental screenshot [(3)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2991#issuecomment-651950941) has the IAS Zone information. 
[image]

I'm a new user, but wanted to help.  Can you indicate what info is still needed, and where in the codebase I can find the device definition files?  I perused the source but didn't find it.  I'd like to contribute PR.

GeospatialDaryl, 2020-07-09:
[image]

[image]

SwoopX, 2020-07-09:
[image]

Select the IAS zone cluster in your case and scroll down for the read button. Make sure the device is awake, so trigger it's function. The button must be pressed 1-2 seconds afterwards.

GeospatialDaryl, 2020-07-11:
Thanks for the update - and the clear instructions.   I can get the IAS zone cluster panel up, but can't get the read button to populate any of the attributes.  I've observed the network object on the gui blinking red on the left-hand button, but even pressing the reset/join button on the device and then re-querying with the read button hasn't returned any info.
[image]

Thanks for the great instructions and I'll continue to experiment with trying to poll the Sonoff device.

GeospatialDaryl, 2020-07-11:
Again - thanks for your patience, clear instructions, and help.
[image]

cllatser, 2020-07-13:
Good morning, I make my contribution to this sensor. When it detects movement, the check is set to Alarm 1 and in the IAS Zone cluster the zone status parameter changes to 0x0001.

Thank you so much.
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-07-13:
Yes, thanks. That's what I needed.

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

pippo73, 2020-09-23:
> Closing this as device is now supported with .79 beta.

Hi all, 
I've bought 5 sensors. I don't know why only 2 of them are connected.
I've tried to connect the other 3 in the same way but it does not work. 
What should I try? are there any magical thing to make it work?

Mimiix, 2020-09-23:
@pippo73 Feel free to open a own Issue (User question).

SwoopX, 2020-09-23:
PR is already prepared.

pippo73, 2020-09-23:
@Mimiix ok, I will

> PR is already prepared.

Sorry @SwoopX what does it mean?

Jimmy-SafeCash, 2020-10-09:
This is now working in 2.05.84 but it doesn't show within Phoscon App (just as events in HA).

Furthermore, I'm pretty underwhelmed by it compared to a Hue motion sensor. I guess it's to be expected with such a cheap price... Shame. (hoping that it can be improved through better integration with deconz perhaps?)

SwoopX, 2020-10-09:
Deconz is not Phoscon. If its missing there, feel free to raise it in the Phoscon repo.

What better integration in deconz do you expect? It is done and you get what you paid for.

Jimmy-SafeCash, 2020-10-09:
> Deconz is not Phoscon. If its missing there, feel free to raise it in the Phoscon repo.
> 
> What better integration in deconz do you expect? It is done and you get what you paid for.

Yes, perhaps should have left Phoscon out as it's not related (but thought it would be useful for the many people who will be using it).

I'm not expecting anything, just stating that for my setup the sensor doesn't work that well. I wasn't sure whether that was due to preliminary support in deconz that will be enhanced, or whether it is just symptomatic of the device and therefore nothing can be done. From your response, I take it's the latter.

SwoopX, 2020-10-09:
It's unfortunately not too uncommon that devices do not work too well or differently than expected. And than mainly happens for cheaper devices. I specifically recal the Hive motion sensor here...

If you could be a bit more elaborate on hat the device does (or doesn't do), we might judge if it is device related.

Jimmy-SafeCash, 2020-10-09:
> It's unfortunately not too uncommon that devices do not work too well or differently than expected. And than mainly happens for cheaper devices. I specifically recal the Hive motion sensor here... [...]

Sure. My only comparison is with the Hue Sensor. Compared to the Hue sensor the Sonoff is a lot less reliable. Motion is typically not detected as well (can walk past sometimes and nothing happens, other times fine), the same is witnessed when I have a no motion automation e.g. "2 mins", it doesn't seem to fire reliably. The Hue never skips a beat on that front. 

Wondering if the reset time of the firing of events can be adjusted?

damajor, 2020-11-15:
I have some trouble with these devices.

Here are the info:
[image]
[image]
[image]
[image]
[image]
[image]

And a second one (the cluster data seems to be the same as the first one):
[image]

Thoky81, 2020-12-25:
I would like to ask if somebody else have an following issue with the Sonoff Zigbee motion sensors. I have two of them in different rooms for about 2 weeks and I have realized that they trigger motions it selfs almost periodically. It doesn't matter if is night or day, one of them is triggering exactly every 1,5 hour and the other one every 2 hours. I am using several Aqara Motion sensors too but without this strange behavior. The connection in deconz is green line therefore the issue must be somewhere else.

[image]

damajor, 2020-12-25:
I just checked mine and I got a simillar behavior. The sensor seems to be
unstable and fire event when there is nothing to detect.
I cannot test much now but I think SonOff stuff is not very good quality.

>

Thoky81, 2020-12-26:
@damajor I agree with you. Another magical thing is the blind time which in my case is 10 min! If somebody have any idea how to deal with this It would be great. Thanks

SwoopX, 2020-12-26:
Can you guys please open a separate issue for that? What you experience has nothing to do with the initial device support.

Please take note that we'd require some logs catching this automatic trigger, occurring every 1.5hrs.