dresden-elektronik/deconz-rest-plugin #2992: whole system crashed :-(

ChrisPrefect, 2020-06-28:
What just happened?

Some lights where not responding anymore, so I restarted the raspi pi 4. Now Phoscon isn't loading anymore and doesn't even detect my browser:

[image]

deCONZ forgot my whole network layout (that was a lot of work to order all the 90 nodes!!)

[image]

And a third of the nodes are still not added to the network again.

All the buttons (tradfri remotes) don't work anymore. I had to go out in the rain to unscrew plugs in the garden to turn them off :-( 

This is how the mobile view looks like:

[image]

How can this happen?

This system isn't working at all. I am really down with my nerves. I can't take this anymore :-(

Mimiix, 2020-06-29:
Hi @ChrisPrefect 

Some vital information from the bug report template is missing. Could you please add the environment section?

Thank you!
- Dennis

ChrisPrefect, 2020-06-29:
## Environment

[image]

 - Host system: Raspberry Pi 4
 - Running method: beta image from phoscon with gui
 - Firmware version: newest
 - deCONZ version: .78
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes

Mimiix, 2020-06-30:
Interesting case here. Sorry, did not see the screenshot.  However, i do prefer templates above screens with these things :).

I assume here you use the ubuntu install method (https://phoscon.de/en/conbee2/install#ubuntu)?

The behaviour on the first screenshot often occurs when theres addons in chrome preventing loading. Most of the time ghostery or anything else that tries to hide your browser to websites. 

I do go with you on the fact that stuff should be auto aligning. I might create a feature request for that. Really hate it :/

The SD card from a few weeks ago, what version of deCONZ was that?

ChrisPrefect, 2020-06-30:
I used the images: https://phoscon.de/en/conbee2/sdcard
I chose the beta image and upgraded to the latest version of deconz.

Nothing changed in chrome. And it somehow works with the older clone of the SD card. 

I guess the SD card got corrupted? How can this be after about two month of using a new 32GB sandisk card? Does deconz write a lot of log files?

This is the old SD card: 
[image]

SwoopX, 2020-07-01:
@ChrisPrefect sorry, but for me, I still miss pieces of the puzzle. What crashed, what did you do (afterwards), etc.?

If devices are visible in deconz GUI, but are not reachable any more, that is a big hint that a channel change was involved which did not get propagated to all devices. However, that might still be resolvable via the hidden config page in Phoscon.

Speed up your case by joining Discord!

ChrisPrefect, 2020-07-02:
I think the SD card got corrupted? But I can't verify this. The older copy of the SD card works again (more or less).

I did not do a channel change manually. Could this happen by accident?

I tried the Discord channel, but it was not helpful, just a lot of chatter and many people talking multiple issues simultaneously. 

Is there a paid support available? I invested about 250 hours into getting my 140 node zigbee network to be reliable. Nothing helped so far. Right now, some tradfri remotes just stopped working and I have to bind them again. And Some light groups take 5-10 seconds to react to a simple on/off command with the remotes. With Alexa, only half of the lights react at all. It is so frustrating. I also added four Tradfri-repeaters to get better signal, but it did not help at all.

Smanar, 2020-07-02:
I think same, a database corruption (all is working with old SD card) or a SD card corruption.

BTW have you access at the "broken" SD card ? To try to retreive the database and copy them on the new one ? Like this you will be sure if the problem is from the DB or not, and perhaps retreive your old config.

You can find some information about the previous used channel on the hidden config here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues
And in deconz for the actual one.

ChrisPrefect, 2020-07-02:
> Wich one pairing mode have you used for tradfri remote ? Direct binding using remote group (and not light group), or Phoscon scenes ?

Uh, that sounds interesting. Where do I find information about these different binding methods? I just added lights in phoscon, added remotes and then added the remotes to light groups.

Would another binding method also solve the issue that when switching a group with alexa only a handful of lights in that group actually react?

Smanar, 2020-07-03:
It s a long story, and I haven't the resume, but it can explain why you have loose your "binding", they was stored in database too.
Realy bad luck you haven't a more recent backup.

So yes there is another method, and no, it don't solve all issues, it s just another method.

When you include a remote (no works for all remotes), you will see a new group with the remote name, this group is special, if you put a bulb in it, the 2 devices can make request in "group", and for that the coordinator is not usefull, so it can work even without conbee.
But it can work only for "native zigbee command", so on/off + dimmer ok, but not for the color.

And for "pro" users, sorry for "simplification" ^^.

closed 2020-08-01 by stale[bot]