dresden-elektronik/deconz-rest-plugin #2994: deconz.0 adapter wont start

moschnetwork, 2020-06-29:
hey guys,

i hope someone can help me.
ioBroker.admin 4.1.1
deConz Version: 2.5.72
API Version: 1.16.0

here are my log file:

`
host.raspberrypi | 2020-06-29 10:41:15.881 | info | Restart adapter system.adapter.deconz.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2020-06-29 10:41:15.881 | info | instance system.adapter.deconz.0 terminated with code 0 (NO_ERROR)
host.raspberrypi | 2020-06-29 10:41:15.880 | error | Caught by controller[1]:     at process._tickCallback (internal/process/next_tick.js:68:7)
host.raspberrypi | 2020-06-29 10:41:15.880 | error | Caught by controller[1]:     at Request.request [as _callback] (/opt/iobroker/node_modules/iobroker.deconz/main.js:859:36)
host.raspberrypi | 2020-06-29 10:41:15.880 | error | Caught by controller[1]:     at JSON.parse (<anonymous>)
host.raspberrypi | 2020-06-29 10:41:15.879 | error | Caught by controller[1]: SyntaxError: Unexpected token < in JSON at position 5
host.raspberrypi | 2020-06-29 10:41:15.878 | error | Caught  by controller[0]: This error originated either by throwing inside of an  async function without a catch block, or by rejecting a promise which  was not handled with .catch(). The promise rejected
deconz.0 | 2020-06-29 10:41:15.315 | info | (27507) Terminated (NO_ERROR): Without reason
deconz.0 | 2020-06-29 10:41:15.314 | info | (27507) terminating
deconz.0 | 2020-06-29 10:41:15.278 | info | (27507) cleaned everything up...
deconz.0 | 2020-06-29 10:41:15.275 | warn | (27507)  Could not connect to websocket instance of deConz/Phoscon. Error:  WebSocket was closed before the connection was established
deconz.0 | 2020-06-29 10:41:15.273 | error | (27507) SyntaxError: Unexpected token < in JSON at position 5     at JSON.parse (<anonymous>)     at Request.request [as _callback] (/opt/iobroker/node_modules/iobroker.deconz/main.js:859:36)     at p
deconz.0 | 2020-06-29 10:41:15.265 | error | (27507) unhandled promise rejection: Unexpected token < in JSON at position 5
deconz.0 | 2020-06-29 10:41:15.264 | error | Unhandled  promise rejection. This error originated either by throwing inside of  an async function without a catch block, or by rejecting a promise which  was not handled with .catch().
deconz.0 | 2020-06-29 10:41:15.141 | info | (27507) starting. Version 1.3.9 in /opt/iobroker/node_modules/iobroker.deconz, node: v10.21.0, js-controller: 3.1.4
host.raspberrypi | 2020-06-29 10:41:12.572 | info | instance system.adapter.deconz.0 started with pid 27507
host.raspberrypi | 2020-06-29 10:41:12.518 | info | "system.adapter.deconz.0" enabled

`

SwoopX, 2020-06-29:
This seems to be a ioBroker specific issue

Mimiix, 2020-06-29:
Hi @moschnetwork 

This issue is not related to this repository, as it seems like a ioBroker problem. Please post your issue here: https://github.com/ioBroker/ioBroker/

Closing this.

closed 2020-06-29 by Mimiix