dresden-elektronik/deconz-rest-plugin #2995: Expose more information for the centralite pearl thermostat.

apmon, 2020-06-29:
I was able to successfully add a Centralite 3157100 to my conbee I stick with 2.05.78 and do see the thermostat temperature sensor in the API and home assistant. Thanks for getting this working in #2630 However currently it seems to only expose the current temperature and not any of the other information such as temperature set point, fan setting, cooling/heating mode. If the AC/heater is actually currently running. It also doesn't appear to be settable through the API, or at least it doesn't appear in the lights section of the API. The thermostat control in home assistant therefore also doesn't appear to be able to set the thermostat. I was able to set the temperature set point directly in deconz gui and read the information from the respective zigbee clusters. Is it easy to expose these other information through the API as well? I would like to write some automations to control the A/C through a temperature sensor in a different room instead of the room the thermostat is currently physically located.

SwoopX, 2020-06-29:
The API does seem to only expose a rather limited set of attribtues in its current state. Not sure how extensive the changes might become without digging deeper into the code.

Regardless, can you please read the attributes of the pertaining clusters and chare a screenshot here?

Smanar, 2020-06-29:
For information, I m working too on thermostat, and have adding some more information, and command for API, the "mode" for exemple.
The code is soon finished.

for exemple https://github.com/Smanar/deconz-rest-plugin/commit/394d44b27d773f25d6ab44aedfa4a6fb6947b8e6#diff-97253d482e643c24fce6275e9992ceee

Smanar, 2020-06-29:
The device I m ATM
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2979 the remote
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2978 the wired device

apmon, 2020-06-30:
Thanks for looking into this!

[image]
[image]
[image]
[image]

Of those attributes, it seems "Occupied Cooling Setpoint 0x0011" "Occupied Heating Setpoint 0x0012" and "Fan mode 0x0000" would be most useful to me. Even more so if they are settable/writable through the API.

I wonder if the "Thermostat running state 0x0029" provides the current state of the thermostat, i.e. A/C or heater actively running. If that is of use, I can see if that attribute changes as the thermostat does it's thing and how to interpret those likely flags.

Smanar, 2020-06-30:
For your device It s perhaps not yet done but
0011 or 0012 are used it s the setpoint value (and it s working for other devices)
0029 is used too it's the "on" filed, but this one is read only.

ATM I have some problem with setpoint fonction, it seem the deconz core miss a field.

Smanar, 2020-06-30:
```
 * option             | read/write | attribute | description
 * -------------------|------------|-----------|---------------------
 * state.on           | read only  | 0x0029    | running state on/off
 * state.temperature  | read only  | 0x0000    | measured temperature
 * config.heatsetpoint| read write | 0x0012    | heating setpoint
 * config.mode        | read write | 0x001C    | System mode
 * config.scheduleron | read write | 0x0025    | scheduler on/off
 * config.offset      | read write | 0x0010    | temperature offset
 * config.scheduler   | read write | (command) | scheduled setpoints
```

SwoopX, 2020-07-25:
Is that stull relevant? We've put quite some work in thermostats lately.