dresden-elektronik/deconz-rest-plugin #2997: Feature Request: Auto alignment of VNC view nodes.

Mimiix, 2020-06-30:
# Feature request type
I'd like to have the VNC view to auto align. Moving those things can be a pain. 

## Description
I'd like to have a general button that just aligns the network nodes. Might be useful if you can group them too and then align yourself. However, it just feels like a pain in the ass to do it yourself.

## Considered alternatives
NONE

## Additional context
-

morfei1, 2020-06-30:
It would be a very usefull feature. It becomes even harder if one uses a mobile and lose the center of the map.

EarlOfBurton, 2021-02-14:
I would also love to have that function. For me a "grid snap" or "guidelines" would be the best solution, since I want to adjust the position of the nodes myself (sorting by room or type...).
Since all nodes are the same size, maybe a "fixed grid" that can be turned off and on, would be a good solution? Just for reference I'm adding a screenshot of a mostly sorted system, how I would like to have a kind of snap align. Kind regards.
[image]

univok, 2021-03-13:
I would appreciate a grid as well. Plus something like a "zoom all button", because you can loose overview completly when accidentally moving the scroll bars or zooming. It's like looking for a stamp in the dark on the grass of a huge soccer field with a tiny flashlight.

EarlOfBurton, 2021-03-14:
> Plus something like a "zoom all button"

True that! Great idea.

MarkoPaasila, 2021-04-08:
I would like to have nodes auto-arrange according to some algorithm, like [the D3 force-directed-graph](https://observablehq.com/@d3/force-directed-graph) based on link strength. That would naturally group strongly linked nodes together - and the opposite for weak links. Several other softwares have something like this (D3, yED, graphviz), so it shouldn't be too hard to implement.

rEs-84, 2022-12-06:
Might of zoomed or moved the nodes out of the view.
Still no way to reset the map/view back to normal? or locate a node on the map from double clicking it from the list?

Mimiix, 2025-05-12:
@manup  Can we have this now 🥺