dresden-elektronik/deconz-rest-plugin #2998: Deconz change itself it's channel and pan id -> all lights not reachable

realwax, 2020-06-30:
Today  I noticed a major issue. 
Lights were suddenly not reachable. Ikea and dresden elektronik flspp.
A closer look showed that network id and channel changed itself!
I did a scan to get back the pan id and reentered it in the gui and switched channels back. 
Lights are still not reachable. What am I missing? 
Reference to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1226
@manup You helped out there. Can you help me please. Readding all is massive...

Cheers

Wolfgang

Smanar, 2020-06-30:
Hello, have you tried the hidden config https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues ?

realwax, 2020-06-30:
@Smanar Thanks - yes - right now - that worked. 
The interesting thing is changing back pan id and network key wasn't enough. Rolling back config did it.

closed 2020-06-30 by realwax