dresden-elektronik/deconz-rest-plugin #2999: Tuya/Jinvoo curtain switch (SM-SW101-ZC)

Pedder007, 2020-06-30:
## Device

- Product name: SM-SW101-ZC
- Manufacturer: Tuya/Jinvoo (Heiman, read from the node list as vendor)
- Model identifier: TS0302
- Device type: 
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [X] Other: Smart Wall Switch / Curtain Switch

## Screenshots

[image]

[image]

[image]

[image]
[image]

[image]
[image]
[image]
[image]
[image]

Pedder007, 2020-07-02:
Update Basic Cluster Info (did reading again)
[image]

Having now the manufacturer name in the field: _TYZB01_dazsid15
I searched an found this: https://github.com/Koenkk/zigbee-herdsman-converters/issues/621

I Also tested to steer the device direct from the deConz Hui and it worked:
[image]

So it seems already to work with the ConBee II stick and deConz, but not in Pimatic, which might have to be solved there then

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

Pedder007, 2020-07-19:
Hey @SwoopX, thanks for your effort 👍
... but I‘m a bit confused 😊What does ‚supported with .79 beta‘ mean?
Can I use the device now by Pimatic?

SwoopX, 2020-07-19:
When you have subscribed to the apt beta channel and updated to the current version, then yes. Also marthoc should have updated the container already.

Pedder007, 2020-07-19:
Not quiet sure what you mean, but I will spend some research and convinced that I will find the answer 😉
Thanks again!

Pedder007, 2020-07-20:
Hi again @SwoopX, I'm sorry but I couldn't find anything which seems to fit to your hint. I still have no clue at all what I have to update from where. I'm running Pimatic with the raspbee 0.1.5 plugin on a Raspi4 with a ConBee II and deConz in headless mode. 
Phoscon Vers. is 2.05.75 and stick firmware is 264A0700. 
Could you support with a link? That would be really great :-)

SwoopX, 2020-07-20:
@Pedder007 I'm afraid I dont know Pimatic so I probably cannot help you there. How have you installed deconz anyway. You are already 4 versions behind 😉

If it is a package pimatics folks provide, they maybe should update.

Pedder007, 2020-07-20:
Ok, I understand this an searched a bit more. I think I now also got your point. You updated/enhanced the latest deconz version 2.05.79... to use the curtain switch, right?
I was a bit lost as you only wrote .79 :-)
If this is the case, then it‘s definitely a question of integration in the respective Pimatic plugin.
As written somewhere, via the Deconz-gui I have already been able to steer the device, also with .75.
Thanks for now then 👍

rkupfer2020, 2020-07-23:
Hi, 
I use the same Heimann / _TYZB01_dazsid15 curtain switch with Conbee and Deconz. I already upated the Docker Container to 2.05.79. But I am still not able to use the Deconz GUI to open/close the curtains. Only with VNC they respond to the commands.
Under Home Assistant I can open and stop the curtains, but not close them. Anybody has an idea how to solve the issue?
Best regards
Roger

SwoopX, 2020-07-23:
> But I am still not able to use the Deconz GUI to open/close the curtains.

I assume you mean Phoscon in your case. This is kind of normal as devices have to be added seperately. I'd recommend to raise a corresponding request at the Phoscon repository.

> Only with VNC they respond to the commands.

This is actually the deconz GUI. If all works well there, then the device doesn't have any issue.

> Under Home Assistant I can open and stop the curtains, but not close them.

Probably the wrong API commands are used in this case. See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3063

cobirnm, 2020-09-05:
> ## Device
> * Product name: SM-SW101-ZC
> * Manufacturer: Tuya/Jinvoo (Heiman, read from the node list as vendor)
> * Model identifier: TS0302
> * Device type:
>
>   * [ ]  Light
>   * [ ]  Lock
>   * [ ] [...]

Have you been able to Calibrate the Switch?

djfanatix, 2020-09-10:
Did you try to modify 0x0017 or 0x007 to 0x002 (calibration mode)? Does this work?

Pedder007, 2020-09-21:
Hi, sorry didn't find the time to continue here earlier.
I now just tested again (after first updating the PhosconApp and ConBee II firmware to newest releases).
The PhosconApp now shows the device, but as a 'light'!? This happened three times now, allthough I used the discovering mode for switches.
Via the deconz-gui I was able to switch up/down/close, so for me nothing seems having changed to the status above.
I fear as long as the device is only discovered as a light I will also not get it integrated in my HO Pimatic.