dresden-elektronik/deconz-rest-plugin #3000: Conbee II restarts

probize, 2020-06-30:
I have now battled with this for months. I have been running Hassio on a (virtual) ubuntu LTS 18 on my Synolgy DS918+ where i have attached the Conbee II to the VM. It has been working fine. And I have been following the updates from hassio over time. From time to time i lost the USB device and had to re-connect it in my management for the VMs. When it happens I of cause lost the connectivity to the Zigbee network and reboots of the Deconz was also optimal. Lately i lost the USB more often, and sometimes It was almost impossible to attach the USB again. I then took the stick out and upgraded it 

from : deCONZ_ConBeeII_0x264a0700.bin.GCF
to : deCONZ_ConBeeII_0x26580700.bin.GCF

I moved the stick to my Windows 10, and now I heard the USB connect/disconnect sounds multiple time both before and after upgrade. I tried to move it back, but it did not help at all so i ordered a new one. And firmware upgraded to the latest release. It was then running for some time but after some time i was losing the connection again. and i decided to install the system direct on the Synology docker instance (https://community.home-assistant.io/t/hass-io-on-synology-dsm-native-package/125559)

I have now downgraded and the disconnect seem a bit less.

Events from the log
device disconnected reason: 1, index: 0 (reason: 4, also seen)
device state timeout (handled)
void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
wait reconnect 15 seconds
wait reconnect 1-15

System is sometimes operational for hours and days but could as well reset just after reconnecting.

I have searched the forums and seen that access right could be the cause as well as drivers. I have followed a lot of the recommendations but i even has issues on Win10 so i have sticked with that issue for now. the recommended driver is not upgradable to since it isnt signed

CFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
\\.\COM4         | 0x1CF1 | 0x0030  | DE2214291  | ConBee II

Smanar, 2020-07-01:
Using windows is a good idea.
You have a procedure here to update the firmware with a longuer timeout https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493
You can update with the same firmware.

When you have made the update on wndows, do you have used this command ? Do you still listen the "USB connect/disconnect" ? If you still listen it after the update, you you have the command result ?

probize, 2020-07-01:
Yes, I have no problems upgrading and downgrading the firmware. I have only used the 0x264a0700 and 0x26580700. I have been able to update as well from the docker container. 

After the upgrade, I still get the Windows Announcement (tone) from time to time. I have started deConz up on my WIN10 and I see the network forming, but I also experience that the application closes the network view and i get back to the view where I can connect. I have searched for debug for windows but have not found any... Only on Linux so I do now know if it is a bad platform to troubleshoot from... I thinking about starting up one of mine Raspberry 3Bs and connect it here. But the wife is not happy at the moment... She wishes we were back with manual curtains.

Mimiix, 2020-07-01:
This feels like a broken device. Maybe due to the heat(when using it, it warms up) stuff gets broken.

Smanar, 2020-07-01:
I think same, it s realy stable on windows, if you have too this problem on windows it s not good.
You can try the USB extension, or an HUS HUB, but for me you can contact the support too.

probize, 2020-07-01:
I thought as well it was a hardware defect. That is why I ordered a new one. But the new one had a lower Id. I use a 2 meter usb2 extension cord. The conbee 2 is placed at my roof, just next to my Synology and cable modem but they run without any issue. The Conbee 2 may only have run for a hour or two before it began the boot loop.

manup, 2020-07-01:
Hmm with firmware version 0x26580700 and an USB extension cable it should be way more stable especially on native OS like Win10. One important thing to keep in mind is to bring the ConBee II as far away as possible from any hard drives or USB 3.0 devices and perhaps also try different USB ports if that's possible since these can be really nasty to 2.4 GHz RF like Zigbee.

Smanar, 2020-07-02:
The second conbee have the same disconnection problem ?

probize, 2020-07-03:
Yes the condition are the same.,,, I have bought an active 3M USB extender (https://www.deltaco.se/sites/cdn/InformationDocumentsLibrary/USB2-EX3M_PRIME_Datablad.pdf) but the result are the same, I will properbly have to get my 3rd device...

Smanar, 2020-07-03:
So 2 differents devices, same problem on windows + USB extension cable, it works for exemple during 1h then the device disconnect  ? And you have tried the firmware update on both (do you have the command result) ?

>bring the ConBee II as far away as possible from any hard drives or USB 3.0 devices and perhaps also try different USB ports if that's possible since these can be really nasty to 2.4 GHz RF like Zigbee

The power alimentation can be an issue too, but I never see it on a WIN10 machine.

But seriously you can contact the support, they can propose an exchange

probize, 2020-07-03:
Yes i find it very odd. I have been running with my fist devices for months. And yes from time to time I had to mount the USB multiple times before the VM attached to it. And sometimes it my have rebooted sense it stopped replying on my request from home assistant or the scheduled task did not complete  all my blinds procedures. I did not investigate it that much since there was always a new thing going on with home assistant. But finally my wife complained and I found the issue. I wonder much why the second device in fact is even worse than the first, but as well only temporary. 

Before I wrote this email I made my order on Amazon. So 3rd device it is.

probize, 2020-07-11:
New Conbee 2 now mounted (3rd) device, it seems like the restarts is gone... The docker instance is totally cleared and i have chosen not to firmware upgrade, But it flaging for an upgrade in deconz 2.05.78 22.5.2020 so i upgraded from the Phoscon Web inteface 

264A0700 from 
26580700 to

It seems not to have any restarts anymore but the log is clogged with the database open and state 3

23:38:43:471 don't close database yet, keep open for 900 seconds
23:38:48:220 device state timeout ignored in state 3
23:38:56:577 don't close database yet, keep open for 900 seconds
23:38:57:720 device state timeout ignored in state 3
23:39:03:461 don't close database yet, keep open for 900 seconds
23:39:08:220 device state timeout ignored in state 3

Mimiix, 2020-07-12:
As we told you on discord, that's just a info message. Nothing wrong.

probize, 2020-07-12:
Hi Mimiix you are quite fast across all platforms. I have waited a bit for updating this. But it seems like the new conbee ii works perfectly. I had it running for 2 days on my laptop and it kept the connections and did not come with the windows sound of disconnect and connect in that period.

closed 2020-07-12 by probize

lukinno1, 2020-11-20:
Hello 

My conbee II usb is still disconnect/connect every 5s.
Problem started after update to last firmware. 
I use windows 10.

How can i solve this problem?

Mimiix, 2020-11-20:
@lukinno1 Can you open a own issue?