dresden-elektronik/deconz-rest-plugin #3001: Conbee II - endless connecting/disconnecting loop on USB level

MNicks, 2020-07-01:
Not sure if this is the right place to report Conbee II specific issues which are not directly related to deCONZ itself, but I give it try...

In my case I'm running Hassio as VM within Proxmox on an Intel Nuc with an attached Conbee II USB stick. This setup runs over month w/o any problems. But out of nowhere since some days I discovered the following statements in the Syslog. The Conbee II stick seems to be have trouble on USB level and sticks into a endless connecting/disconnecting loop. 

As I did no changes on hardware level (e.g. no firmware updates, no other USB devices, ...) my first idea was the stick is somehow technically broken. But looking to the issue list (e.g. #1598, #3000) it seems others have similar problems. 

Unfortunately I wasn't able to find an answer on what causes the problem or how to fix it and would be appreciated for any help.
 
```
[...]
Jul 01 01:47:27 pve kernel: usb 1-3: USB disconnect, device number 74
Jul 01 01:47:27 pve kernel: usb 1-3: new full-speed USB device number 75 using xhci_hcd
Jul 01 01:47:32 pve kernel: usb 1-3: unable to read config index 0 descriptor/all
Jul 01 01:47:32 pve kernel: usb 1-3: can't read configurations, error -110
Jul 01 01:47:32 pve kernel: usb 1-3: new full-speed USB device number 76 using xhci_hcd
Jul 01 01:47:33 pve kernel: usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Jul 01 01:47:33 pve kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul 01 01:47:33 pve kernel: usb 1-3: Product: ConBee II
Jul 01 01:47:33 pve kernel: usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Jul 01 01:47:33 pve kernel: usb 1-3: SerialNumber: DE1964774
Jul 01 01:47:33 pve kernel: cdc_acm 1-3:1.0: ttyACM0: USB ACM device
Jul 01 01:47:54 pve kernel: usb 1-3: USB disconnect, device number 76
Jul 01 01:47:54 pve kernel: usb 1-3: new full-speed USB device number 77 using xhci_hcd
Jul 01 01:47:54 pve kernel: usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Jul 01 01:47:54 pve kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul 01 01:47:54 pve kernel: usb 1-3: Product: ConBee II
Jul 01 01:47:54 pve kernel: usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Jul 01 01:47:54 pve kernel: usb 1-3: SerialNumber: DE1964774
Jul 01 01:47:54 pve kernel: cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[...]
```

Smanar, 2020-07-01:
Here you have a procedure with longuer timeout https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493 using (-t 60)

Make the try on windows, it s easier, because you will have less problem on it, and if you still have problem you will listen the usb disconnection/reconnection sound.

You can update with the same firmware you already have.

Mimiix, 2020-07-01:
Goodmorning @MNicks , 

No worries, you are in the right spot.

Are you able to add the stick on a windows native machine and see what happens then? Does it still do this?

Just to make sure it is not a hardware fault.

MNicks, 2020-07-02:
On Windows the stick initially shows the same behavior (continuous registering and deregister as COM device) but nevertheless deConz as well as GCFFlasher were able to pick up the device. This gave me the change to update to the latest firmware version (0x26580700) but with this deConz wasn't able to establish a connection to Conbee anymore. So I started a 2nd try by using the previous firmware version (0x26570700) which does the trick and the stick now is back to life.
I've still no idea what causes this problem, but as everything is now back to normal I will close this issue.

closed 2020-07-02 by MNicks

Smanar, 2020-07-03:
I m seing this problem every week, but the good new is it happen only one time at the same user (for the moment)

antonakv, 2020-07-04:
I have the same reconnects in loop with usb cable and without
```
[  224.732489] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[  232.959705] usb 1-1.3: new full-speed USB device number 20 using xhci_hcd
[  233.098801] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  233.098809] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  233.098814] usb 1-1.3: Product: ConBee II
[  233.098819] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  233.098824] usb 1-1.3: SerialNumber: DE2191879
[  233.105472] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device
[  237.021676] usb 1-1.3: USB disconnect, device number 20
[  237.022202] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[  245.249780] usb 1-1.3: new full-speed USB device number 21 using xhci_hcd
[  245.388802] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  245.388819] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  245.388832] usb 1-1.3: Product: ConBee II
[  245.388846] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  245.388858] usb 1-1.3: SerialNumber: DE2191879
[  245.401631] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device
[  249.311694] usb 1-1.3: USB disconnect, device number 21
[  249.312109] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[  257.799785] usb 1-1.3: new full-speed USB device number 22 using xhci_hcd
[  257.938934] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  257.938942] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  257.938948] usb 1-1.3: Product: ConBee II
[  257.938953] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  257.938957] usb 1-1.3: SerialNumber: DE2191879
[  257.941646] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device
[  261.857547] usb 1-1.3: USB disconnect, device number 22
[  261.857947] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[  270.090109] usb 1-1.3: new full-speed USB device number 23 using xhci_hcd
[  270.229207] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  270.229223] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  270.229236] usb 1-1.3: Product: ConBee II
[  270.229249] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  270.229262] usb 1-1.3: SerialNumber: DE2191879
[  270.234057] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device
[  274.147281] usb 1-1.3: USB disconnect, device number 23
[  274.147704] cdc_acm 1-1.3:1.0: failed to set dtr/rts
```
Firmware is latest offered by Phoscon app. I was able to update to the latest one, then after reboot I see reconnect loop

Smanar, 2020-07-04:
You can use the same you already have, but can you show use the command line you have used and the result ?

SwoopX, 2020-07-04:
@antonakv If you're up to an experiment which could potentially solve this, ping me on our Discord.

antonakv, 2020-07-06:
@SwoopX I'm up to an experiment. Let's try.

SwoopX, 2020-07-06:
Cool. Like I said, would be best to ping me in our Discord (see readme). Best in about 2h from now. Would also be more interactive

SwoopX, 2020-07-08:
@antonakv It appears that `ModemManager` is running. Please stop or disable it and try again. The latest combination of deconz and firmware should prevent that in future.

pittbull, 2020-09-07:
I am experiencing this issue now running Home Assistant on Proxmox. There is now ModemManager in HASS so how can I get my Conbee II stable?

Smanar, 2020-09-07:
Have you tried the procedure ? Or make in on another machine, in all case you can't do it on HASSIO.

pittbull, 2020-09-07:
By procedure you mean updating the firmware?

pittbull, 2020-09-07:
Connecting it to my physical Win 10 produces only connects and reconnects as well.

Smanar, 2020-09-07:
Yep, perfect.
So make the firmware update using windows (even the same firwmare you have), and with the command with "-t 60", and after you can try again the conbee on the proxmox

pittbull, 2020-09-07:
Well - kinda hard to do when it's stuck in a connect/reconnect loop and neither GCFFlasher nor deConz finds the stick.

pittbull, 2020-09-07:
Went by the store and exchanged it, and when I plugged it in Windows found it and tried to install a driver. That failed. GCFFlasher or deConz still cannot find it and I'm getting the connect/disconnect sounds repeatedly.

Smanar, 2020-09-07:
Even the command `GCFFlasher -d COM7 -t 60 -f ../firmware/deCONZ_ConBeeII_0x26580700.bin.GCF` are not working ?

Do you have a capture with the error message on windows

pittbull, 2020-09-07:
The command shows nothing connected. Left the stick in for 10 minutes and repeatedly retried the command - nothing.

Windows did recognise a device when I first plugged it in, but failed to find a driver for it.

Smanar, 2020-09-07:
Perhaps it s a bad port ?
Have you tried to use the command "GCFFlasher -l" repeatedly (you need to have running when the conbee is connected, and with the disconnection ...) to be sure for the port ?

You can try to "-R 100" in the command (when you are sure for the port) but not sure it work on windows.

pittbull, 2020-09-07:
No. This has nothing to do with the port on the computer. This started initially as a disconnect/reconnect issue on Proxmox. I moved to the Windows 10 on your recommendation, and I see the same behaviour: disconnects and reconnects.

I have, as stated, been at the store just now and exchanged the device for a new one - with the same result; both on Windows 10 and Proxmox.

Smanar, 2020-09-07:
You have a store where you can buy conbee next to you ?

I know it s a new device, but I think there is more chance you have the same problem on 2 conbee (this problem is realy classic) than you have a problem  on a promox AND a windows system.

Remember you need to use a real computer not a VM.
Can you make a copy paste with the command line you have used and the error message from it ?

pittbull, 2020-09-07:
Yes, I have. (Lucky me!)

Agreed - this is a problem with the Conbee usb device.

I _am_ using a physical Windows laptop.

I was able to install your supplied driver, and upon doing so the device became available on COM6. I proceeded to patch firmware 26580700 successfully and I will now retry the device on my Proxmox server.

Should this repatching of the firmware eliminate the connect/reconnects?

Smanar, 2020-09-07:
The driver is only for win7, the support in win10 is native (but I m not sure)

And yes, I have already see  several dizanes users with this problem, but I haven't see someone have the bug 2 times.
The majority have it at first try

pittbull, 2020-09-07:
Well - the patching worked so that is out of the way.

I do still seem to have the same issues on my server. I am testing and debugging now.

pittbull, 2020-09-07:
Well, we are back to square one:

- new issue: "failed to set dtr/rts"
- multiple connec/reconnect
Result: VM locks up

This is really not very good. How can we mitigate this?

Smanar, 2020-09-07:
On your VM, If I remember you have a setting to configure usb 2.0/3.0, set it to 2.0

pittbull, 2020-09-07:
ModemManager is not loaded.

I can choose to force it to USB 3.0 so I assume that makes it USB 2.0 implicitly.

Smanar, 2020-09-07:
No zigbee don't like usb 3.0 ^^. to much perturbation, better to use usb 2.0

pittbull, 2020-09-07:
I am not using USB 3.0 I was just answering your question.

So - clearly this is a problem with the usb stick, question:

- Do you have a firmware without the bug that loops the restarts?

Smanar, 2020-09-07:
I don't think the problem is from the stick, re-make a try on windows, I don't think you still have the disconnect loop.

pittbull, 2020-09-07:
So what do you think is the case?

pittbull, 2020-09-07:
A small extract of my logs. 

`Sep  7 23:20:19 pve kernel: [19520.505473] cdc_acm 1-8.3.1:1.0: ttyACM0: USB ACM device
Sep  7 23:20:18 pve kernel: [19519.857934] usb 1-8.3.1: USB disconnect, device number 90
Sep  7 23:20:19 pve kernel: [19520.505023] usb 1-8.3.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  7 23:20:19 pve kernel: [19520.505026] usb 1-8.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  7 23:20:19 pve kernel: [19520.505028] usb 1-8.3.1: Product: ConBee II
Sep  7 23:20:19 pve kernel: [19520.505029] usb 1-8.3.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  7 23:20:19 pve kernel: [19520.505030] usb 1-8.3.1: SerialNumber: DE2228163
Sep  7 23:20:19 pve kernel: [19520.505473] cdc_acm 1-8.3.1:1.0: ttyACM0: USB ACM device
Sep  7 23:20:22 pve kernel: [19524.341768] usb 1-8.3.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  7 23:20:22 pve kernel: [19524.341772] usb 1-8.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  7 23:20:22 pve kernel: [19524.341774] usb 1-8.3.1: Product: ConBee II
Sep  7 23:20:22 pve kernel: [19524.341776] usb 1-8.3.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  7 23:20:22 pve kernel: [19524.341778] usb 1-8.3.1: SerialNumber: DE2228163
Sep  7 23:20:22 pve kernel: [19524.341768] usb 1-8.3.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  7 23:20:22 pve kernel: [19524.341772] usb 1-8.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  7 23:20:22 pve kernel: [19524.341774] usb 1-8.3.1: Product: ConBee II
Sep  7 23:20:22 pve kernel: [19524.341776] usb 1-8.3.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  7 23:20:22 pve kernel: [19524.341778] usb 1-8.3.1: SerialNumber: DE2228163
Sep  7 23:20:41 pve kernel: [19543.153909] usb 1-8.3.1: USB disconnect, device number 92
Sep  7 23:20:41 pve kernel: [19543.153909] usb 1-8.3.1: USB disconnect, device number 92
Sep  7 23:20:45 pve kernel: [19546.993918] usb 1-8.3.1: USB disconnect, device number 93
Sep  7 23:20:46 pve kernel: [19547.462032] usb 1-8.3.1: new full-speed USB device number 94 using ehci-pci
Sep  7 23:20:46 pve kernel: [19547.673659] usb 1-8.3.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  7 23:20:46 pve kernel: [19547.673663] usb 1-8.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  7 23:20:46 pve kernel: [19547.673665] usb 1-8.3.1: Product: ConBee II
Sep  7 23:20:46 pve kernel: [19547.673667] usb 1-8.3.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  7 23:20:46 pve kernel: [19547.673669] usb 1-8.3.1: SerialNumber: DE2228163
Sep  7 23:20:45 pve kernel: [19546.993918] usb 1-8.3.1: USB disconnect, device number 93
Sep  7 23:20:46 pve kernel: [19547.462032] usb 1-8.3.1: new full-speed USB device number 94 using ehci-pci
Sep  7 23:20:46 pve kernel: [19547.673659] usb 1-8.3.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  7 23:20:46 pve kernel: [19547.673663] usb 1-8.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  7 23:20:46 pve kernel: [19547.673665] usb 1-8.3.1: Product: ConBee II
Sep  7 23:20:46 pve kernel: [19547.673667] usb 1-8.3.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  7 23:20:46 pve kernel: [19547.673669] usb 1-8.3.1: SerialNumber: DE2228163
Sep  7 23:21:06 pve kernel: [19567.945999] usb 1-8.3.1: new full-speed USB device number 95 using ehci-pci
Sep  7 23:21:06 pve kernel: [19567.945999] usb 1-8.3.1: new full-speed USB device number 95 using ehci-pci
Sep  7 23:21:09 pve kernel: [19571.313943] usb 1-8.3.1: USB disconnect, device number 95
Sep  7 23:21:09 pve kernel: [19571.313943] usb 1-8.3.1: USB disconnect, device number 95
Sep  7 23:21:10 pve kernel: [19572.010387] cdc_acm 1-8.3.1:1.0: ttyACM0: USB ACM device
Sep  7 23:21:10 pve kernel: [19572.010387] cdc_acm 1-8.3.1:1.0: ttyACM0: USB ACM device
Sep  7 23:21:25 pve kernel: [19586.929914] usb 1-8.3.1: USB disconnect, device number 96
Sep  7 23:21:25 pve kernel: [19586.929914] usb 1-8.3.1: USB disconnect, device number 96`

Mimiix, 2020-09-08:
@pitbull If you are using a VM: Please note that this is not recommened or even supported. Use a native install method.

I still believe it's a firmware issue here. Follow [this guide](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) to update accordingly.

If you are still having issues: Create a own user question issue 😄

pittbull, 2020-09-08:
@Mimiix - not sure what you are referring to as not recommended or supported? Are you saying that using the Conbee II in a virtualised environment is not recommended or supported?

For the sake of clarity I am not trying to patch the firmware on the Proxmox server, for that I use my Windows laptop. My problem though is on the Proxmox with repeated connects and reconnects (as per log)

Smanar, 2020-09-08:
> 
> 
> So what do you think is the case?

You still have the bug on windows ?

pittbull, 2020-09-08:
@Smanar No - when I installed the driver you suggested I was able to find the device on COM6 and successfully patched it with the latest firmware.

Mimiix, 2020-09-08:
@pittbull I am saying that it is not recommend to update the firmware in a VM. Usage is absolutely fine (Tho i do personally recommend to use a native OS but i am using a VM too 😄)>
 
But, please open your [own issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=User+Question&template=user-question.md&title=) instead of continuing here.

pittbull, 2020-09-08:
Great - got a little nervous there. ;)

An update from my side: As i continued to see the reconnect-loop I googled a bit further and found a post over at Proxmox stating that others where having the same issue and that one resolution was to downgrade the Proxmox version. 

I have now downgraded it (and all the vm's are still running - phew), and now I have added the usb port to HA again and from the looks of things it works. There are no usb traces in the logs aside from the reset that happens when the vm boots, so fingers crossed that this issue is resolved. :)

Thank you kindly for your insights and comments.

Smanar, 2020-09-08:
Ha nice, good to know. you are not the only issue with proxmox here. Thx.

adrianmihalko, 2020-09-10:
I have same problem on Esxi, endless connect-disconnect.

Smanar, 2020-09-10:
Have you try the command line with "t 60" on a real machine ?

Mimiix, 2020-09-11:
@adrianmihalko Please open a own issue  :)

0x53A, 2021-10-02:
I know this is an old issue, but since this is one of the top search results, I wanted to add that the [method above](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3001#issuecomment-688474820) worked perfectly for me.

I got my new Conbee II today. It worked fine, then I updated the firmware, then it got stuck in a connect-disconnect loop.
The fw-update was done through the phoscon web app.

If I plugged it into a USB port on windows, I got the de-do-di (connected) and de-du-do (disconnected) every few seconds.

-----------

I opened device manager and saw that under ``Ports``, the COM5 was connecting/disconnecting.

Then I downloaded the second-latest firmware from http://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D.

Running ``GCFFlasher -d COM5 -t 60 -f ./deCONZ_ConBeeII_0x26700700.bin.GCF`` worked. It waited a few seconds for the next connect, and then successfully flashed it.

Important: You need to check which COM port number it is on your local computer and change the cli arguments to that.

bohmandan, 2021-12-12:
> I know this is an old issue, but since this is one of the top search results, I wanted to add that the [method [...]

Same here! Worked fine even though it kept up with the reconnect loops.

antonakv, 2021-12-13:
I just thrown away Conbee II to garbage after that firmware update loop I had. And use other Zigbee 3 usb device. 
This bug is not acceptable, and it's good call to never use Conbee II anymore.

Bellavista, 2022-04-20:
I have a homeassistant installation on a virtual machine on pc windows 10, conbee II, zegbee2MQTT 
After a crash ... days of beeps ... corrupt databases due to constant disconnection ...
I downgraded to deCONZ_ConBeeII_0x26700700.bin.GCF.
for the moment a quiet night
[image]

Bellavista, 2022-05-06:
Nothing to do!!!!
I readong that the problem has existed for about 2 years and there is no valid solution
the only thing I can think of ...... I do not recommend buying the Combee II

Mimiix, 2022-05-06:
This is most, if not all, the time related to the virtual machine software, power supply or interference.

I am closing this issue as it's growing into an complaint form. Feel free to ask help on the forums if you encounter this issue : forum.phoscon.de