dresden-elektronik/deconz-rest-plugin #3002: Mute version warning if latest version is beta/pre release and not released

mruepp, 2020-07-01:
## Describe the bug
Latest release is labeled as beta/pre release 2.5.78 and not yet released in the stable apt channel 
http://phoscon.de/apt/deconz buster InRelease

Despite this, homebridge.log prints:
raspbee: warning: not using recommended deCONZ gateway version 2.5.78

## Steps to reproduce the behavior
Install latest deconz 2.05.77 from http://phoscon.de/apt/deconz buster InRelease 

## Expected behavior
No warning if using stable channel

## Environment
Raspberry Pi 3 B+
Raspbee
Deconz Headless

Node.js Version | v12.18.
NPM v6.14.5
Phoscon 2.05.77
Raspbee Firmware 26350500

SwoopX, 2020-07-01:
This is nothing what originates from deconz...

Mimiix, 2020-07-01:
@ebaauw Can you elaborate on this? Is this indeed a homebridge issue or is it with us?

mruepp, 2020-07-01:
Its a homebridge logger message:
[7/1/2020, 8:45:44 PM] [Hue] Initializing Hue platform...
[7/1/2020, 8:45:44 PM] [Hue] homebridge-hue v0.11.72, node v12.18.2, homebridge v1.1.1
[7/1/2020, 8:45:44 PM] [Hue] warning: not using recommended node version v12.18.1 LTS
...
[7/1/2020, 8:45:45 PM] [Hue] raspbee: dresden elektronik deCONZ gateway v2.5.77, api v1.16.0
[7/1/2020, 8:45:45 PM] [Hue] raspbee: warning: not using recommended deCONZ gateway version 2.5.78

ebaauw, 2020-07-02:
Homebridge Hue v0.11.72 has been developed and tested against deCONZ v2.05.78.

morfei1, 2020-07-02:
It's taken from discord on 30.06.2020 - user **@johnnygeezz**

"I see this message in my Homebridge log:    

[6/30/2020, 07:31:52] [Hue] Phoscon-GW: dresden elektronik deCONZ gateway v2.5.77, api v1.16.0
[6/30/2020, 07:31:52] [Hue] Phoscon-GW: warning: not using recommended deCONZ gateway version 2.5.78

Where can I find the instructions to update to 2.5.78?
or does it matter?"

mruepp, 2020-07-02:
The only issue is that  deCONZ v2.05.78 is labeled prerelease and is not released on the stable channel as latest but Homebridge Hue v0.11.72 is tested against and released as latest. If these are dependent, I would expect an apt-get update && apt-get upgrade to get the latest deconz package. If v2.05.78 will be released, then its just a matter of waiting and everything resolves ok.

Mimiix, 2020-07-02:
@mruepp That is something on homebridge's end, not on ours. 

I am closing this issue as this is not on our end :)

closed 2020-07-02 by Mimiix