dresden-elektronik/deconz-rest-plugin #3003: Heiman Siren does not trigger alert

sonofsilberling, 2020-07-01:
## Describe the bug
Via node-red, the Heiman Siren (warning device) triggered nicely by setting alert to "lselect" or "blink" and turned off by setting alert to "none" - but not anymore while I cannot say when this happened.
The siren works, it is connected and does make a very loud sound when I trigger it through Phoscon.

The error node red shows is
"{"address":"/lights/11/state/on","description":"parameter, on, not available","type":6,"post":{"on":true,"alert":"blink"}}"
Somehow it looks like something is adding a "on":"true" command. See node-red setup in the attached screenshot

[image]

Is this a setup error in node-red or am I missing something?

## Steps to reproduce the behavior
Set alert to "lselect" or "blink"

## Expected behavior
A very loud or a very blinking Heiman siren

## Environment
- Deconz 2.05.75
- Firmware 264A0700 
- Device: Conbee II
- Host system: Raspberry Pi 
- Running method: Raspian 
- Do you use an USB extension cable: yes

## Additional context
Lights api:
"
"etag": "d9d50f8ac5e7e3f4ba70607951a49c42",
"hascolor": false,
"manufacturername": "Heiman",
"modelid": "WarningDevice",
"name": "Warning device 11",
"state":{
"alert": "none",
"reachable": true
},
"swversion": "2018.03.21",
"type": "Warning device"
"

SwoopX, 2020-07-01:
Does it react as expected on `lselect` and just failing on `blink`?

sonofsilberling, 2020-07-01:
I'm afraid the result on setting alert to "lselect" or "select" is the same outcome and the same message.

SwoopX, 2020-07-07:
Does that also happen when you directly issue this via REST API, so without NR?

sonofsilberling, 2020-07-08:
Thank you for following this up.

In my case lights/11 is the siren and a PUT against lights/11/state with 
`{ "alert": "none" }`
or
`{ "alert": "blink" }`
is successful and does result in the expected behaviour. I do this via a Chrome extension without problems.

SwoopX, 2020-07-08:
So, then your gave yourself already the answer: the issue is with NR 😉

sonofsilberling, 2020-07-08:
Thank you for that. I'll follow up with Node Red.

SwoopX, 2020-07-08:
Mind closing this one then?

closed 2020-07-08 by sonofsilberling

sonofsilberling, 2020-07-09:
For anyone else to come across the same issue: I now bypass Node Red's deconz implementation and issue a rest PUT statement via node red's HTTP request node, which works fine.