dresden-elektronik/deconz-rest-plugin #3004: HEIMAN RC-EM

cllatser, 2020-07-01:
## Device

- Product name: HEIMAN RC-EM
- Manufacturer: ETIGER
- Model identifier: RC-EM
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [X] Other: Remote Alarm

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-07-07:
You are aware that this no regular remote but for steering security gateways? And it will especially not work if deconz or your middleware goes down.

cllatser, 2020-07-08:
Yes I know, I have a small flat. I don't need high security. I started recently and now I have everything connected to the Conbee II. I bought this because the seller's website sold it as the Heiman RS1RC. I did not want to put an RF bridge. Thank you very much.

https://es.aliexpress.com/item/32975159931.html?spm=a2g0s.9042311.0.0.501e63c0274yWo

Note: I have all the sensors connected directly to the Conbee II and the Conbee II to the Rpi4 and the Rpi4 and router lte to the UPS, so I would just go without an alarm if the Rsi4 failed :)

SwoopX, 2020-07-08:
What do you plan to do with it anyway?

cllatser, 2020-07-08:
Simply that each button has an event. And then I in Node-red will assign them to the control panel.

SwoopX, 2020-07-08:
Alright, just wanted to set expectations correctly. Hope it works the same way as the other remote...

cllatser, 2020-07-08:
You will make me happy. Thank you very much.

SwoopX, 2020-07-08:
For the remote to properly join, the coordinator has to be extended by "IAS ACE" cluster manually in deconz GUI. For that, go to _Edit -> Network settings -> Endpoints_ (or press F9 -> Endpoints) and add `0x0501` to the `in clusters` of endpoint 0x01.

grafik

After successful join, the device should send IAS ACE commands on all its buttons as below:

**Top button**: Arm Mode - Arm Day/Home Zones Only -> translates to 1001
**Right button**: Arm Mode - Disarm -> translates to 2001
**Bottom button**: Command - Emergency -> translates to 3001
**Left button**: Arm Mode: Arm All -Zones -> translates to 4001

cllatser, 2020-07-08:
Hello, I was away from home. I already tried before opening this incident the cluster 0x0501 and it did not send me any event. Do I have to wait for an update?

SwoopX, 2020-07-08:
yes, what else?

cllatser, 2020-07-08:
Excuse my English. I have already put the 0x0501 in "in clusters". I have paired the controller with Phoscon as a switch. But I see neither in HA nor in Node-Red any control event. Do I have to wait for the next version of deconz or would it already have to work? Thank you very much.
[image]

SwoopX, 2020-07-08:
I have updated the codebase to handle it in future. That has not yet been merged and a new version hasn't been released. The comment I added is just a set of prerequisites and explains what to expect.

cllatser, 2020-07-08:
Thank you very much. :)

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

Xam54580, 2020-11-20:
> Excuse mon anglais. J'ai déjà mis le 0x0501 dans "en clusters". J'ai couplé le contrôleur avec Phoscon en tant que commutateur. Mais je ne vois ni dans HA ni dans Node-Red aucun événement de contrôle. Dois-je attendre la prochaine version de deconz ou est-ce que cela devrait déjà fonctionner? Merci beaucoup.
> [image]

Hello did you manage to make it work? I have the same problem