dresden-elektronik/deconz-rest-plugin #3007: Issue with left and right buttons on IKEA TRÅDFRI remote control

maxnumberone, 2020-07-02:
Hello,
I have an issue with an IKEA TRÅDFRI remote control.
The left and right buttons are having an assignment issue in the Phoscon App:
[image]
When I select the button to change the color temperature of lights, it is assigning the action to a long press.
Unfortunately there's no possible long press for these 2 buttons on my remote, when you keep one of them pressed it keeps repeating short press/clicks, no long press.
Is it possible this issue comes from the firmware installed on my remote, I have 2.3.014, maybe it was updated by Ikea in a more recent release?
Thank you for your help

morfei1, 2020-07-02:
Hello! I think you need to post this issue in - https://github.com/dresden-elektronik/phoscon-app-beta

Mimiix, 2020-07-02:
Hi @maxnumberone !

As @morfei1 suggested, this belongs in the Phoscon Repository. I will move it :)