dresden-elektronik/deconz-rest-plugin #3008: Heiman TS003 rellay/switch | Functionality issues

CarpeDiemRo, 2020-07-02:
## Describe the bug

I have two relays with serial 
QS Zigbee S04 2C LN  2 ways [https://imgur.com/Ipnchwd](url)
QS Zigbee S05 LN with 1 way [https://imgur.com/kBCrK1a](url)

They are displayed in Deconz interface as Heiman TS003 [https://imgur.com/qmfvyS6](url)

Even though the initialization in Deconz was straight forward and they were correctly displayed in the interface, they did not work as it should. Meaning that Switch ON and OFF action is not working or, very rarely is working with a huge delay. 

I am using Home Assistant integration and I tried to control them from the interface. The toggle goes to Switch ON state and then, immediately goes back.

From the Deconz interface, the same situation. 

## Steps to reproduce the behavior
1. Pair the switches with Deconz
2. Try to Switch ON / OFF

## Expected behavior
The switch must be able to be controlled by the Deconz

## Screenshots
**One Way switch in VNC**
[https://imgur.com/37WrslV](url)
[https://imgur.com/TmEI08e](url)
[https://imgur.com/JoO5UQh](url)

Two ways switch in VNC
[https://imgur.com/cYWxxpP](url)
[https://imgur.com/AnAa2zF](url)
[https://imgur.com/UmeKFZj](url)

Both switches in the Zigbee context
[https://imgur.com/MoOpzNI](url)

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
Raspbian / Home Assistent deCONZ Add-on
No USB extension cable
Firmware version: (264A0700)
deCONZ version: (2.05.78)
### deCONZ Logs
12:50:49:790 binding for cluster 0x0006 of 0x680AE2FFFE096E78 exists (verified by reporting)
12:50:49:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x680AE2FFFE096E78 (seems to be active)
12:50:49:790 binding for cluster 0x0006 of 0x680AE2FFFE096E78 exists (verified by reporting)
12:50:49:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x680AE2FFFE096E78 (seems to be active)
12:50:50:854 don't close database yet, keep open for 900 seconds
12:50:51:790 binding for cluster 0x0006 of 0xEC1BBDFFFE9F7584 exists (verified by reporting)
12:50:51:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xEC1BBDFFFE9F7584 (seems to be active)
12:50:51:790 binding for cluster 0x0006 of 0xEC1BBDFFFE9F7584 exists (verified by reporting)
12:50:51:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xEC1BBDFFFE9F7584 (seems to be active)
12:50:55:964 don't close database yet, keep open for 900 seconds
12:50:57:790 binding for cluster 0x0006 of 0x680AE2FFFE096E78 exists (verified by reporting)
12:50:57:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x680AE2FFFE096E78 (seems to be active)
12:50:57:790 binding for cluster 0x0006 of 0x680AE2FFFE096E78 exists (verified by reporting)
12:50:57:790 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x680AE2FFFE096E78 (seems to be active)

12:52:15:429 	0x00124B001E733081 force poll (2)
12:52:15:434 ZCL attribute report 0x00124B001E733081 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

## Additional context
If I press very very quick on Light toggle in Home assistant, some action are taken into account around 20 - 30% from switch ON/OFF

12:57:19:582 	0xEC1BBDFFFE9F7584 force poll (2)
12:57:19:586 ZCL attribute report 0xEC1BBDFFFE9F7584 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:57:19:762 	0xEC1BBDFFFE9F7584 force poll (2)
12:57:19:766 ZCL attribute report 0xEC1BBDFFFE9F7584 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:57:19:959 	0xEC1BBDFFFE9F7584 force poll (2)
12:57:19:974 ZCL attribute report 0xEC1BBDFFFE9F7584 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
12:57:20:147 	0xEC1BBDFFFE9F7584 force poll (2)

ebaauw, 2020-07-02:
Two server _On/Off_ clusters on the same endpoint?!  That’s a first.  I doubt that’s in line with the Zigbee standard and I’m pretty sure neither the deCONZ core nor the REST API plugin can handle this.

Does the device really advertise the cluster twice oneachbendpoint?  Can you sniff the _Simple descriptor_ responses to confirm?

Can you check in the GUI to which cluster the device responds, when sending an _On_ or _Off_ command?  Can you read attributes on each cluster?  Does the _OnOff_ attribute reflect the current device state?

CarpeDiemRo, 2020-07-02:
Let take it one by one
> Two server _On/Off_ clusters on the same endpoint?! That’s a first. I doubt that’s in line with the Zigbee standard and I’m pretty sure neither the deCONZ core nor the REST API plugin can handle this.

I supposed you are meaning to 2 ways switch. I deleted from deConz and I re-added again successfully. I received the following attributes 
[image]
[image]

> Does the device really advertise the cluster twice oneachbendpoint? Can you sniff the _Simple descriptor_ responses to confirm?

I really don't know how to sniff thee Zigbee Network. It is possible from deConz UI?

> Can you check in the GUI to which cluster the device responds, when sending an _On_ or _Off_ command? Can you read attributes on each cluster? Does the _OnOff_ attribute reflect the current device state?
So, the behavior is:
1. I toggled the switch from the first circuit of 2 ways switch. A blue dot is blinking on the corresponding entity 
2. I toggled the second switch of the two ways switch. A blue dot is blinking as well. 
3. A blue dot is blinking on the consequent devices that are linked to the entity.
I supposed this worked as designed. However, no command is sent to the physical relay and the switch is pushed back to the initial state. 
I tested in parallel with deConz UI. When I pressed the toggle button from the Home Assistant, the bulb symbol color becomes yellow for a moment and go back to grey. 
You can check this movie https://youtu.be/2NtvlMnHNxM
4. The same situation with one way switch.

ebaauw, 2020-07-02:
> I supposed you are meaning to 2 ways switch.

Actually both switches.  The 1 way shows a single endpoint (as expected), but with two times the 0x0006 cluster; the 2-way shows two endpoints (as expected), but again with two 0x0006 clusters.

Did you try each of the two clusters on the 1 way switch, and all four clusters on the 2 way switch?

CarpeDiemRo, 2020-07-02:
I tried to removed each one and to work with them separately. The same result as I described. It is the first time when I saw this situation.

ebaauw, 2020-07-02:
> It is the first time when I saw this situation.

Same here.  If they don’t work with the GUI, I don’t think there’s a way to make them work with the REST API plugin.

CarpeDiemRo, 2020-07-02:
I will try to integrate with zigbee2mqtt in order to see if they are broken or in a deConz bug. I will come back with more details

CarpeDiemRo, 2020-07-02:
I tested with zigbee2mqtt stick and it is working perfectly

zigbee2mqtt:info  2020-07-02 18:58:55: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe9f7584', payload '{"state_left":"ON","linkquality":44}'
zigbee2mqtt:info  2020-07-02 18:58:55: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe9f7584', payload '{"state_left":"ON","linkquality":42}'
zigbee2mqtt:info  2020-07-02 18:58:56: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe9f7584', payload '{"state_left":"OFF","linkquality":42}'
zigbee2mqtt:info  2020-07-02 18:58:56: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe9f7584', payload '{"state_left":"OFF","linkquality":44}'

CarpeDiemRo, 2020-07-03:
From my perspective it is a deConz issue.

Smanar, 2020-07-03:
>I am using Home Assistant integration and I tried to control them from the interface. The toggle goes to Switch ON state and then, immediately goes back.

Are you listening the relay ? because on your picture you are not using output, so it can be just a flase/missing state return ?
And for exemple I know a device that dont work without bulb connected at output.

CarpeDiemRo, 2020-07-03:
@Smanar Not the case, because with Zigbee2Mqtt are working perfectly without any bulb attached.

Smanar, 2020-07-03:
But you are using bulb with Zigbee2mqtt ?

CarpeDiemRo, 2020-07-03:
As I mention, I tried with the same relays in the same setup to integrate it with zigbee2mqtt hub and they are working perfectly. Are you familiar with zigbee2mqtt?

Smanar, 2020-07-03:
>zigbee2mqtt:info 2020-07-02 18:58:55: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe9f7584', payload '{"state_left":"ON","linkquality":44}'

If you want, I can send the same message but without changing the output relay ?

CarpeDiemRo, 2020-07-03:
@Smanar Of course I have listen the relay. No sound. For your clarity, please have a look at the attached movies from the bug description.

CarpeDiemRo, 2020-07-06:
Do you know to who can I address this issue?

SwoopX, 2020-07-07:
When you can compile the rest plugin yourself (as per the readme instructions), I can provide you with some code which outputs the communication responses.

CarpeDiemRo, 2020-07-16:
@SwoopX I don't know how to do this. Can you give me, please, some guidelines.

For me it is very strange the fact that Conbee recognize the device, but nevertheless it is not functional.

SwoopX, 2020-08-06:
Missed that one. The compile instructions are mentioned in the readme of this repo. I can preper a special version for compilation then but it might be a good idea to try it anyway with the currently available code here.

SwoopX, 2020-08-14:
The upcoming release gets rid of duplicate cluster entries. Can you please double check again and report back, once it is available to you?

ArniPL, 2020-08-29:
I have the exact same issues with the exact same switch.

Note, that setting the on/off state via the cluster commands (in deCONZ GUI) works perfectly without issues. But setting the state via the REST API, still doesn't work.

Is there some way that we could execute the cluster command directly with the API? That would be a usable workaround before the device is fixed in deCONZ.

I can't check if the new 2.05.80 release fixes this, because I am using deCONZ add-on for Home Assistant, which is currently on the 2.05.79 version with [no easy way](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-docker) to force core deCONZ upgrade.

@CarpeDiemRo Looking for the workaround, I found the [deconz-cli-plugin](https://github.com/ma-ca/deconz-cli-plugin), where it seems easy to execute cluster commands, but the setup is not that straightforward.

Mimiix, 2020-08-29:
@ArniPL As far as i am concerned you can't do that with the API.

However, you are trying to fix a issue that is fixed in a newer version. We do not maintain the addon, so i can't do anything on this. You could test it on a stand alone instalation. 

@CarpeDiemRo : Is the issue fixed (as @SwoopX requested?)

ArniPL, 2020-08-31:
@Mimiix Thank you for the API clarification.

As for the issue at hand, I got the impression from what @SwoopX said, that we don't know if the deduplication feature in new version fixes this. We'll need to wait until HA addon gets updated, or as you said - test that on a standalone installation.

SwoopX, 2020-08-31:
@ArniPL I can already give you the HA answer: there is no intention to move to .80 since the feature to upgrade the firmware via Phoscon has been temporary disabled.

ArniPL, 2020-08-31:
@SwoopX Ah, yes. I see that now in the changelog. Thank you. Alright then - gonna wait for a release that's ready for HA.

ArniPL, 2020-09-27:
I can confirm that this is now resolved in [deCONZ 2.05.81](https://github.com/home-assistant/hassio-addons/pull/1573).

Mimiix, 2020-09-28:
As @ArniPL  mentioned, it seems to be fixed! Thanks all for cooperation :)

closed 2020-09-28 by Mimiix

cpuks, 2020-10-14:
Yes, 1 gang and 2 gang no neutral needed. I've got them both - 1 gang works and 2 gang seems to be broken, just ordered one and over weekend I'll try to gather intel :-) tomorrow 15th so should be new deconz release as well so I'll install GUI too prior my investigation.
1 gang seems to works OK as it was deceted as Hineman light switchable - same as 1 gang with neutral.

[image]

Smanar, 2020-10-14:
Oups, right, I was stupid, thoses device are light, so no need to add them in whitelist, they are working natively.