dresden-elektronik/deconz-rest-plugin #3009: New device EcoBright 8 button wall remote

KekieM, 2020-07-02:
[image]

- Product name:  ED-10015
- Manufacturer: EcoDiM
- Model identifier: ED-10015
- Device type:
  - [ ] Light
  - [ ] Lock
  - [X ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

https://www.ecobright.nl/z-wave-wandschakelaar/z-wave-wandschakelaar-draadloos-zwart-4-zones/

## Screenshots

[image]
[image]
[image]
[image]
[image]

Sjonnie2018, 2020-07-05:
You're linking to the Z-wave version of the switch. For those interested this is the link to the ZigBee version: https://www.ecobright.nl/zigbee-wandschakelaar/zigbee-wandschakelaar-draadloos-zwart-4-zones/

SwoopX, 2020-07-06:
Can you please read the attributes of the power configuration cluster and then provide an updated screenshot? Thanks!

KekieM, 2020-07-06:
Power configuration 1 (does not seem to change with read)
[image]

Power configuration 2 ( updated after read)
[image]

Power configuration 3 (no change after read)
[image]

Power configuration 4 (no change after read)
[image]

SwoopX, 2020-07-19:
Unfortunately, beta .79 contains an incorrect button map. A confirmed fix will be available with the following version.

KekieM, 2020-07-27:
> Unfortunately, beta .79 contains an incorrect button map. A confirmed fix will be available with the following version.

is that the reason only the top row of buttons seems to work?

SwoopX, 2020-07-28:
Yes, ideed. I ordered 2 of them but unfortunately, they arrived after our release deadline so I couldn't verify.

wvuyk, 2020-08-29:
awaiting merge triggers the awful bot?

SwoopX, 2020-08-30:
> awaiting merge triggers the awful bot?
> Could need some updates there...

No, it doesn't. And why don't you consider checking release notes? It's a valuable source to understand, what has happened.

wvuyk, 2020-08-30:
I did a search for the EcoDim 8 button wall remote regularly but guess I missed it then? Sorry for stepping on your toes here, as that is not what I wanted. 
I was monitoring this thread because I would like to use the switch as well. This thread did not show the switch was implemented,  so I assumed it was not finished yet?

SwoopX, 2020-08-30:
Np, guess you really overlooked it. Check out my recent commit for all the details (the release notes also mention it): https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3073

However, please take note that the device now exposes only one `ZHASwitch` resource instead of 4. I'd recommend resetting and re-pairing it.

I'm closing this issue now since the device is fully supported with version 2.05.80.

closed 2020-08-30 by SwoopX

Mimiix, 2020-08-31:
I recognize this one. You got me to ship it right? @SwoopX ?

SwoopX, 2020-08-31:
Guilty 😜

ronaldt80, 2020-10-07:
Perhaps a silly question but how do I use this switch in the Phoscon App? Is that possible? I have paired it and see it as available in my devices (switches) but can not seem to link the switch to a light/group.  Other switches (eg Aqara) work smoothly. I use a Conbee II, 2.05.80 / 8/14/2020 with FW v.26580700

SwoopX, 2020-10-07:
Not sure since I don't use Phoscon. However, the device groups should be visible and usable in the old webapp.

ronaldt80, 2020-10-10:
Thanks @SwoopX - that indeed works! Even after linking the devices in the old webapp it doesn't show up in Phoscon though.

SwoopX, 2020-10-10:
Glad that it works. You might want to raise the visibility topic in the Phoscon beta repo.