dresden-elektronik/deconz-rest-plugin #3011: WS-EUB1-ZG

rosti76, 2020-07-03:
## Device

- Product name: Zigbee Smart Switch
- Manufacturer: _TZE200_7tdtqgwv
- Model identifier: TS0601
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ x] Other: Mains powered Wall Swith

## Screenshots
[image]
[image]

Smanar, 2020-07-03:
for information , another device with tuya manufacture specific cluster EF00 > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2879

Smanar, 2020-07-13:
Ok in fact it s not another device, but a duplicate ^^

Can you go on the other issue and try the code proposed, pls ?

closed 2020-08-14 by stale[bot]