dresden-elektronik/deconz-rest-plugin #3012: Device "eth0" does not exist.

ebreg, 2020-07-03:
## Device "eth0" does not exist.
My Raspberry Pi Zero is connected to the local network via Wifi. Since installing deconz the syslog is flooded with the above message.

> Jul  3 15:59:54 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:01:22 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:02:50 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:04:19 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:05:46 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:07:13 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:08:41 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.
> Jul  3 16:10:10 homebridge deCONZ-WIFI2.sh[325]: Device "eth0" does not exist.

The messages can be disabled by `sudo systemctl disable deconz-wifi.service`, but that only helps until the next reboot.

## Expected behavior
Messages are turned off.

## Environment
**Host system:** Raspberry Pi Zero WH
**Running method:** Raspbian (Homebridge Raspberry Pi Image)
**Firmware version:** 2610700
**deCONZ version:** 2.05.77
**Device:** RaspBee II

## Additional context
This issue seems to be related to #1738.

SwoopX, 2020-07-06:
You can ultimately mute the service with `sudo systemctl mask deconz-wifi`. However, the messages should be gone with the next release. Requested the necessary changes...

ebreg, 2020-07-06:
Perfect, thanks.

SwoopX, 2020-07-28:
Bump, still not fixed.

closed 2020-09-11 by stale[bot]