dresden-elektronik/deconz-rest-plugin #3014: Deconz websocket stopped sending config objects for Ikea On/Off switches

Digital-Thor, 2020-07-03:
I've been enjoying working with conbee/deconz for about six months.  Thanks for all the hard work maintaining and improving them.  Below is a problem that I keep encountering.

## No config objects sent by websocket
I often encounter a problem after rebooting the Raspberry Pi configured with a Conbee II.  The problem also seems to occur if I leave the Pi off overnight.  After the Pi powers-up, the deconz websocket no longer emits config objects from Ikea Tradfri On/Off switches, though state objects are emitted. 

## Steps to reproduce the behavior

  1. Pair IKEA Tradfri On/Off switches
  2. Reboot Pi 3b+ with Conbee 2 or leave the Pi off overnight.
  3. Once Pi is powered up again, Deconz does send a state object when On/Off switch (button) is pressed, but no more config objects.
  4. Pair a new On/Off switch, and config objects come from it, but not from previous devices (that were paired in step 1).

## Possible Clue
I suspect that the problem may have something to do with the fact that the On/Off switches (from step 1) are now connected directly to the light bulb and not the controller.

[image]

## Environment
 - Host systems: Raspberry Pi 3b+.  Also had same problem on Pi 4b
 - Running method: Raspbian
 - Firmware version: 264A0700
 - deCONZ version: 2.05.77
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## Additional context
 - I have two Conbee Devices on my network.  Each uses a different websocket address and Zigbee channel.
 - One network has about 20 devices (sensors and switches).  
 - The log below is for the smaller network shown in the screen shot above.  There is another powered-down light which is not shown.

### deCONZ Logs
```
pi@rpad3bp:~$ sudo systemctl status deconz

● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz.service.d
           └─override.conf
   Active: active (running) since Fri 2020-07-03 11:52:10 PDT; 26s ago
 Main PID: 13641 (deCONZ)
    Tasks: 6 (limit: 2077)
   Memory: 9.9M
   CGroup: /system.slice/deconz.service
           └─13641 /usr/bin/deCONZ -platform minimal --http-port=80 --ws-port=443 --dbg-info=2 --dbg-error=1 --auto-connect=1 --dbg-zcl=1 --dbg-zdp=1 --dbg-aps=1

Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:11:751 New websocket 10.0.1.40:37936 (state: 3)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:12:194 UPNP socket not bound, state: 0
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:13:701 New websocket 10.0.1.40:37960 (state: 3)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:943 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:943 COM: /dev/ttyS0 :  (0x0000/0x0000)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:943 dev /dev/ttyS0 (/dev/ttyS0)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:943 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:943 GW firmware update select USB device
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:15:944 GW update firmware not found:
Jul 03 11:52:17 rpad3bp deCONZ[13641]: 18:52:17:408 Announced to internet http://dresden-light.appspot.com/discover

pi@rpad3bp:~$ journalctl -au deconz

-- Logs begin at Fri 2020-07-03 10:39:50 PDT, end at Fri 2020-07-03 12:11:43 PDT. --
Jul 03 11:52:10 rpad3bp systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jul 03 11:52:10 rpad3bp deCONZ[13641]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Jul 03 11:52:10 rpad3bp deCONZ[13641]: libpng warning: iCCP: known incorrect sRGB profile
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:519 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:534 CTRL. 3.27.211:52:10:747 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:748 COM: /dev/ttyS0 :  (0x0000/0x0000)
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:748 dev /dev/ttyS0 (/dev/ttyS0)
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:748 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:10:748 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:11:107 gw run mode: systemd/headless
Jul 03 11:52:11 rpad3bp deCONZ[13641]: 11:52:11:107 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
[... 413 lines cut ...]
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:174 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedSt
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:174 DB sql exec UPDATE devices SET nwk = 0 WHERE mac = '00:21:2e:ff:ff:05:41:be';INSERT INTO devices (mac,nwk,timestamp) SELECT '00:21:2e:ff
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:189 DB saved in 16 ms
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:190 ~Resource() /lights 0x7ef0a050
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:190 DB pushZdpDescriptorDb()
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:191 sql exec SELECT * FROM nodes WHERE mac='d0:cf:5e:ff:fe:ca:76:e6-01' COLLATE NOCASE AND state != 'deleted'
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:192 don't close database yet, keep open for 900 seconds
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:193 LightNode 3: dgp_3_61_1 added
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:193 ~Resource() /lights 0x7ef0a050
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:193 DB pushZdpDescriptorDb()
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:193 DB save zll database items 0x00000800
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:193 DB sql exec UPDATE devices SET nwk = 10059 WHERE mac = 'd0:cf:5e:ff:fe:ca:76:e6';INSERT INTO devices (mac,nwk,timestamp) SELECT 'd0:cf:5
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:195 DB saved in 1 ms
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:195 DB pushZdpDescriptorDb()
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:201 ~Resource() /lights 0x7ef0a050
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:201 SensorNode 2 set node 0xccccccfffe5442dd
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:206 DB pushZdpDescriptorDb()
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:206 DB save zll database items 0x00000800
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:206 DB sql exec UPDATE devices SET nwk = 34290 WHERE mac = 'cc:cc:cc:ff:fe:54:42:dd';INSERT INTO devices (mac,nwk,timestamp) SELECT 'cc:cc:c
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:207 DB saved in 0 ms
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:208 ~Resource() /lights 0x7ef0a050
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:210 DB pushZdpDescriptorDb()
Jul 03 11:52:41 rpad3bp deCONZ[13641]: 18:52:41:211 sql exec SELECT * FROM nodes WHERE mac='d0:cf:5e:ff:fe:fe:a2:ad-01' COLLATE NOCASE AND state != 'deleted'

```

ebaauw, 2020-07-03:
> I suspect that the problem may have something to do with the fact that the On/Off switches (from step 1) are now connected directly to the light bulb and not the controller.

I don’t think so.

The switch sends groupcasts when pressing a button, which the REST API plugin translates to `buttonevent` values, producing the `state` notifications.  I think, after pairing and discovering the group, the only message leading to a `config` notification would be the attribute report that the sensor sends to the coordinator.  This is a unicast, for which the switch would expect an ACK message.  Most likely, it removed the binding and/or reporting configuration since it didn’t receive any for an extended period, because the coordinator was powered down.

I think the API plugin should eventually re-create the binding and reportIng config, but you might also do so manually from the GUI.

Digital-Thor, 2020-07-04:
Thanks @ebaauw for explaining the protocol.  The switches have been running for days without binding again.  So I tried to do a binding in `deconz-gui` from the the latest switch to one of the initially-paired switches, but binding timed-out.  Specifically, I tried binding the `Poll Control` clusters, then the `Groups` clusters, and then the `Basic` clusters, but got a timeout for each of those cases. 
[image]

I'm not that familiar with binding, but wondering if I could do this using the API instead since my App is likely to encounter this problem frequently.  Do you know if binding configures the `devicemembership` array in `/api/<apikey>/groups`, or is that a different relationship?

ebaauw, 2020-07-04:
You need to bind the _Power configuration_ cluster to the first endpoint on the coordinator.  Make sure to wake the switch before and while pressing bind.

The client clusters are bound to a group alright, or you wouldn’t see any `buttonevent` notofications.  The REST API plugin eavesdrops on the commands that the switch sends to the group, populating `config.group` and creating the `/groups` resource with the `devicemembership`.

Digital-Thor, 2020-07-04:
Thanks @ebaauw.  I was able to get binding to work using that approach in the **deconz-gui** and `config` notifications are once again emitting from the websocket.  However, I couldn't see any associated changes when getting from the `/sensors` or `/groups` endpoints until I explicitly set the group value within the Binding Dropbox as shown here:

[image]

Since my App will need to rebuild this relationship automatically after power-up, it will need to use the deconz API.  I tried `PUT  /groups/groupID` with `body.devicemembership` containing an array of sensor IDs like this:

```
curl -i -X PUT -H 'Content-Type: application/json' -d '{"devicemembership":["1","2","3","4"]}' http://10.0.1.40/api/authkey/groups/10

HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Type: text/html; charset=utf-8
Content-Length:0
ETag:"6e45171922e573cf4c431af9bb551aa0"
```
Although it returns a 200, there's no response body showing a change to `devicemembership`.  Getting from `/groups` confirms no change.  I guess I'm not that surprised since `devicemembership` is not included in the [documentation here](https://dresden-elektronik.github.io/deconz-rest-doc/groups/#setattr). However, there must be a way that Phoscon pulls it off. 
 Thoughts?

ebaauw, 2020-07-05:
Afaik, you cannot update `devicemembership` in a `/groups` resource, nor can you update `config.group` in a `/sensors` resource.  Both are updated by the API plugin, as it sees group commands from the controller.  To which group these commands are sent is determined by the binding of the corresponding client cluster, usually _On/Off_ and _Level Control_, sometimes also _Scenes_ and _Color Control_.  I don’t think the client _Groups_ cluster does anything here, but some devices have a weird implementation of the binding table, where they don’t seem to look at the source cluster and apply the binding to all clusters.

I don’t use Phoscon, but I think it creates rules instead of using the group bindings.  There is some obscure code in the API plugin to create binding using rules, but I never understood that part, and haven’t been able to use it.

closed 2020-08-02 by stale[bot]