dresden-elektronik/deconz-rest-plugin #3015: Firmware update fails

JMBokhorst, 2020-07-04:
Hi, 

I bought a new conbee II for my home assistant (been using the first conbee for a long time). After I plugged it in, I wanted to do a firmware update, however this failed (see error below). After the error occurred, the usb isn't recognised anymore. Do you have any idea how I can recover my conbee? 

Thanks in advance. 

```
Device Path : /dev/ttyACM0
 
-------------------------------------------------------------------
 
Firmware available for flashing:
 
deCONZ_ConBeeII_0x264a0700.bin.GCF
deCONZ_ConBeeII_0x26580700.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF
deCONZ_Rpi_0x26350500.bin.GCF
 
Enter the firmware file name from above, including extension,
or press Enter now to exit.
 
File Name : deCONZ_ConBeeII_0x26580700.bin.GCF
 
-------------------------------------------------------------------
 
Device: /dev/ttyACM0
 
Firmware File: deCONZ_ConBeeII_0x26580700.bin.GCF
 
Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y
 
Flashing...
 
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 264A0700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 160930 bytes: |====error: timeout flashing firmware after 3006 ms
 
Flashing Error! Please re-run this script...
```

Running on ubuntu 18.04 with docker. 

I have tried to recover the bricked USB device on a rasp-pi following the manual upgrade steps. However, it seems the firmware is been corrupted completely as the device is not able to connect correctly: 

```
[2227002.233815] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[2227005.687735] usb 1-2: USB disconnect, device number 46
[2227014.075799] usb 1-2: new full-speed USB device number 47 using xhci_hcd
[2227014.225990] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030
[2227014.225996] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[2227014.226000] usb 1-2: Product: ConBee II
[2227014.226003] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[2227014.226006] usb 1-2: SerialNumber: DE2215531
[2227014.226726] cdc_acm 1-2:1.0: ttyACM0: USB ACM device

```

Then I tried to connect to another machine that runs ubuntu 18.04, but I got the same results. 

```
[67811.446199] cdc_acm 3-2:1.0: failed to set dtr/rts
[67820.010355] usb 3-2: new full-speed USB device number 103 using ohci-pci
[67820.217126] usb 3-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[67820.217133] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[67820.217136] usb 3-2: Product: ConBee II
[67820.217139] usb 3-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[67820.217142] usb 3-2: SerialNumber: DE2215531
[67820.219366] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
[67823.765272] usb 3-2: USB disconnect, device number 103
[67823.766192] cdc_acm 3-2:1.0: failed to set dtr/rts
[67832.334197] usb 3-2: new full-speed USB device number 104 using ohci-pci
[67832.541141] usb 3-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[67832.541147] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[67832.541150] usb 3-2: Product: ConBee II
[67832.541153] usb 3-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[67832.541156] usb 3-2: SerialNumber: DE2215531
[67832.543337] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
[67836.071991] usb 3-2: USB disconnect, device number 104
```
Is there a way to recover the stick?

Smanar, 2020-07-04:
Hello, you have a procedure here to make firmare update with longuer timeout using "-t 60" option.

JMBokhorst, 2020-07-04:
Hello,

Yes, I have tried that. The first upgrade I did was via the upgrade steps in the docker. This resulted in the first output. Afterwards, it seems to be unable to get the device to be detected correctly. So I haven't been able to have a second go at the upgrade. 

Do you have any idea how to get the devices detected first, so I can flash it again?

Smanar, 2020-07-04:
Not with docker, with a real OS. Try with windows for exemple.

> The firmware update button doesn't work properly in the HA addon or docker installations. Nor does it work fine in Virtual machines. On this page you can find out how to update. Please use a native installation to update.

Mimiix, 2020-07-04:
You can't update the firmware in HA Addon or Docker version.

JMBokhorst, 2020-07-04:
The initial upgrade was via Docker, afterwards I tried on ubuntu 18.04 directly and on the latest version of rasbian. I will try again with windows and see if that works.

JMBokhorst, 2020-07-04:
Ok, I have tried via windows and that seems to be working :) Thanks! 

So in summary: 
- I first tried to update the software via docker following [these](https://github.com/marthoc/docker-deconz) update steps. This failed and gave me the time out error.
- I tried to update the software on the host machine running ubuntu 18.04, however ubuntu wasn't able to recognise the USB anymore getting stuck in an infinite (re-)connect/disconnect loop. 
- I tried on another machine running ubuntu 18.04 and on a raspberry pi with the same connection issues.
- I tried via windows, that seems to be working ok and I was able to flash the USB. 
- No further issue's with ubuntu as well :)

closed 2020-07-04 by JMBokhorst