dresden-elektronik/deconz-rest-plugin #3017: How to reset ?

Galiwyx, 2020-07-06:
Hello,

I'm trying using deCONZ with homebridge on docker. I use unRaid as OS. 

First I checked if conbee II stick was discovered : 
[image]

Second, I go to https://phoscon.de/pwa to add the new Conbee USB stick
[image]

Buty stick is already here, I don't understand why, it's a clean installation of deCONZ. How can I remove the previous installation ? Of course I can't login (connection failed).

Galiwyx, 2020-07-06:
The problem was the url, it's not  https://phoscon.de/pwa but  https://phoscon.de/app

Mimiix, 2020-07-07:
So it is fixed :)?

Galiwyx, 2020-07-07:
Yes :)

closed 2020-07-07 by Mimiix