dresden-elektronik/deconz-rest-plugin #3020: Climax EMD-1ZBS power meter

Ocramius, 2020-07-07:
Hey there! First of all, thanks for all the hard work around deconz: looks awesome so far!

I'm here to ask for device support as per https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

## Device

This is a Climax EMD-1ZBS (https://zigbeealliance.org/zigbee_products/energy-meter-emd-1zbs/ & https://www.climax.com.tw/new/downloads/EMD-1ZBS%20Energy%20Meter_Product%20News_11.11.2016.pdf):
 

[image]

Not much available online about how it operates. Retailer is Lupus - they sell it re-branded as "Electric Meter Smart Home Module". It uses a diode to detect the blinking light on a power meter, and infers KWh metrics from that.

- Product name: EMD-1ZBS
- Manufacturer: Climax
- Model identifier: **no model identifier displayed**
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [x] Other: power meter

## Screenshots

Cluster screenshot: 

[image]

[image]

[image]

Lemme know if I can help in any way to get this one detected: couldn't find much about it in other gateways.

SwoopX, 2020-07-07:
You forgot the screenshot for the simple metering cluster.

Ocramius, 2020-07-08:
Thanks @SwoopX: I've added the screenshots of all cluster info sub-sections (didn't know you could select them :D )

SwoopX, 2020-07-08:
Thanks, but it looks like you haven't read the attributes via the button...

Ocramius, 2020-07-08:
This is after pressing "read": will retry though 👍

SwoopX, 2020-07-08:
The device was probably not awake.

Ocramius, 2020-07-08:
I've re-done the reads: since I don't really understand how the GUI works, I've selected each field individually and used the "read" button each time.

Here are the results:

### 0000 Basic

[image]

### 0001 Power Configuration

[image]

### 0003 Identify

[image]

### 0702 Simple Metering

[image]

Ocramius, 2020-07-08:
> The device was probably not awake.

Hmm, I've forced it to try "pairing" (not sure if that's what it's called in the ZigBee world). How would that be forced on other appliances, such as other meters? Trying to query through `Identify` GUI seems to only produce timeouts.

SwoopX, 2020-07-08:
Now it looks ok-ish. But the values are all 0, probably due to a reset?

Got no clue what the measurement units are. For now, I'd assume it's all in W.

Ocramius, 2020-07-08:
> probably due to a reset?

Hard to say: this thing came with zero manual, besides "press and hold for 10 seconds to start looking for a gateway".
I can most certainly act as guinea pig to get this polished over time.

> For now, I'd assume it's all in W.

Very likely

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

Ocramius, 2020-07-19:
Amazing, will try reading through the diffs and give it a spin soon 👍

00lex, 2020-07-21:
its frustrating to get or just find a shop for some zigbee devices :(
@Ocramius can you give us a link? 

thank you

Ocramius, 2020-07-21:
@00lex here you go: https://www.lupus-electronics.de/shop/Alarm-Smarthome/Smarthome-Hausautomation/LUPUSEC-Hauptstromz%E4hler-p.html

Indeed, I had to contact Climax too in order to find a local reseller :shrug:

spripe, 2020-10-03:
Any solution for this?
My meter is also reporting 0 kWh

spripe, 2020-10-03:
And would be nice if it could report watt and not kWh :(

Ocramius, 2020-10-03:
@spripe I suggest opening a new issue: my device now connects and shows up (as a siren) within Phoscon, but I didn't have time to debug further into metrics: doesn't seem to pick up anything meaningful, for now.

SwoopX, 2020-10-03:
> And would be nice if it could report watt and not kWh :(

Not gonna happen. Firstly, what you get is dependent on the device (99% report any values smaller than kWh) and secondly, the values exposed by the REST API are integers, no doubles.

If you guys don't get any values in the simple metering cluster via deconz GUI, then there's not much we can do. In that case I'd suspect it to not collect any usable data.

luka6000, 2021-03-10:
I'm very interested in this meter. Found some docs
https://fccid.io/GX9EMDZB/User-Manual/Users-Manual-3242046.pdf
Implementation looks ok. Looking at the screenshots I can see that Basic-Device enabled attribute is set to false. Maybe that's why no values are returned?

Ocramius, 2021-03-10:
Attaching it here, since I don't trust the external URI to stay stable :D 

[Users-Manual-3242046.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6115439/Users-Manual-3242046.pdf)

luka6000, 2021-03-11:
@Ocramius, so can you confirm that this meter works? Can you set attribute to enable the readings?

Ocramius, 2021-03-11:
@luka6000 I haven't really been able to get back to it - will certainly try it though, as I'm setting up my small solar farm, so it should come back to my priorities soon-ish :)