dresden-elektronik/deconz-rest-plugin #3023: Cannot pair IKEA TRADFRI LED1837R5 / GU10

excedra, 2020-07-08:
I have an issue paring my IKEA TRADFRI LED1837R5.

No matter what I'll do, I cannot get them into paring mode. For paring, Deconz/the Pi is directly placed on the Lamp (with an powerbank), so distance should be no issue. 

Fresh ones (directly out of the box) do also not start a pairing process on the first power on.

I've tried resetting them:
- 6 times on/off. This does not work. Simply no reaction.
- I've bought an Philips Dimmer Switch (https://www.reddit.com/r/tradfri/comments/dm0rv3/psa_moving_tradfri_gu10_bulbls_from_hue_hub_to/). This does reset the bulb (seems so, it does do the on/on/flickering thing) but the bulb doesn't do any paring.

I've tried different ways on the Deconz side...
- Start paring before resetting the bulb
- Start paring after resetting the bulb
- Start paring while resetting the bulb
- Set paring time to 10 minutes

How to get them paired? Other devices word fine.

_Originally posted by @excedra in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2238#issuecomment-653782561_

excedra, 2020-07-08:
> I have an issue paring my IKEA TRADFRI LED1837R5.
> 
> No matter what I'll do, I cannot get them paired. For paring, Deconz/the Pi is directly placed on the Lamp (with an powerbank), so distance should be no issue.
> 
> Fresh ones (directly out of the box) do also not start a pairing process on the first power on.
> 
> I've tried resetting them:
> 
> * 6 times on/off. This does not work. Simply no reaction.
> * I've bought an Philips Dimmer Switch (https://www.reddit.com/r/tradfri/comments/dm0rv3/psa_moving_tradfri_gu10_bulbls_from_hue_hub_to/). This does reset the bulb (seems so, it does do the on/on/flickering thing) but the bulb doesn't do any paring.
> 
> I've tried different ways on the Deconz side...
> 
> * Start paring before resetting the bulb
> * Start paring after resetting the bulb
> * Start paring while resetting the bulb
> * Set paring time to 10 minutes
> 
> How to get them paired? Other devices word fine.
> 
> _Originally posted by @excedra in [#2238 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2238#issuecomment-653782561)_

Mimiix, 2020-07-08:
Hi!
What does the logging show?

Can you also provide some info about your environment in terms of versions, install method and hardware?

excedra, 2020-07-08:
Hardware: ConBee II
Version: 2.05.77 /22.5.2020
Firmware: 264A0700

Running on an Orange pi zero with Debian 10.4.
Installed usind the deb-package via apt/repo.

Date is correct on the system.
Zigbee-Channel is currently 11

The reset using the Philips Dimming Remote was at 15:12:20:

> 15:12:20:334 Mgmt_Lqi_req zdpSeq: 131 to 0x680AE2FFFE6EA49A start index 0
15:12:20:334 APS-DATA.request id: 162, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
15:12:20:361 APS-DATA.confirm id: 162, status: 0x00 SUCCESS
15:12:20:361 APS-DATA.confirm request id: 162 -> confirmed, timeout 13588768
15:12:20:375 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -61
15:12:20:375 APS-DATA.indication request id: 162 -> finished
15:12:20:376 APS-DATA.request id: 162 erase from queue
15:12:20:376 ZDP status = 0x00 -> SUCCESS
15:12:20:376 ZDP Mgmt_Lqi_rsp zdpSeq: 131 from 0x680AE2FFFE6EA49A total: 5, startIndex: 0, listCount: 3
15:12:20:376     * neighbor: 0x00212EFFFF059482 (0x0000), LQI: 163, relation: 0x02 rxOnWHenIdle: 1
15:12:20:376     * neighbor: 0x680AE2FFFE43F767 (0x05E7), LQI: 73, relation: 0x02 rxOnWHenIdle: 1
15:12:20:376     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 150, relation: 0x00 rxOnWHenIdle: 1
15:12:21:532 Current channel 11
15:12:21:537 CTRL got nwk update id 1
15:12:21:542 Device TTL 1387 s flags: 0x7
15:12:22:734 Mgmt_Lqi_req zdpSeq: 134 to 0x680AE2FFFE6EA49A start index 3
15:12:22:734 APS-DATA.request id: 175, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
15:12:22:758 APS-DATA.confirm id: 175, status: 0x00 SUCCESS
15:12:22:759 APS-DATA.confirm request id: 175 -> confirmed, timeout 13588768
15:12:22:773 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -59
15:12:22:774 APS-DATA.indication request id: 175 -> finished
15:12:22:774 APS-DATA.request id: 175 erase from queue
15:12:22:774 ZDP status = 0x00 -> SUCCESS
15:12:22:774 ZDP Mgmt_Lqi_rsp zdpSeq: 134 from 0x680AE2FFFE6EA49A total: 5, startIndex: 3, listCount: 2
15:12:22:774     * neighbor: 0x680AE2FFFEC3D80A (0xBEEF), LQI: 160, relation: 0x01 rxOnWHenIdle: 0
15:12:22:774     * neighbor: 0x001788010872FE40 (0xDEC2), LQI: 116, relation: 0x01 rxOnWHenIdle: 0
15:12:25:613 Mgmt_Lqi_req zdpSeq: 137 to 0x00212EFFFF059482 start index 0
15:12:25:613 APS-DATA.request id: 191, addrmode: 0x03, addr: 0x00212effff059482, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
15:12:25:629 APS-DATA.confirm id: 191, status: 0x00 SUCCESS
15:12:25:629 APS-DATA.confirm request id: 191 -> confirmed, timeout 13588768
15:12:25:635 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -59
15:12:25:635 APS-DATA.indication request id: 191 -> finished
15:12:25:635 APS-DATA.request id: 191 erase from queue
15:12:25:636 ZDP status = 0x00 -> SUCCESS
15:12:25:636 ZDP Mgmt_Lqi_rsp zdpSeq: 137 from 0x00212EFFFF059482 total: 2, startIndex: 0, listCount: 2
15:12:25:636     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 252, relation: 0x02 rxOnWHenIdle: 1
15:12:25:636     * neighbor: 0x680AE2FFFE6EA49A (0x9226), LQI: 252, relation: 0x02 rxOnWHenIdle: 1
15:12:27:700 GW update firmware not found: deCONZ_ConBeeII_0x26490700.bin.GCF
15:12:28:662 Mgmt_Lqi_req zdpSeq: 140 to 0x680AE2FFFE43F767 start index 0
15:12:28:662 APS-DATA.request id: 207, addrmode: 0x03, addr: 0x680ae2fffe43f767, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
15:12:28:685 APS-DATA.confirm id: 207, status: 0x00 SUCCESS
15:12:28:685 APS-DATA.confirm request id: 207 -> confirmed, timeout 12622568
15:12:28:722 APS-DATA.indication srcAddr: 0x05e7, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -61
15:12:28:722 APS-DATA.indication request id: 207 -> finished
15:12:28:722 APS-DATA.request id: 207 erase from queue
15:12:28:722 ZDP status = 0x00 -> SUCCESS
15:12:28:722 ZDP Mgmt_Lqi_rsp zdpSeq: 140 from 0x680AE2FFFE43F767 total: 3, startIndex: 0, listCount: 3
15:12:28:722     * neighbor: 0x00212EFFFF059482 (0x0000), LQI: 32, relation: 0x02 rxOnWHenIdle: 1
15:12:28:722     * neighbor: 0x680AE2FFFE6EA49A (0x9226), LQI: 68, relation: 0x02 rxOnWHenIdle: 1
15:12:28:722     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 37, relation: 0x02 rxOnWHenIdle: 1
15:12:32:501 Mgmt_Lqi_req zdpSeq: 145 to 0x680AE2FFFE6EA49A start index 0
15:12:32:501 APS-DATA.request id: 228, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
15:12:32:522 APS-DATA.confirm id: 228, status: 0x00 SUCCESS
15:12:32:522 APS-DATA.confirm request id: 228 -> confirmed, timeout 13595880
15:12:32:537 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -61
15:12:32:537 APS-DATA.indication request id: 228 -> finished
15:12:32:538 APS-DATA.request id: 228 erase from queue
15:12:32:538 ZDP status = 0x00 -> SUCCESS
[... 91 lines cut ...]
15:12:55:095 aps request id: 33 finished, erase from queue
15:12:57:657 GW update firmware not found: deCONZ_ConBeeII_0x26490700.bin.GCF
15:12:57:733 Mgmt_Lqi_req zdpSeq: 172 to 0x680AE2FFFE6EA49A start index 0
[... cut ...]
15:12:59:872 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -62
15:12:59:872 APS-DATA.indication request id: 120 -> finished
15:12:59:872 APS-DATA.request id: 120 erase from queue
15:12:59:872 ZDP status = 0x00 -> SUCCESS
15:12:59:872 ZDP Mgmt_Lqi_rsp zdpSeq: 174 from 0x680AE2FFFE6EA49A total: 5, startIndex: 3, listCount: 2
15:12:59:872     * neighbor: 0x680AE2FFFEC3D80A (0xBEEF), LQI: 160, relation: 0x01 rxOnWHenIdle: 0
15:12:59:872     * neighbor: 0x001788010872FE40 (0xDEC2), LQI: 116, relation: 0x01 rxOnWHenIdle: 0

Mimiix, 2020-07-08:
First things first: please update to 0.78. There is some fix in there for IKEA bulbs. This might be affected. Also make sure to update firmware.

After updating: Check if you can add the light with Phoscon , searching lights.  If you still can't : Please provide me a log of the search:)

excedra, 2020-07-08:
OK, Updated. Now on 2.05.78 / 22.5.2020 and Firmware 26580700 - but still no change...

Reset of the bulb was at 17:49:50.

17:48:47:992 APS-DATA.confirm id: 71, status: 0x00 SUCCESS
17:48:47:992 APS-DATA.confirm request id: 71 -> confirmed, timeout 31509648
17:48:47:998 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 239, rssi: -64
17:48:47:999 APS-DATA.indication request id: 71 -> finished
17:48:47:999 APS-DATA.request id: 71 erase from queue
17:48:47:999 ZDP status = 0x00 -> SUCCESS
17:48:47:999 ZDP Mgmt_Lqi_rsp zdpSeq: 52 from 0x00212EFFFF059482 total: 3, startIndex: 0, listCount: 2
17:48:47:999     * neighbor: 0x680AE2FFFE6EA49A (0x9226), LQI: 243, relation: 0x02 rxOnWHenIdle: 1
17:48:47:999     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 255, relation: 0x02 rxOnWHenIdle: 1
17:48:48:987 Current channel 11
17:48:48:992 CTRL got nwk update id 1
17:48:48:996 Device TTL 3372 s flags: 0x7
17:48:50:232 APS-DATA.indication srcAddr: 0xdec2, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0xFC00, lqi: 239, rssi: -64
17:48:50:233 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 001
17:48:50:238 APS-DATA.indication srcAddr: 0xdec2, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -64
17:48:50:358 Mgmt_Lqi_req zdpSeq: 54 to 0x00212EFFFF059482 start index 2
17:48:50:358 APS-DATA.request id: 88, addrmode: 0x03, addr: 0x00212effff059482, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
17:48:50:375 APS-DATA.confirm id: 88, status: 0x00 SUCCESS
17:48:50:375 APS-DATA.confirm request id: 88 -> confirmed, timeout 34908552
17:48:50:382 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 121, rssi: 4
17:48:50:382 APS-DATA.indication request id: 88 -> finished
17:48:50:382 APS-DATA.request id: 88 erase from queue
17:48:50:382 ZDP status = 0x00 -> SUCCESS
17:48:50:382 ZDP Mgmt_Lqi_rsp zdpSeq: 54 from 0x00212EFFFF059482 total: 3, startIndex: 2, listCount: 1
17:48:50:382     * neighbor: 0x680AE2FFFE43F767 (0x05E7), LQI: 36, relation: 0x02 rxOnWHenIdle: 1
17:48:50:947 APS-DATA.indication srcAddr: 0xdec2, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 239, rssi: -64
17:48:50:947 ZDP device announce: 0x001788010872FE40, 0xDEC2, 0x80
17:48:50:948 DeviceAnnce of SensorNode: 0x001788010872FE40 [1]
17:48:50:985 APS-DATA.indication srcAddr: 0xdec2, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0xFC00, lqi: 239, rssi: -64
17:48:50:986 no button map for: RWL021 ep: 0x02 cl: 0xFC00 cmd: 0x00 pl[0]: 001
17:48:52:989 don't close database yet, keep open for 900 seconds
17:48:53:718 Mgmt_Lqi_req zdpSeq: 56 to 0x680AE2FFFE43F767 start index 0
17:48:53:718 APS-DATA.request id: 107, addrmode: 0x03, addr: 0x680ae2fffe43f767, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
17:48:53:739 APS-DATA.confirm id: 107, status: 0x00 SUCCESS
17:48:53:739 APS-DATA.confirm request id: 107 -> confirmed, timeout 34831200
17:48:53:782 APS-DATA.indication srcAddr: 0x05e7, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 239, rssi: -64
17:48:53:782 APS-DATA.indication request id: 107 -> finished
17:48:53:782 APS-DATA.request id: 107 erase from queue
17:48:53:782 ZDP status = 0x00 -> SUCCESS
17:48:53:782 ZDP Mgmt_Lqi_rsp zdpSeq: 56 from 0x680AE2FFFE43F767 total: 3, startIndex: 0, listCount: 3
17:48:53:782     * neighbor: 0x00212EFFFF059482 (0x0000), LQI: 46, relation: 0x02 rxOnWHenIdle: 1
17:48:53:782     * neighbor: 0x680AE2FFFE6EA49A (0x9226), LQI: 77, relation: 0x02 rxOnWHenIdle: 1
17:48:53:782     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 31, relation: 0x02 rxOnWHenIdle: 1
17:48:55:638 APS-DATA.request id: 118, addrmode: 0x02, addr: 0xfffd, profile: 0x0000, cluster: 0x0000, ep: 0x00 -> 0x00 queue: 1 len: 11 tx.options 0x00
17:48:55:639 send NWK_addr_req to 0xF0D1B80000148FC3, last seen 0 s, last seen by neighbors 0 s
17:48:55:717 APS-DATA.confirm id: 118, status: 0x00 SUCCESS
17:48:57:558 Mgmt_Lqi_req zdpSeq: 60 to 0x680AE2FFFE6EA49A start index 0
17:48:57:558 APS-DATA.request id: 128, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
17:48:57:581 APS-DATA.confirm id: 128, status: 0x00 SUCCESS
17:48:57:581 APS-DATA.confirm request id: 128 -> confirmed, timeout 34222712
17:48:57:595 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 215, rssi: -67
17:48:57:595 APS-DATA.indication request id: 128 -> finished
17:48:57:595 APS-DATA.request id: 128 erase from queue
17:48:57:595 ZDP status = 0x00 -> SUCCESS
17:48:57:595 ZDP Mgmt_Lqi_rsp zdpSeq: 60 from 0x680AE2FFFE6EA49A total: 5, startIndex: 0, listCount: 3
17:48:57:595     * neighbor: 0x00212EFFFF059482 (0x0000), LQI: 153, relation: 0x02 rxOnWHenIdle: 1
17:48:57:595     * neighbor: 0x680AE2FFFE43F767 (0x05E7), LQI: 82, relation: 0x02 rxOnWHenIdle: 1
17:48:57:595     * neighbor: 0x04CF8CDF3C763F7A (0xF0A7), LQI: 152, relation: 0x00 rxOnWHenIdle: 1
17:48:58:991 don't close database yet, keep open for 900 seconds
17:48:59:958 Mgmt_Lqi_req zdpSeq: 62 to 0x680AE2FFFE6EA49A start index 3
17:48:59:958 APS-DATA.request id: 140, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
17:48:59:979 APS-DATA.confirm id: 140, status: 0x00 SUCCESS
[... cut ...]
17:49:27:319 APS-DATA.request id: 43, addrmode: 0x03, addr: 0x680ae2fffe6ea49a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
17:49:27:340 APS-DATA.confirm id: 43, status: 0x00 SUCCESS
17:49:27:340 APS-DATA.confirm request id: 43 -> confirmed, timeout 34830960
17:49:27:356 APS-DATA.indication srcAddr: 0x9226, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 239, rssi: -64
17:49:27:356 APS-DATA.indication request id: 43 -> finished
17:49:27:356 APS-DATA.request id: 43 erase from queue
17:49:27:356 ZDP status = 0x00 -> SUCCESS

SwoopX, 2020-07-10:
I don't see any hints deconz entered pairing mode nor any device makes any attempt to pair. How do you pair, Phoscon lights search?

ursusHU, 2020-07-16:
I have LED1737R5/LED1739R5, in my case I had a similar problem. Try to reset the button not with 6 turn off and on, but with 7. Strange but this way it was working for me.

MattWestb, 2020-07-16:
@ursusHU I think it was the reset problems with the WS2.0. From the release note:
"WS 2.0 opal bulb V-2.0.023.
Improvement ImageID updated in attribute.
Improvement factory reset time is aligned with existing devices."
I don't know if the the LED1537R6/LED1739R5 its the WS2.0 but the new version with glass its and with the update firmware its behaving like other Ikea bulb. Most Ikea bulbs have getting Z3 upgrade and fixes and some old LL (RGBW) have also getting an fixes the last month.

excedra, 2020-07-16:
@SwoopX: Can you please provide my the parameters for deCONZ for the loggin you need? I am not familiar with the logs of deCONZ/Zigbee. Then I will re-run this.

Hessenpower01, 2020-07-16:
Similar here.

Resetting bulb w 6x on/off works fine. Additional on/off and it pairs. But only one bulb pairs, not able to pair additional ones.

Running a Conbee II on a Pi 4B w Home Assistant. 2.05.78 and 264A0700

Mimiix, 2020-07-16:
To answer your last question @Hessenpower01 : You can't use the HA addon to update the firmware. See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-docker

and https://github.com/home-assistant/hassio-addons/issues/1347

MattWestb, 2020-07-16:
@Hessenpower01 Its normal with deCONZ need a reboot after some pared devices and also after some time running you must restart for paring to working.
@Mimiix / Dennis it's some for writing in the wicki or putting on the todo list with experience date.

Mimiix, 2020-07-16:
@MattWestb The HA addon isn't anything we manage. They are working on it. It is in the wiki.

MattWestb, 2020-07-17:
> @MattWestb The HA addon isn't anything we manage. They are working on it. It is in the wiki.

This bug was present in the Phoscon-GW official downloaded image from DE that i was first installed for more then 2 years ago.
And if its also present in the HA addon i have not tested then i running it stand alone but i think its more then likely then its one dervat of the core code in Phoscon.
If you don't believe my:
1 do some time travel in all the slated issues and you  finding many cases that coming from this bug direct or indirect.
2 make a new install and trying the scenario out. 
3 make the same as 2 but with one older version.
4 (most likely) Slat the case.
5 flagging the case at bugg and putting one experens date on it (< 2 years pleas).

SwoopX, 2020-07-17:
@excedra Well, it seems that you already enabled almost all debug output that is possbile, so from that perspective, there's probably not more that can be done. Another approach to be tried could be to enter joining directly via deconz GUI.

Mimiix, 2020-07-17:
@MattWestb I am not sure what you are on about. The bug you refer to is pickedup in the HA addon. The firmware itself can be updated trough that button on the stand alone images. 

If you have any issue or want to discuss your problems on this git: Open a seperate issue on your own.

MattWestb, 2020-07-17:
I was addressing Hessenpower01s problem with Paring not working after some time and its well documented and the root its in the firmware and / or deCONZ so you getting the paring problem in phoscon, old web app and in deconz-rest-plugin.
And i was as a community member trying to helping other member with one very well known problem.
By the way in the post before it was only referring to official info from Amazon,de (DEs selling contact) and FCC-ID (RaspBee II FCC cert) and the company registration in germany so its no secrets at all for EU citizens as you and my but perhaps not for other.

Mimiix, 2020-07-17:
My reply is unrelated to the this particular case and only to your messages. I understand that the information is open, but i won't allow sharing personal information of anyone else **in any case. This includes open information**. 

As i said before: If you have stuff to say, open a own issue on this git page. I don't mind the subject, but you can't hijack other peoples issues. This will be my last reply on this issue of this subject and it won't help @excedra to solve his issue.

Hessenpower01, 2020-07-17:
Thanks all for the help. What made it work:

updated FW with Raspbian
prolonged the search period from 1 minute to 3 minutes

pairing procedure:

- bulb off
- start pairing in Deconz
- put bulb in reset mode (6x on/ off), no additional on/ off!
- wait ...

this way it seems to work stable for me.

MattWestb, 2020-07-17:
Sounds great Hessenpower01 and i hope you have a stale system for your lights.
True updating the firmware in the bulbs 2 its normally helping.

Mimiix, 2020-07-17:
@Hessenpower01 That 3 minutes is imho something that is underrated. I have done that instantly as 1 min seems way to short. Glad you fixed it.

@excedra Can you try and see what @Hessenpower01's method does for you :)?

excedra, 2020-07-21:
@Mimiix: Yes - and not. Finally got them paired. BUT:
- Resetting them with the Philips Remote does not work. The bulb starts the dimming, but with the switch on/off method, it comtinues the dimming longer and powers doesn, then goes dimming on. This is the paring.
- Resetting them as instructed by IKEA (and many others in tuturials, etc.) with delays and other options does not work. You have to hammer the switch on and of as fast as possible. (from the ON state 12 times: 6 offs and 6 ons).
- Start pairing before the reset of the bulb. 1 minute for paring is fine.
- After paring the first one, i've tried paring two of them at the same time. That does not work. (but resetting them together works).
But...

All of this worked only after I completly resetted my deconz and started a new network. Now all is fine.

MattWestb, 2020-07-21:
Great that you have getting you system up and running.
Ikea bulbs can being one PITA to resetting and some was having bugs that its fixed the last month with new firmware so look if deCONZ can upgrade the firmware in you bulbs.
If the OTA upgrade not working first restart deCONZ and wait little and aktivate the OTA updater plugin and then repower one bulb at the time (taking around one hour / bulb).

ursusHU, 2020-08-17:
@excedra may I ask you how did you reset the deConz? I have a very similar problem, but would like to avoid reinstalling my whole RPI.

MattWestb, 2020-09-06:
I can confirming that LED1837R5 (and problerly all other IKEA WS2 lights) can doing some very strange things and not being able to resetting them and therefore not getting paired.
I was having 3 in one groupe and was reconfiguration one of them then the binding was not done correct (of my). Was sending one reset to the device and it was leaving the network and flashing for conformation it have being reseted. Was doing little more things and some minutes later i was trying to pairing it but no reaction from it (no blinking after 6 on/off or fading then being in pairing mode after power on). Was trying on /  off reset for 30 minutes without success.
Was using the old webbapp and stretching for touchlink devices and its there on channel 20 with one unique panid (if ZB3 devices is failing pairing they going in find and bind mode and its having doing that = own network). Blinking the device and its the right one and then sending reset and its confirming it with flashing the light. Paring it and all is OK.
That the device is making its own network then can't pairing is OK, but refusing being reseted with "on/off" is not and very bad behaviour !!!

One great thing having one RaspBee I that have working touchlink in the firmware and the old webb app :-))
If its dying i have one HUE remote dimmer for security for doing touch resets of bad devices.

jppataki, 2020-10-04:
The Seven on-off in a rapid sequence did it for me

closed 2020-11-09 by stale[bot]