dresden-elektronik/deconz-rest-plugin #3026: Tuya Wireless Control Sensor

kwatt67, 2020-07-10:
Hi,

I can get the Conbee to see this device but not to be able to do anything with it, any help would be appreciated.

I hope the info is okay to see what it is etc.

Thanks 

K.

- Product name: Zigbee Temperature Humidity Sensor Tuya Wireless Control Sensor w/ Display (Unbranded)
- Manufacturer: 0x1002
- Model identifier: Doesn't seem avaialble
- Device type: Temperature/Humidity sensor
  - [ ] Sensor

[image]
[image]
[image]

Mimiix, 2020-07-10:
@kwatt67 What do you mean with "not able to do anything with it"?

kwatt67, 2020-07-10:
@Mimiix I have the Conbee connected to Homeseer visa JowiHue and the plugin cannot see the sensor at all.

Though messing about with it I've managed to delete the sensor completely, re-adding it at the moment. Or, attempting to.

Wim (JowiHue) seemed to think it was not passing info to the RestAPI. Meaning it is not passing any information to that at all, his comment was:

"I am not sure as there seem to be different models around. One is found here: https://github.com/dresden-elektroni...in/issues/2442 which is solved it seems, but other do not see the device past to the REST Apii. JowiHue needs the devices through the REST API. Do any of you use POSTman? You could check that wat if the rest api is populating the devices or not? Or on the github, let the developers there check the details of deCONZ to see if it should work. You need to post the details of the clusters to let them check in detail.

The plugin 'should' accept this device out of the box normally once populated in the rest api"

That I suspect may make it all make sense, I hope.

K.

Mimiix, 2020-07-10:
First off: What version of deCONZ and the firmware are you on? 

If you are sure you are on the latest versions, use the [RestAPI](https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/) to see how it is exposed.

kwatt67, 2020-07-10:
The deCONZ version is: Commercial Version -  2.05.77

I just tried to connect postman per the instructions and I got this:

[
    {
        "error": {
            "address": "/",
            "description": "body contains invalid JSON",
            "type": 2
        }
    }
]

Mimiix, 2020-07-10:
If you can, please come to Discord :) Easier helping you that way.

kwatt67, 2020-07-10:
On there not but I've not used that ever so not entirely sure what I'm doing is correct. ;)

kwatt67, 2020-07-10:
I'll try that again, in English (that makes sense)... I am on there but seems I don't have permission to send messages.

Smanar, 2020-07-10:
New capture full 

https://media.discordapp.net/attachments/692306044495593544/731189167127658506/screenshot.jpg?width=272&height=300

kmplngj, 2020-08-01:
Any news on this? Is the device supported? I'm planning to buy this for my setup.

kwatt67, 2020-08-01:
Hi Jan,

Working perfectly here.

K.

SwoopX, 2020-08-02:
Closing this as the sensor is already supported.

closed 2020-08-02 by SwoopX