dresden-elektronik/deconz-rest-plugin #3028: Aqara Wireless Switch Controller (WXKG06LM)

elpablillo, 2020-07-10:
Hi, I've bought specifically the model WXKG03LM, but I've received WXKG06LM. I can pair it the device but PhosconApp says that  cannot find any device, so, it shows in the GUI and that's it. Can you add this model ? 
Thanks 

## Device

- Product name: Aqara Wireless Switch Controller
- Manufacturer: 0x1037
- Model identifier: 
- Device type: RFD
  - [ ] Light
  - [ ] Lock
  - [X] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [X] Other: Switch

## Screenshots

[image]

Mimiix, 2020-07-10:
I think it is implemented already. What does the RestAPI show?

If you see devices within deCONZ but not in Phoscon, that is most of the time perfectly normal. Phoscon is closed source and needs devices to be added manually by their Developer. However, the device will still be usable with the API/Home-Assistant/OpenHAB etc.

elpablillo, 2020-07-10:
RestAPI nothing... no switch, I've tried to add it many times, I delete the node in GUI, I put the device in pair mode, scan in the RestAPI, the device link to something, in the GUI appears the device ( but with no name,Model,Vender,etc..it's in the 1st pic) but in the RestAPI  it falis
[image]

SwoopX, 2020-07-10:
Please provide all requested screenshots. The basic cluster is missing (when the attributes have been read)

elpablillo, 2020-07-10:
I'm not sure what screenshot do you need... this one? (I'm starting so I don't have much) 
[image]

Mimiix, 2020-07-10:
Please refer to https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

elpablillo, 2020-07-10:
Oh, thanks for the link...
[image]

SwoopX, 2020-07-10:
Huh? Isn't that an Aqara Opple 3 gang?

elpablillo, 2020-07-10:
Well, not very well... I have two devices of this model and both with the same result, they show in the GUI but that's all, not in HA and not in PhosconApp...And I have the last firmware v2.5.77, So...What am I doing wrong?; the process it is not complicate, it's quite easy :(

Mimiix, 2020-07-10:
Can you provide some HA logging?

If you need guidance, join the Discord. In the readme there is a URL to the server :)

elpablillo, 2020-07-13:
Hi all, are you going to add the lumi.remote.b186acn02 to the Firmware version V2_05_78, I didn't see those new lines in the Pre-release, and I wouldn't want you to forget about my device :). Thanks

ekesgin, 2020-07-22:
I'm also waiting the support for this device.

king7222, 2020-07-24:
+1 here 😊, exactly like elpablillo I bought two WXKG03LM and received two WXKG06LM, and they are not detected in my Jeedom with Deconz plug-in after activating the include mode 😕.
I hope this device will be included in the next update, because I have no use for these switches otherwise. 

Thanks for all the work

Smanar, 2020-07-25:
What is the model id, it s not the same device than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3066 ?

king7222, 2020-07-25:
Yes it has only one switch but the model id is the same : lumi.remote.b186acn02

Smanar, 2020-07-25:
Yep this device is in the PR > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3053/commits/7c01bb1dbefad841684f8d02863d2f7ed208383d

king7222, 2020-08-16:
Thank you very much for your work, Wxkg06lm (lumi.remote.b186acn02) just works perfectly with 2.05.80 👍
I could add my two switches in Jeedom without any issue. 
Thanks

elpablillo, 2020-08-16:
Well, in my case it still shows in the deconz-GUI, but still not showing in the WebApp... and in HA it only shows the battery level entity, i had to go to Developer Tools > events and listen the deconz_event: 
{
    "event_type": "deconz_event",
    "data": {
        "id": "lumi_remote_b186acn02_10",
        "unique_id": "00:15:8d:00:04:52:50:4f",
        "event": 1002
    },
    "origin": "LOCAL",
    "time_fired": "2020-08-16T10:03:39.391325+00:00",
    "context": {
        "id": "c1e74d05dfa711ea92fc5d5adfd4cf0f",
        "parent_id": null,
        "user_id": null
    }
}
Getting the ID and the event I could create an automation. 
event_data:
  event: 1002
  id: lumi_remote_b186acn02_9
event_type: deconz_event
platform: event
So, it is working but in a rustic way.

SwoopX, 2020-08-16:
Phoscon is not deconz. You might want to request support for it at the Phoscon beta repository.

closed 2020-09-13 by stale[bot]

maximilien03, 2020-11-07:
Hi !
I also got two of those WXKG06LM. It looks like the old references (which were perfectly working with Phoscon app) are not produced anymore.
Pairing was successful with Phoscon app. However, they don't appear in the switch editor. I guess they are not fully supported yet.
HA allows me to make automations based on events, but it would be way better if I could configure those switches with Phoscon.