dresden-elektronik/deconz-rest-plugin #3032: Aquara remote switch WXKG03LM doesn't appear in deCONZ GUI

LestatSan, 2020-07-11:
Hi, 

I use a conbee II on a RPI3B+ with raspbian strech. I use the phoscon App (last version) to set up my instalation.
I installed deCONZ GUI (2.5.77) on a PC witch windows 10.
I use the last version oh the conbee II firmware

I'd like to bind my switch (WXKG03LM) directly with my light.

I succeed to pair the switches in the Phoscon App, but they're not present in the network in deCONZ GUI... 

Does anyone have an idea to solve it ?

Thanks

Smanar, 2020-07-11:
Hello, there is lot of thing I don't understand

- Not possible make direct bind with Xiaomi.
- Not possible having device in phoscon but not in deconz.

BTW, we have find a new model ID for a xiaomi device yesterday, if you can give us your, I have perhaps more information.

LestatSan, 2020-07-11:
> Not possible make direct bind with Xiaomi.
Ok...  This is what I was worried about.... Is there a list of the switches we can direct bind ? I didn't found anything.

>Not possible having device in phoscon but not in deconz.
That's a thing I dont understand... You can see the screen captures with the links below :
https://zupimages.net/viewer.php?id=20/28/zr9k.png
0x0000 is the conbee II
0xC4AF is a Trafri light
0xB87A is a Trafri switch

I have 2 more Aqara switches in Phoson App

https://zupimages.net/viewer.php?id=20/28/n5fo.png

The references I found on the switch : 
Model : WXKG03LM
FCC ID : 2AKIT-WXKG02LM
IC / 22635-WXKG02LM
Lumi United technology

And in Phosocn App
The default name of the switch : lumi.remote.b186acn01 2
Product : WXKG03LM Rev.2
Version :  20180525

Smanar, 2020-07-11:
I think all other switch, except Xiaomi ^^. It s easy to reconize them, if you have a group created in same time you include the switch, it s supported.
The opple can, but idk if it s implemented. Choose one and make a search on the issue with the name.

>    { VENDOR_JENNIC, "lumi.remote.b186acn01", jennicMacPrefix },  // Xiaomi single button wall switch WXKG03LM 2018
It seem you have the correct model id, your device is supported.

And I don't understand what happen on your configuration, are you sure your switches are still on the API, you can see button even when using them ?
You can have a device in deconz and not in phoscon, but not the reverse.

LestatSan, 2020-07-11:
>I think all other switch, except Xiaomi ^^. It s easy to reconize them, if you have a group created in same time you include the switch, it s supported.

Arf... Ok. I'll deal with this for now, i'll see later or for the next ones. Thank you.

>And I don't understand what happen on your configuration, are you sure your switches are still on the API, you can see button even when using them ?
You can have a device in deconz and not in phoscon, but not the reverse.

I made a group, and the on/off works fine but I just have the single tap, not the double tap and long tap, but I don't know if it's implemented.
Jeedom uses the API to implement the devices, and I can use the switch with Jeedom (and I have access to the double and long tap).
Just too be sure, I tried with deCONZ on a Ubuntu, and I had the same issue.
May be this switch need a very long time (+20min) or request to be seen in deconz ?

I had a second issue wich seems to be known : The tradfri remote didn't create its own group in Phoscon App. I had to make a group and pair the remote witch the light and the tradfri remote's group was created. But I don't think it's related.

Smanar, 2020-07-11:
If I m right on deconz you haven't GUI naturally.
So what is your setting ? you are using deconz on another machine ? Another VM ? Using X-Forwarding (I already have this kind of bug with it, was like the user have 2 deconz running in same time, action on one was invisible on the second) ?

LestatSan, 2020-07-11:
You're right, Jeedom haven't Deconz with GUI, so I use deconz desktop on another machine.

I have Jeedom running on a Rpi3B+. My Conbee II is plug into it with an usb extension. I paired the light and switches with Phoscon with it. During my reseach to have a direct binding (and/or switch groups in deconz) I was recommended to install deconz with GUI on a computer with no other deconz instance. So I installed it on a PC with Windows 10Pro (and on another with ubuntu 20.04, just to try). The screen captures were made on my W10 computer.

Smanar, 2020-07-11:
So you have paired your device on the PI
And using deconz ATM on windows for test, and both capture, deconz and phoscon are from the windows machine ?

Have you used the backup/restaure feature available in Phoscon/Gateway/backup options ?
If no, make a backup on Phoscon/Raspberry and a restaure in Phoscon/windows.

The network setting is stored on the device but the device are stored on the host.

LestatSan, 2020-07-11:
Both capture are from the same windows machine.

I made a backup on phoscon/RPI  and a restore on phoscon/W10. It didn't change anything.
Earlier, I paired all my devices (3light tradfri, 2 xiaomi aqara and 2 trafri remote). A see all of them in phoscon, and only the light and one tradfri remote in deconz.

SwoopX, 2020-07-11:
Take some debug output from the raspbee. When it's booted, first stop deconz with `sudo systemctl stop deconz-gui`. Then, start it again as follows `/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > debug`. Let it run for about 2 mins and share the debug file.

LestatSan, 2020-07-12:
I dont have deconz-gui running on my RPi, so I wrote : 

sudo systemctl stop deconz
/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > debug

It returned me : 
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile

You will find the debug in the attached file

[debug deconz lestatsan.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4908188/debug.deconz.lestatsan.txt)

Smanar, 2020-07-12:
So first in this log

I can see the remote group > 09:26:21:465 Sqlite group: name = TRADFRI remote control 
I can see the Xiaomi device, why not in deconz ? In deconz there is something in panel/ node list, can you check if the device aaren't here ?
The gateway have failed to connect, idk if it s something normal ?

LestatSan, 2020-07-13:
There's nothing more in panel node list

For the gateway, idk, the Conbee was plugged...

Smanar, 2020-07-13:
Are you sure you havent many deconz instance running ??
What is the result of command
`ps ax | grep deCONZ`

Because on your last log, you have 0xCa4F in your deconz screen capture, but not in your log, and almost all, deconz have failed to connect, so it s not possible this log come from a working network at this moment.

LestatSan, 2020-07-14:
Normaly, I've got only on instance running.... The command returned this : 
`  380 ?        Ss     3:13 /bin/bash /usr/bin/deCONZ-update2.sh
 1790 ?        S      0:00 sudo deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-info=0 --dbg-aps=0 --dbg-zcl=0 --dbg-zdp=0 --dbg-http=0
 1795 ?        Sl    59:42 deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-info=0 --dbg-aps=0 --dbg-zcl=0 --dbg-zdp=0 --dbg-http=0
25814 pts/0    S+     0:00 grep --color=auto deCONZ
`

Does that mean I've got 2 instances running ?

Thanks for your help

Smanar, 2020-07-14:
Nope it s ok, sudo is an instance too.

But ATM you haven't deconz with GUI running ? Can you make the same command in the same situation than when you have take the capture in deconz ?

LestatSan, 2020-07-14:
I can do it, but it's not on the same machine. 
All the tests were made witch my Jeedom's RPI with no GUI. I only used the GUI on an other machine to get the capture and check /see the zigbee network

Smanar, 2020-07-14:
For me there is a database mismatch somewhere.
BTW your problem is rather the direct binding.
On the normal installation (with jeedom, headless), on the homepage, you haven't a group called "TRADFRI remote control" with 5 device in it ?

LestatSan, 2020-07-14:
I had it, but wasn't automaticaly created, it appears after a direct bind between the remote and the ligth. Right now, in phoscon, the ConbeeII doesn't have the IP adress of my Pi and there's no groups or light in the phoscon app. I'm gonna loose my mind with this...

I think you're right, I have a database mismatch and I think it should be a (huge ?) part of my problem : 
Right now, I can control the light with Jeedom, but I don't have anything (light, remote controls....) in phoscon app.

I think I'll reset the gateway and reload the firmware to get a "fresh install"

Smanar, 2020-07-14:
No, I don't thnik you need to reset something, and reset the conbee or the firmware will chnage nothing on your installation.
For me the problem is software or from your network, the conbee haven't the jeedom adress, but wich one it have ?
It s realy like you have find a way to run 2 instances in same time ^^.

About the pairing, remember there is 2 method for the remote, the 4 press for inclusion in network, and the long press for direct bind. Never use the last one with deconz.

LestatSan, 2020-07-14:
I think I found why the conbee didn't have the jeedom adress... I checked the zigbee network with my computer wich has the deCONZ GUI and I didn't quit disconnect correctly the conbee.... Now it's ok, the conbee have my Jeedom adress.

> It s realy like you have find a way to run 2 instances in same time ^^.
Oh lucky me, I'm gonna be rich lol

About the pairing with the ikea remote, I used the 4 press button method, the remote appeared in phoscon app, but the remote group wasn't created. Then I paired the remote and the light with the long press. That created the remote network in deCONZ.

What's the problem with the long press ?

LestatSan, 2020-07-14:
In sum the situation here : 

XIAOMI : 
 - We can't direct bond Xiaomi switches => ok
 - I can't see the Xiaomi switches in deCONZ GUI, BUT they're present in phoscon app and works fine with Jeedom.

Ikea :
 - When I paired the Tradfri remote in phoscon app. It had never created a switch group. So, I made a light group with my light and the switch, then I paired the switch (long press for direct bind)  with the light => The remote group appeared in phoscon app.

Smanar, 2020-07-14:
The correct method is
- include the bulb
- include the remote with the 4 press
- bind them using the group (and idk why you haven't group created at this moment)

If you use the long press on remote, on normal way , the both device will leave the deconz network, to join a new network and bind them together.

LestatSan, 2020-07-14:
Ok, thanks for your help !

closed 2020-08-14 by stale[bot]