dresden-elektronik/deconz-rest-plugin #3034: Can't read configurations error -71 / -110

gvdhoven, 2020-07-11:
## Describe the bug
Deconz II zigbee stick loses configuration

## Steps to reproduce the behavior
After plugging the stick in my Photon OS / Docker image it gives (at) random errors about the configuration

## Expected behavior
n/a

## Screenshots
[image]
[image]

## Environment
 - Host system: PC (even tried fresh install)
 - Running method: Marthoc Docker container (latest version)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: yes (also tested without, tested also different ports as can be seen in the screenshot)

## Additional context
Eventually caused a kernel panic (have only seen that once) -> Ran memtest 4 passes without issues. 
System has been running stable for months without any hardware changes.

gvdhoven, 2020-07-11:
More info:
[image]

Contents of `/etc/udev/rules.d/98-docker-usb.rules`:
```
# ConBee II disbale modem manager and mtp probing
SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="ttyUSB.zigbee"
```

Even tried without first two lines.

Smanar, 2020-07-11:
But it s not a problem from your usb port ?

Can you make a try with a simple USB key ? to check you can use  the USB port ?

gvdhoven, 2020-07-11:
USB drive works; here are the logs from docker after re-attaching it (with USB extension cable) to the USB2 port:
```
2020-07-11T16:30:33.555790748Z [marthoc/deconz] Starting deCONZ...,
2020-07-11T16:30:33.555811665Z [marthoc/deconz] Current deCONZ version: 2.05.78,
2020-07-11T16:30:33.555814677Z [marthoc/deconz] Web UI port: 8080,
2020-07-11T16:30:33.555816653Z [marthoc/deconz] Websockets port: 8443,
2020-07-11T16:30:33.555819080Z [marthoc/deconz] VNC port: 5900,
2020-07-11T16:30:33.728398912Z /usr/bin/xauth:  file /root/.Xauthority does not exist,
2020-07-11T16:30:33.830472813Z ,
2020-07-11T16:30:33.830513830Z New '7969bd5999e3:0 (root)' desktop at :0 on machine 7969bd5999e3,
2020-07-11T16:30:33.830518720Z ,
2020-07-11T16:30:33.830521759Z Starting applications specified in /etc/X11/Xvnc-session,
2020-07-11T16:30:33.830524787Z Log file is /root/.vnc/7969bd5999e3:0.log,
2020-07-11T16:30:33.830527592Z ,
2020-07-11T16:30:33.830530175Z Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd 7969bd5999e3:0 to connect to the VNC server.,
2020-07-11T16:30:33.830533178Z ,
2020-07-11T16:30:36.913085882Z libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri),
2020-07-11T16:30:36.913501345Z libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri),
2020-07-11T16:30:37.013022389Z libpng warning: iCCP: known incorrect sRGB profile,
2020-07-11T16:30:37.018392368Z 18:30:37:018 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/,
2020-07-11T16:30:37.055418760Z 18:30:37:034 CTRL. 3.16.218:30:37:055 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt,
2020-07-11T16:30:37.095170145Z 18:30:37:095 parent process /bin/sh,
2020-07-11T16:30:37.095262723Z 18:30:37:095 gw run mode: docker,
2020-07-11T16:30:37.095270763Z 18:30:37:095 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version,
2020-07-11T16:30:37.097861686Z 18:30:37:098 DB sqlite version 3.16.2,
2020-07-11T16:30:37.098188324Z 18:30:37:098 DB PRAGMA page_count: 42,
2020-07-11T16:30:37.098191179Z 18:30:37:098 DB PRAGMA page_size: 4096,
2020-07-11T16:30:37.098193076Z 18:30:37:098 DB PRAGMA freelist_count: 4,
2020-07-11T16:30:37.098211940Z 18:30:37:098 DB file size 172032 bytes, free pages 4,
2020-07-11T16:30:37.098272037Z 18:30:37:098 DB PRAGMA user_version: 6,
2020-07-11T16:30:37.098274589Z 18:30:37:098 DB cleanup,
2020-07-11T16:30:37.099453926Z 18:30:37:099 DB create temporary views,
2020-07-11T16:30:37.104256396Z 18:30:37:104 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:37.104746393Z 18:30:37:105 started websocket server at port 8443,
2020-07-11T16:30:37.106579132Z 18:30:37:106 found node plugin: libde_rest_plugin.so - REST API Plugin,
2020-07-11T16:30:37.107500318Z 18:30:37:107 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin,
2020-07-11T16:30:37.112245713Z 18:30:37:112 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin,
2020-07-11T16:30:42.541358875Z 18:30:42:541 Announced to internet https://phoscon.de/discover,
2020-07-11T16:30:45.096447978Z 18:30:45:096 New websocket 172.18.0.10:59702 (state: 3) ,
2020-07-11T16:30:47.265762784Z PROTO: CRC error,
2020-07-11T16:30:47.265789439Z PROTO: CRC error,
2020-07-11T16:30:47.265792033Z 18:30:47:266 Device firmware version 0x26580700,
2020-07-11T16:30:47.269755858Z 18:30:47:270 unlocked max nodes: 200,
2020-07-11T16:30:47.341744602Z 18:30:47:342 Device protocol version: 0x010B,
2020-07-11T16:30:47.346779978Z 18:30:47:347 new node - ext: 0x00212effff04f654, nwk: 0x0000,
2020-07-11T16:30:47.378805720Z 18:30:47:379 SensorNode 3 set node 0x001788010644a00d,
2020-07-11T16:30:47.378973777Z 18:30:47:379 SensorNode 4 set node 0x001788010644a00d,
2020-07-11T16:30:47.378980585Z 18:30:47:379 SensorNode 5 set node 0x001788010644a00d,
2020-07-11T16:30:47.386924703Z 18:30:47:387 SensorNode 6 set node 0x0017880106441058,
2020-07-11T16:30:47.386986461Z 18:30:47:387 SensorNode 8 set node 0x0017880106441058,
2020-07-11T16:30:47.387003235Z 18:30:47:387 SensorNode 7 set node 0x0017880106441058,
2020-07-11T16:30:47.394753083Z 18:30:47:395 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.402112140Z 18:30:47:402 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.402372217Z 18:30:47:402 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.402663584Z 18:30:47:403 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.402859450Z 18:30:47:403 LightNode 3: Philips White Ambiance 4 added,
2020-07-11T16:30:47.421200126Z 18:30:47:421 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.421231555Z 18:30:47:421 LightNode 4: Philips White Ambiance 3 added,
2020-07-11T16:30:47.439537588Z 18:30:47:439 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.439599381Z 18:30:47:440 LightNode 6: Philips White Ambiance 2 added,
2020-07-11T16:30:47.447249992Z 18:30:47:447 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.447324475Z 18:30:47:447 LightNode 7: Philips White Ambiance 5 added,
2020-07-11T16:30:47.458822637Z 18:30:47:459 don't close database yet, keep open for 900 seconds,
2020-07-11T16:30:47.458865745Z 18:30:47:459 LightNode 8: Philips White Ambiance 6 added,
[... cut ...]
2020-07-11T16:33:08.516984377Z 18:33:08:517 0x00178801083bd67f found group 0xFFF0,
2020-07-11T16:33:08.516986830Z 18:33:08:517 0x00178801083bd67f found group 0x0002,
2020-07-11T16:33:09.261023030Z 18:33:09:261 verified group capacity: 23 and group count: 2 of LightNode 0x00178801083bb4ca,
2020-07-11T16:33:09.261048484Z 18:33:09:261 0x00178801083bb4ca found group 0xFFF0,
2020-07-11T16:33:09.261050992Z 18:33:09:261 0x00178801083bb4ca found group 0x0002,
2020-07-11T16:33:14.244932955Z 18:33:14:245 don't close database yet, keep open for 900 seconds,
```

Mimiix, 2020-07-11:
As you are running it on PhotonOS, what happens if you use it with a native installation?

SwoopX, 2020-07-11:
From the last logs above, all is good.

gvdhoven, 2020-07-11:
@Mimiix i have been running it from PhotonOS for about 6 months without problems.

@SwoopX thanx for checking; the console message hasn't popped up yet. If it does, i will copy and paste a new part of logs. However, i do see twice a CRC error:
```
2020-07-11T16:30:47.265762784Z PROTO: CRC error,
2020-07-11T16:30:47.265789439Z PROTO: CRC error,
```

But i assume i can ignore that. This is also 'normal'?:
```
2020-07-11T17:48:45.088442887Z 19:48:45:088 0x0017880106B9635E error APSDE-DATA.confirm: 0xE9 on task
2020-07-11T17:46:36.979869170Z 19:46:36:980 0x00178801068D1C28 error APSDE-DATA.confirm: 0xE9 on task
```

And my server just died again;
```
7/11/2020, 6:24:32 PMnode: Deconz
msg : string[55]
"WebSocket error: Error: getaddrinfo ENOTFOUND ha-deconz"
7/11/2020, 6:30:35 PMnode: Deconz
msg : string[61]
"WebSocket error: Error: connect ECONNREFUSED 172.18.0.18:8443"
```

gvdhoven, 2020-07-11:
btw, i since now seriously doubt it is an issue with Deconz; since after my server hang; i got it to kernel panic 3 times, i'll close the issue for now;

[image]

closed 2020-07-11 by gvdhoven

SwoopX, 2020-07-11:
@gvdhoven The CRC erro can happen, afaik, a serial transmission got broken. The APSDE-DATA errors can occur from time to time, also nothing unusual. The last error messages do not originate from deconz...

You may want to check your HDD if your RAM is ok. Can also be a possible root cause.

gvdhoven, 2020-07-11:
For a kernel panic? I am running from an SSD, got any tips for me to check that for errors?

btw; i just 'resetted' my bios to factory defaults and that changed quite some things with CPU and timings (no clue that i changed that in the past), let's see what that does.

Mimiix, 2020-07-11:
There were some changes in the firmware with the udev configurations. I've seen atleast one case with unraid where this became an issue.