dresden-elektronik/deconz-rest-plugin #3035: Cannot create backup on Windows 10

Plantje1978, 2020-07-11:
I am using Windows 10 and I want to move my Domoticz and deCONZ installation to another machine, because this one is terribly slow!

If I use the Phoscon web page and try to create a backup I get: "Creating backup failed" If I try to do the same thing on the server itself it doesn't even show the page to click the button to start creating the backup.
Now, I don't know where the backup file will be created, but since the machine is very slow I actually need every click to be spot on...

Smanar, 2020-07-12:
Hello (I m thorgal), I have already found an issue with same problem (on windows too) but no answer > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1892

I have put some things to try on the other forum, can you tell me is one of them have worked ?

Smanar, 2020-07-12:
BTW, if we want to do it manually wich one files are vital ?
The network config is on the device, so only the zll.db is usefull ?

On the Phoscon backup there is to a file called "sessions.defaut", what is the utility of this file ?

Mimiix, 2020-07-13:
@Plantje1978 First off: Please stick with the template. The version information is missing. This might not help you, but if we get more reports on a specific version it might be related. 

Which page are you using to access the phoscon webpage? The internal IP + port or are you using the phoscon.de/webdiscover?

Plantje1978, 2020-07-13:
Hi,

I will try to use the template. I didn't do it as I saw previous issues being logged for this issue. Unfortunately those have been closed without people getting back to the issue with the solution or additional info.

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
Version
 - Host system: PC 
 - Running method: Windows 10
 - Firmware version: 26330500)
 - deCONZ version: 2.05.69
 - Device: ConBee I
 - Do you use an USB extension cable: Yes

### deCONZ Logs
Enabling logging seems to be a Pi only thing; not sure how to do that on Windows

## Additional context
This seems to be some sort of rights issue. I am not sure where the code tries to write the file, but I have provided "everyone" full control on the %appdata% deCONZ folder and on another deCONZ folder.

I am using the internal IP + port to access the Phoscon webpage

Mimiix, 2020-07-13:
I'd suggest updating to the lastest versions first. It might been solved in those.

Plantje1978, 2020-07-13:
Will do... just to be on the safe side: I can just download the executable and install it without loosing anything?

Smanar, 2020-07-13:
Yep you will loose nothing. Network config is on the key, and device config are on the Host.

BTW I have a found a way (not easy) to enable debug on windows, I m seing so much debug line in the API code, so it can help to find the problem

> Unfortunately it seem the only way to have debugging on windows is using that https://docs.microsoft.com/en-us/sysint ... /debugview
> 
> Cf https://github.com/dresden-elektronik/d ... issues/980

Plantje1978, 2020-07-13:
Installed the latest version....now it states I need to update the firmware... no problem. However, it states, that I need to create a backup first....you see where I am going with this?

The issue still exists.... no way to create a backup.

Mimiix, 2020-07-13:
Just install it on a different user account or different PC. There's no actual configuration stored on the stick.

Plantje1978, 2020-07-13:
I don't get it.... this IS an attempt to install on a different machine. And the PC is this slow that it is not very easy to create a different user account just like that. Finally, I don't have the idea that upgrading the firmware version will all of a sudden enable the backup

Smanar, 2020-07-13:
Me too, new deconz version ok, but firmware ...

But if you can enable loging using the app, I think we will find the problem, take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L16762 there is so much debug information.

Mimiix, 2020-07-14:
@Plantje1978 I've asked @manup to take a look at this ;)

Plantje1978, 2020-07-14:
Great! Thanks!

I have control over the machine again and setup RDP so I don't have to rely on Teamviewer. If there is anything I need to test or set, please let me know.

Could you let me know where the code tries to write the backup? Is that something on the server? Or is it trying to push it for download?

Smanar, 2020-07-14:
It use 7za.exe to make an archive with files "deCONZ.conf" "zll.db" and "session.default" and after send it using the http server.

Plantje1978, 2020-07-14:
FYI: tried it again on the machine itself, this time using Chrome as browser: same result.

Smanar, 2020-07-14:
Your machine is realyy not powerfull enought to run the app for debugging on windows ?

Plantje1978, 2020-07-14:
No, I don't think so. Creating a backup is needed, because I want to move to a more powerful machine. I am not sure about development times, but I think I will wait for a bit

Smanar, 2020-07-14:
and if you make a search on your computer you cn't find thoses file ?  "deCONZ.tar"with files "deCONZ.conf" "zll.db" and "session.default"

Plantje1978, 2020-07-16:
If I find those files, how will it help me? I can try to do it manually, but I hope I will not break my entire home automation!

Smanar, 2020-07-16:
if you find it,
- you will know where decone make "his magic"
- we will know the bug is after the files creation, perhaps using the zip application
- we can make ourself the archive to make restauration.

Plantje1978, 2020-07-17:
Ok, well... fair enough... 

Some questions though...

- Has anyone ever been able to do a proper backup and restore using Windows?

- If I do a backup this tar file is created containing the three others?

I have my new machine ready, so I could always try to look for it and restore. Before doing this however, I need to check how again to search the whole hard drive as Windows by defaults doesn't search specific folders.

Will let you know!

Smanar, 2020-07-17:
Yep, It works on my side, but I have a special config, always admin, full right, security disabled.
On my side file is created at C:\Users\smanar\AppData\Local\dresden-elektronik\deCONZ, and yes the archive contain the 3 files

But deconz create first the file, and after propose a download, so it mean in your situation the bug appear before the file creation.

Plantje1978, 2020-07-17:
I will do some checking again and see if I can further loosen up some rights :) I will dismantle this machine after I am done migrating anyway. 

I had two of these tablets: an EXOPC and a WeTab (they actually are the same). The EXOPC had a bulky battery, not something safe to keep in the house. The WeTab indicates it has only 10% battery power left while it is constantly on the charger.

Plantje1978, 2020-07-17:
Hmmm... well ok... 
If I go to C:\Users\Familie Hendrix\AppData\Local\dresden-elektronik\deCONZ (which is my %appdata% location) I can see zll.db, session.default and config.ini (so not deCONZ.conf).
I tried making a copy of config.ini and rename it to deCONZ.conf. After that I tried creating a new backup. It still failed and my deCONZ.conf file was gone. And the files zll.db, session.default and config.ini had a new timestamp. (Weird enough all the same, but not my current timestamp.)
So, something is definitely happening there!

Smanar, 2020-07-17:
I don't think it s same file, the deCONZ.conf is created by the code, so if you haven't here, it mean you have the error at this place

BTW there is too an error check in the code https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L16856

Like
>if (securityMode != 3) { return false; } // High - No master but TC link key

Plantje1978, 2020-07-17:
I think I've found the issue...
I've located 7za.exe in the bin folder. And if I double click the executable it states that the app cannot be run from this computer and that I should ask the software vendor for a new version.
There are no details as to why it cannot be run, but this should definitely help!

Plantje1978, 2020-07-17:
That did the trick! Downloaded some version of 7za.exe and now it does work. Thanks for the help Smanar!

Let's see if this is as expected...
Now I got a raspbee_gateway_config_2020-07-17 and within that I have a deCONZ.tar with the promised files.

Seems fixed! Can I then also use that file for restoring? Or should I use the deCONZ.tar?

Plantje1978, 2020-07-17:
By the way: wanted to share this on the Domoticz forum as well, but the site seems to be down.

Smanar, 2020-07-17:
lol, one more time ^^

raspbee_gateway_config_2020-07-17 and  deCONZ.tar are the same file.
But I m suprised, because the file deCONZ.conf  is created before the use of 7za.exe, so I don't understand why the file is not here.

Plantje1978, 2020-07-17:
But I am happy! Just before the weekend, I have everything in place to migrate my home automation server

SwoopX, 2020-07-17:
Just out of curiosity: how can your current system not be powerful enough? I run deconz, my home automation software and lots of other stuff on a Rpi zero...

Plantje1978, 2020-07-22:
> lol, one more time ^^
> 
> raspbee_gateway_config_2020-07-17 and deCONZ.tar are the same file.
> But I m suprised, because the file deCONZ.conf is created before the use of 7za.exe, so I don't understand why the file is not here.

Isn't the deCONZ.conf created while creating the .tar file using 7za.exe?

Plantje1978, 2020-07-22:
> Just out of curiosity: how can your current system not be powerful enough? I run deconz, my home automation software and lots of other stuff on a Rpi zero...

Quite a long answer to a simple question, but here we go :)
A long time ago, I wanted to ditch my cable box and start using my own hardware for viewing and recording live TV. Windows MediaCenter was the way to go back then. So, I installed a Windows machine.
A little less long time ago, I started my home automation and since I had a machine running 24/7 for my (nowadays) TV Mosaic and Kodi software, I started with Domoticz for Windows and I stuck with it.
When I wanted to read my P1 values (from my smart meters) I needed to have my home automation machine near my smart meters. So I took an old tablet I had lying around. That way, I had a machine with a physical interface. And that happens to be a tablet with an Intel Atom N450 and 2GB (oringally 1GB) of RAM. So, really nothing fancy.

SwoopX, 2020-07-22:
Thanks for highlighting. Then I guess it's time for some upgrade :wink:

Plantje1978, 2020-07-22:
It sure is!

Smanar, 2020-07-22:
> Isn't the deCONZ.conf created while creating the .tar file using 7za.exe?

Deconz create first the file and after use 7za.exe, so on your situation if the bug was from the use of 7za.exe, you will probably have the file, even not used.