dresden-elektronik/deconz-rest-plugin #3036: Device name

cllatser, 2020-07-13:
## Device

- Product name: SNZB-04
- Manufacturer: SONOFF
- Model identifier: DS01
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [X] Sensor Window / Door
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

derVicthor, 2022-12-29:
Device is never available in Phoscon for me. Pairing works, but I never get any device data and it shows up as unavailable from the start. What to do?

[image]