dresden-elektronik/deconz-rest-plugin #3037: Device name

cllatser, 2020-07-13:
## Device

- Product name: SNZB-02
- Manufacturer: SONOFF EWELINK
- Model identifier: TH01
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [X] Sensor Temperature and Humidity
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Screenshots
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

closed 2020-07-19 by SwoopX

Plaatjesdraaier, 2021-11-02:
Unfortunately, the Sonoff devices are only added in the app, but not visible in the visual overview.
Furthermore, they do not indicate any [image]

0nlytom, 2021-11-03:
It actually does work and connects properly. Can you please confirm the `Appilication Version` is 5? I managed to get mine to report data after manually setting the maximum reporting value to 600 second for the Temperature and humidity.

tiix95, 2021-11-07:
Same issue here. Can connect SNZB-02 correctly but Phoscon app shows "Status : Not reachable".
Same version as @Plaatjesdraaier but firmware 26660700

Smanar, 2021-11-09:
Have just buy one and have same issue.

After 3 tries it have finally worked, my procedure
- Long press (around 5s)
- Wait for 3 red blink
- press the button every 0.5s
- Force a read temperature in the GUI (not sure if it have an impact)

0nlytom, 2021-11-10:
The issue is that the reporting time interval is between 30 seconds and 65535 seconds. To get the sensors to increase the reporting frequency, the upper value needs to be changed. Please note, that once the device is pair, it maintains a connection with the coordinator for 30 to 60 seconds. All steps need to be complete before the sensor goes into sleep mode. 
You will need to have deCONZ GUI open for this and pair the device via deCONZ, which is under the `Control` tab. Have the `Cluster info` tab open. 
So, please read through everything first, as there is limited time to change the values.

1. Delete the old device
2. Set deCONZ into Pairing mode and switch to the `Cluster info` tab
2. Reset the sensor, hold the button down until the red LED flashes.
3. Wait for a new device to show up
4. Enable the dropdown menu by click the circle on the far right of the sensor. 

[image]

5. Click on the `Temperature measurement` and go to the `Cluster info` tab.
6. Double click on the `Meaured Value`, this will open a new window allowing you to edit the attributes 
7. Change to the `Max Report Interval` to 900. This is in seconds. 

[image]

8. Now click `write config`. Above the button it edit notify you if it was a success 
9. Do the same for the humidity.

Please note that the less the `Max Report Interval`, is sooner you will have to replace the battery

Smanar, 2021-11-10:
Ha yes it have worked after some tries.
Have used 30/3000/100 as setting on my side.

>The issue is that the reporting time interval is between 30 seconds and 65535 seconds

So it mean there is a problem, deconz need to set the reporting with thoses value for temperature
```
            rq.minInterval = 10;           // value used by Hue bridge
            rq.maxInterval = 300;          // value used by Hue bridge
            rq.reportableChange16bit = 20; // value used by Hue bridge
```

But we are not alone with issue on inclusion for this device ATM
https://github.com/Koenkk/zigbee2mqtt/issues/8971
https://github.com/Koenkk/zigbee2mqtt/issues/5562

Plaatjesdraaier, 2021-11-11:
Hi Onlytom,
Thank your for your clear contribution.
At first it seems to work, but unfortunately the values are not refreshed
[image]

Mimiix, 2021-11-11:
It probably takes a while

Plaatjesdraaier, 2021-11-11:
Everything else updates except the SNZB-02's  
[image]

It is also useful if the update frequency is not too low because, for example, I also use these sensors to activate the fan as soon as the humidity gets too high in the shower.
It is of course the intention here that the sensor detects the increased humidity as soon as the shower is switched on, not when I have finished showering

Mimiix, 2021-11-11:
What happens if you click the button on them?

Plaatjesdraaier, 2021-11-11:
I removed the sensors and added them again with the values:  min 30 /  max 600 / reportable 100 and now it works
(clicking the button did nothing)

Smanar, 2021-11-11:
Yes, have same result.

The device can be included easily , but no report at all (for me it leave the network).
But if you just set a reporting setting like said 0nlytom it can work immediatly.

I have tried to wake up the device to set report after the inclusion procedure, almost impossible, even pressing the button.

andreasgruener, 2021-11-18:
I followed the very precise and detailed description by 0nlytom. Perfect!

20design, 2021-11-23:
For me too, I followed the steps described by 0nlytom and its working fine. Just be fast to add the humidity sensor as well.
Thanks!

mgrn0, 2021-12-07:
adding two of the snzb-02 sensors with latest beta worked out of the box for me without ANY problems, no manual steps required.

camarril, 2021-12-10:
> adding two of the snzb-02 sensors with latest beta worked out of the box for me without ANY problems, no manual steps required.

Thank you!
I was setting up sizbee2mqtt for the fist time yesterday, using the latest stable version (1.22.1) and my SNZB-02 only sent battery and signal strength information once after pairing and then stopped working. No temperature or humidity. Today I tried the latest-dev image (published 11 hours ago) and now it works perfectly.

mgrn0, 2022-01-04:
very weird, both my snzb-02 sensors lost connection today and I cannot get them to re-connect at all. removing (once) and re-pairing does not work, deconz does not 'see' them anymore. updating to latest beta did not help, both sensors are not recognized. battery levels were 100%, using a new battery does not help, neither.

any ideas?

Mimiix, 2022-01-04:
@mgrn0 I suggest opening a forum post and add some logs.

nook24, 2022-01-10:
@mgrn0  I have the exact same issue with the exception that my setup is brand new :(

AndRe5575, 2022-02-09:
@0nlytom is there a way to do this without GUI? I installed deconz without because I also use openmediavault on my PI and therefore I had to use the lite OS version without GUI. Pairing my sonoff sensor just works for about one day and then it´s not found anymore so I have to pair them again.

0nlytom, 2022-02-10:
@AndRe5575 which version are you running? I know that in the latest version `2.13.04` everything work fine. Please try and upgrade.

0nlytom, 2022-02-11:
@AndRe5575 Please try and reconnect the sensors. I successfully re-provisioned one sensor on version `2.14.01`

AndRe5575, 2022-02-11:
@0nlytom I reconnected them. One is working at the moment and one shows no new update since a few hours. The one which is working shows sometimes that the last update was / will be in 1 hour. A bit strange.

Stubbs, 2022-08-18:
It appears that any devices added with an earlier version that had the default `Max Report Interval` set to 65535 still retain this value when re-added to the network, new devices work as expected.

I have tried updating the value while the device is awake immediately after adding to the network as described by @0nlytom however it seems to keep the incorrect value.

Can these devices be factory reset?

TheSOMGuy, 2022-09-22:
Is there an working solution for this problem bc i run in the same situation with a completely new setup. 
My door sensor is recognized as it should be but the Multisensor ( Temp/Humidity ) is not.
I can't use the gui bc of my headless openhabian installation and i have to solve this by hand or has to by new sensors.
And i do prefer the first one.

Any new suggestions?
Thanks

Smanar, 2022-09-27:
What is the issue ?
Have you tried to re-include the device without deleting it ?

BabaIsYou, 2022-09-27:
Can't it be included in a DDF that will be loaded at inclusion ?

Smanar, 2022-09-27:
Yeah can use a DDF to bypass the legacy code, but I don't see what we need.

>it should be but the Multisensor ( Temp/Humidity ) is not.

It's incomplete, bad values ....

BabaIsYou, 2022-09-27:
> @0nlytom I use Version `2.14.01` and a Raspbee II with FW `26720700`

Version 2.15.03 and ConBee II with FW 26720700on RPi4  and no issue with my 3 devices.

nickelnine, 2023-02-06:
Could someone kindly explain what is required to make these devices show values for temperature and humidity? I'm running deconz in Docker:
[image]
I've had the sensor for almost a week and I see only this:
[image]
and
[image]

nook24, 2023-02-06:
In case you are using a Raspberry Pi it is **important to connect the ConBee via an USB extension cord**.  That is very important.
Beside from this, start scanning for new sensors via the Phoscon Website and press the button on the Sonoff sensor to start pairing.

Also make sure that the battery in the Sensor has charge

That's basically it.

dennisb1, 2023-02-12:
> Could someone kindly explain what is required to make these devices show values for temperature and humidity? I'm running deconz in Docker: [image] )

Just a tought but maybe a good idea to do update of deCONZ first? At the moment it is 2.20.01 / 19-9-2022

Smanar, 2023-02-12:
Lol, ha yes lol, haven't see that, 2.13 ^^, I think it s a new record.

misel86, 2023-09-28:
> very weird, both my snzb-02 sensors lost connection today and I cannot get them to re-connect at all. removing (once) and re-pairing does not work, deconz does not 'see' them anymore. updating to [...]

Hello, did you manage to solve this? I have the same setup and the status is Not Reachable after multiple restart/pairing/etc

Smanar, 2023-09-28:
Do you have some logs ? or a capture from the GUI ?
What is your deconz version ? On the capture was the 2.13, so  a realy old one.

misel86, 2023-09-28:
Hello:

[image]

and I'm using DDF from  #6697 with this modification:

[image]

Smanar, 2023-09-28:
So same answer, update your deconz version, you have the 2.13 we are ATM at 2.23

misel86, 2023-09-28:
Oh, okay, looks like I'm using the old docker image. My bad, thanks :D

boelle, 2024-04-11:
i have the same issue and when i try to change the config from within deconz app i fails to write the config

[image]

[image]

even reading config fails

[image]

not that it matter i think but in zigbee2mqtt everything works except the addon stops a few times a week and faisls to restart on its own... i have to reboot the pi completely

Mimiix, 2024-04-11:
@Smanar  can you check the comment of @boelle ?

Smanar, 2024-04-11:
Hello,
What are you trying to do ?
Those device are slepper, so you can need to awake them in same time you press the "read" button, generaly with short press on the reset button.

Can you show the cluster list ? (with pressing the "+" on the node title.

boelle, 2024-04-11:
> What are you trying to do ?

add the device just like a light bulb or switch

> Can you show the cluster list ?

[image]

> And you have no report at all, no temperature, no humidity, no battery ?

correct

> Nothing in logs about this device in deconz/help/debug view, with flag "info" * 2 and "error" * 2 ?

will edit this post tomorrow, i have to switch from zigbee2mqtt to deconz each time i do testing. zigbee2mqtt works but crashes once a week, that is why i try to switch to deconz as i remember that as stable

boelle, 2024-04-12:
so i got one sensor added.... along with some other lights i have

[image]

if i blow air with my mouth it updates instantly

[image]

now to check if i'm also lucky with the other 4 i have

boelle, 2024-04-12:
@Smanar everything worked first time today, not sure why it was such a pain yesterday

Smanar, 2024-04-12:
Lol, zigbee is magic.
No seriously it's not normal, to add new sensor you don't need to use the "permit join" in deconz, just use phoscon, select "add new sensor" and make the procedure on the sensor.
Phoscon will enable itself the "permit join" for a limited period.

boelle, 2024-04-13:
> And you have other problem. Too much device with name like "0xXXXX" ....

those changed when i began to add my light bulbs in phoscon

and yes i would normal use phoscon to add the sensors but that did not worked so i tried with deconz instead, that with all 5 of them, the day before even decons did not work for the sensors

but yeah, very "strange" that it did not work one day and then the next it does

Smanar, 2024-04-13:
>EDIT: btw.... would it be in the near future to see an update function inside phoscon so we can easy update the firmware in devices?

Yes, for the moment you can use the desktop version or enable automatic update with "router" devices, but yes there is nothing for sensor out of deconz.

boelle, 2024-04-13:
i was thinking in phoscon, there are no section for updates

Smanar, 2024-04-13:
Not for sensor, but you can enable OTAU for router device in the settings (automatic working mode, no specific action possibles). And you need to put files in a specific folder.

boelle, 2024-04-13:
that is something i will miss from zigbee2mqtt, they have updates in one single screen for everything

but i guess a stable system that do not crash 1-2 times a week is better than easy updates :-D