dresden-elektronik/deconz-rest-plugin #3039: deCONZ sending multiple zigbee messages for On / Colour / Brightness

paolotremadio, 2020-07-13:
Hi,

I've noticed that deCONZ is sending multiple Zigbee messages for each property I set to a Light via the API. This is annoying because you can actually see the light applying each setting, one by one.

Here's an example:
The light is off; previous colour was red, brightness 10%. Let's assume I'm using Hue E27 as light.

- I set On = true, Bri = 254, Colour X/Y to blue. The light first turns on in red at 10%, then either sets the colour or the brightness, then it sets the remaining property (either colour or brightness).
- On hue, same light the light turns on with all the settings applied.

I'm assuming deCONZ and Hue bridge are sending Zigbee messages in different way.

Is there a reason why deCONZ is handling lights in different way? It is a really minor issue but it spoils the magic of smart lights. The transition is noticeable and I find it very annoying (especially when turning the lights on at night, from a previous setting of 100% to a new setting of 1%).

Do I have to change the API calls to first set bri/xy and then send the On request?

Smanar, 2020-07-13:
Good question.
From my memory it s not possible sending only one command for color and brightness in same time. It s not the same cluster, not the same command.
So I don't understand why you have a difference (I haven't the HUE hub to compare)

paolotremadio, 2020-07-13:
That’s what I thought as well. And yet hue does it... i was wondering if it is a bug on deCONZ. Because it wasn’t always like this for what I can remember :|

Smanar, 2020-07-13:
IDK perhaps a timing problem beetwen the 2 commands.
Do you have the json used ?

ebaauw, 2020-07-13:
deCONZ handles the light the same as the Hue bridge.  As @Smanar indicated, different Zigbee clusters require different commands.  Other than recalling a scene, there’s no way to combine these in a single Zigbee command.  You cannot set `bri` nor `xy` while a light is off.

If you combine the attributes in a single PUT, you shouldn’t be able to notice multiple commands being sent.  I would look for errors in the deCONZ log; you’re probably experiencing interference, causing the next command to be delayed.

paolotremadio, 2020-07-13:
> If you combine the attributes in a single PUT, you shouldn’t be able to notice multiple commands being sent. I would look for errors in the deCONZ log; you’re probably experiencing interference, causing the next command to be delayed.

Do I need a specific attribute to log the errors? What should I be looking for in the logs?

Smanar, 2020-07-13:
If you enable log you will have timestamp to check for delay.
But you can have a problem in the json you have used in the request, it s for that I m asking for it.