dresden-elektronik/deconz-rest-plugin #3040: Legrand Teleruptor

yves19, 2020-07-13:
## Device

- Product name: Legrand Drivia Connected Teleruptor  4 121 70
- Manufacturer: Legrand
- Model identifier: Teleruptor
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [x] Other: Switch

## Screenshots
Clusters
[image]

[image]

Basic Cluster Attributes
[image]

Legrand Specific Cluster
[image]

 Electrical Measurement Cluster
[image]

On-Off Cluster attributes
[image]

Smanar, 2020-07-13:
Yeah new legrand stuff, and again new feature ^^, I realy like netatmo.

I can make it tommorow, will be easy, just some questions
- I m seeing 0 at tension/current/power, but I think only the power is working ?
- Do you think the "button state entry" will be usefull ? For me not, I think knowing if the device is on or off is enought ?
- Can you compile the code on your side to test it ?
- Have you tried the device with the app, to test if it haven't special hidden feature ?

yves19, 2020-07-13:
_I m seeing 0 at tension/current/power, but I think only the power is working ?_
 Yes only power is measured. No Current, no Voltage are measured.

_Do you think the "button state entry" will be usefull ? For me not, I think knowing if the device is on or off is enought ?
The button state entry has no other function than network pairing or resetting. The LED attached to this button can be managed_ through the Specific Cluster Attributes (Option 2 and Option 3).

_Can` you compile the code on your side to test it ?_
Yes I can (when I Have time for this)

_Have you tried the device with the app, to test if it haven't special hidden feature ?_
I have tried the device with decONZ GUI on Win 10 as well as on Debian10. No hidden feature seen. I did not try yet the device with Legrand app

Smanar, 2020-07-14:
You can add wire a physical switch on the Teleruptor, was for this button I m asking (sorry have a bad english).

So only 2 lines to add ^^ (Legrand stuff is realy nice to add in deconz).
To compile it just change the tep 1 by
`git clone --branch Legrand-teleruptor https://github.com/Smanar/deconz-rest-plugin.git`

Smanar, 2020-07-17:
@yves19 Hello, have you make a try ?

yves19, 2020-07-17:
Not yet . Not at home. Probably next week end.
I keep in touch.

yves19, 2020-07-17:
> 
> 
> @yves19 Hello, have you make a try ?

Just tested this evening . It rocks :-)
Thanks.

closed 2020-07-17 by yves19

trungkstn, 2022-08-24:
Anyone test with Contactor (Legrand 412171)

Smanar, 2022-08-24:
The teleruptor is  4 121 70
The contactor is 4 121 71

Both are in the code, perhaps on another issue ? 
```

        sensor->modelId() == QLatin1String("Teleruptor") || //Legrand teleruptor
        sensor->modelId() == QLatin1String("Contactor") || //Legrand Contactor
```

Michel57189, 2024-04-21:
Hello,

I have bought the Legrand Contactor 412171. https://www.legrand.fr/pro/catalogue/contacteur-connecte-drivia-with-netatmo-multifonction-silencieux-1p-20ax-230v-1-module
It's not the same as 412170 Legrand Teleruptor.

I've tested it with deCONZ. It doesn't work as in the Legrand application.
This device should work according to HP/HC (in french) (off peak time / peak time in english) depending on the terminal C1/C2. So I have set up the parameter 0x0000 (option 1) with 0x0004 in FC01 cluster. But in this setting, I should have the possibility to force the switch off and the switch on even if it is peak time.
In deCONZ I only have ''on'' and ''off'' commands that are ineffective in this setting. (These commands are only effective when parameter option1 is 0x0003 in FC01 cluster, but in this case the terminal C1/C2 is ineffective)

See the following screenshots

Please, can you do something for me ...

[image]

[image]

[image]
[image]

[image]

[image]
[image]

Smanar, 2024-04-22:
Hello, yes, I think comment about this device are only on the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3053

But the working mode is normal, what is the problem ? You need to change the working mode "on the fly" ? You can't set one and keep it all the time ?

If it's that It's possible to increase the support for this device, this one is an older one, so need to create a DDF from scratch.
If you can share the "auto generated" one, from the editor it will help me, else I just need 5mn to create a new one, but it can be a little different that the one you already have.

If you want we can try to add a command "config/mode" to switch "auto"/"manu" ?

Do you know what is DDF, and what is your OS ?

Michel57189, 2024-04-22:
Hello,

Thank you for looking into this problem.

I’m using the Jeedom application with deCONZ plugin based on your REST API (I think). It is working on a RPI4 with Debian11.

Before using Jeedom/deCONZ, I was using the LEGRAND application. If you look at the screenshot below, yon can see 3 commands :
- ‘’Auto’’ : where C1/C2 terminal (HP/HC ) is effective
- ‘’Off’’ : to force the switch off
- ‘’Marche forcée’’ : translated as ‘’forced switch on’’ even if it’s peak time.
[image]

So, as you suggest, I think you can add a command "config/mode" to switch "auto"/"manu" ?
But, to answer your question I don’t know what is the DDF is and how to find it. Maybe you can tell me where I can find it …

Smanar, 2024-04-23:
On jeedom it's easy, except if you are using docker or VM.
Somes explanations here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

To make short I will give you a DDF, you just need to put it on one of the 2 possibles folders
- /usr/share/deCONZ/devices/
- /home/<DECONZUSER>/.local/share/ (I think here <DECONZUSER> is jeedom ?

To do this you can use all the tools you want, command line or application
https://community.jeedom.com/t/ddf-deconz/99220/18
https://community.jeedom.com/t/deconz-creation-dun-ddf/90540/9
It's possible too jeedom have a file explorator, but it seem users have right acces when using it.

Just install the DDF and restart the jeedom daemon

For the DDF (can use the name you want, with extension *.json)

```
{
  "schema": "devcap1.schema.json",
  "ddfvalidate": false,
  "manufacturername": "Legrand",
  "modelid": "Contactor",
  "product": "Contactor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 4, "manu" : 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/mode",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "eval": "if (Attr.val == 4) { Item.val = 'auto'; }  else if (Attr.val == 3) { Item.val = 'manu'; }",
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0000",
            "cl": "0xFC01",
            "dt": "0x09",
            "ep": "0x01",
            "eval": "if (Item.val == 'auto') { 4 } else if (Item.val == 'manu') { 3 }",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

If you have the GUI (but from my memory jeedom use headless OS) you can right clic on the node, then "edit DDF" to see if it work.
Else you can see direclty on the API using phoscon/help/API Information/Light
Or you can check the "donnée brutes" on jeedom. If it work you need to see the new field "config/mode".

For the moment it's the base, I will explain later how to use it (but you will need "créer des commandes" on jeedom, ca m'etonnerait que ce soit natif.
I have Legrand device too, not the same, but devices that use the same cluster, so if you have problem, I can make test on my side, but for me it's fine.

Michel57189, 2024-04-23:
Hello,

I have created the DDF file in /usr/share/deCONZ/devices/legrand/

I think this new DDF is considered but I'm not sure that it works as we want

Before this opération, I have these commands on Jeedom
[image]
and in  phoscon/help/API Information, i have :
"schema": "devcap1.schema.json",
    "manufacturername": "Legrand",
    "modelid": "Contactor",
    "status": "Draft",
    "subdevices": [
        {
            "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "state/on",
                    "description": "True when device is on; false when off.",
                    "refresh.interval": 5
                },
                {
                    "name": "state/alert",
                    "description": "The currently active alert effect.",
                    "default": "none"
                },
                {
                    "name": "state/reachable"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "dst.ep": 1,
[... 142 lines cut ...]
                    "at": "0x0000",
                    "dt": "0x10",
                    "min": 1,
                    "max": 300
                }
            ]
        }
    ]
}

So Config/mode is here But I don't know how to use it

I've tried to create an action command : 01-fc01.config::mode::auto
But then I have this error  : Erreur lors de la requete : 127.0.0.1:8484/api/DC1503E087/sensors//config(PUT), data : {"mode":"auto"} erreur : 3 => resource, /sensors/config, not available

And if I want to erase the divice and to include it I only have these commands :
[image]

I don't know if I made a mistake ???

For a few days I will not able to make another tests. But as soon I can do it, I will answer here to continue 

Thank You

Michel

Smanar, 2024-04-24:
>I don't know if I made a mistake ???

No, you are perfect. The DDF is installed, you know how to make request, perefect.

I haven't set the part for the power measurement yet, my DDF is lighter for the moment, I will add it on next one.

`>erreur : 3 => resource, /sensors/config, not available`
Can be my fault, you have too much "/" in your request 
`Erreur lors de la requete : 127.0.0.1:8484/api/DC1503E087/sensors//config(PUT), data : {"mode":"auto"} `
But it don't seem to have impact.

For the moment I m trying the request on my side, and it seem the "write" part is not working, I m asking to other devs what I m doing wrong, I hope having a better DDF tommorow.

Michel57189, 2024-04-24:
Hello
Thank you for your time
If you find the solution to the problem, I Will do new test as soon as possible

Smanar, 2024-04-25:
So I have a good and a bad new.
The good one is the bug is found and Manup have already corrected it https://github.com/dresden-elektronik/deconz-lib/commit/d82fac9f55e9656b28c96bbf0d97c5d4ad4356a5
The bad one is this bug was in deconz core, so hard for you to have it immediatly, need to re compile code, I m asking if I can compile a part and give you the result, but I think the only way is recompiling all the application, and not something easy on a production machine.
So we will probably need the next version to finish the DDF.

Have added the consumption part, only power is working on Legrand stuff.

```
{
  "schema": "devcap1.schema.json",
  "ddfvalidate": false,
  "manufacturername": "Legrand",
  "modelid": "Contactor",
  "product": "Contactor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 4, "manu" : 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/mode",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "eval": "if (Attr.val == 4) { Item.val = 'auto'; }  else if (Attr.val == 3) { Item.val = 'manu'; }",
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0000",
            "cl": "0xFC01",
            "dt": "0x09",
            "ep": "0x01",
            "eval": "if (Item.val == 'auto') { 4 } else if (Item.val == 'manu') { 3 };",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0001",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 800,
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 5,
          "max": 300,
          "change": "0x00000002"
        }
      ]
    }
  ]
}
```

I will ping you on this issue when I will have more information to finish the DDF.

Michel57189, 2024-04-25:
Very good job
Tanks tu you
So I am waiting for your ping

Smanar, 2024-05-16:
@Michel57189 There is a new deconz version, the beta 0.27.
And this one work better for us.

I have another problem for the moment, I m still on it. Previously only sensor had the "config" field, from my mind this rule was removed, for exemple to use the "power on state" for light, but I haven't succed on my side, so for your test Have put the "config/mode" on the sensor device, the ZHApower.
It's still a little hacky but it work on my side for the cable outlet.

```

{
  "schema": "devcap1.schema.json",
  "ddfvalidate": false,
  "manufacturername": "Legrand",
  "modelid": "Contactor",
  "product": "Contactor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0001",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "meta": {
        "values": {
          "config/mode": {"auto": 4, "manu" : 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/mode",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0xFC01",
            "ep": "0x01",
            "eval": "if (Attr.val == 4) { Item.val = 'auto'; }  else if (Attr.val == 3) { Item.val = 'manu'; }",
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0000",
            "cl": "0xFC01",
            "dt": "0x09",
            "ep": "0x01",
            "eval": "if (Item.val == 'auto') { 4 } else if (Item.val == 'manu') { 3 };",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/power",
          "refresh.interval": 800,
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 5,
          "max": 300,
          "change": "0x00000002"
        }
      ]
    }
  ]
}
```

Smanar, 2024-05-16:
And about the new version, I know you are on jeedom that is an easier OS to use, but perhaps the "DDF Bundle Support" part can interest you too ? 

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.27.0-beta

We are searching tester ^^

Michel57189, 2024-05-16:
Hello

First, thank you for not forgetting my topic

So, it's not easy for me to test instanly because the deCONZ updates must first be installed by the jeedom team and I don't really know how to do with beta versions. But I know there are solutions. I will lookfor or ask on jeedom community.

One question please to be sure to understand what you mean : if the ''config/mode'' commands are not outomatically, will I have to create them on FC01 cluster or on another ?

Michel57189, 2024-05-16:
Regarding your request about "DDF Bundle Support", I'm not sure to be the right person. But I'm going to read and understand what it is. And if you really want it I would like to try.

Smanar, 2024-05-17:
>But I'm going to read and understand what it is. And if you really want it I would like to try.

It's an easier way to use DDF, without file manipulation. But you will need too the version 0.27 version.

>One question please to be sure to understand what you mean : if the ''config/mode'' commands are not outomatically, will I have to create them on FC01 cluster or on another ?

Here I don't understand the question, I have adapted the previous DDF, you will have a "config/mode" command on the ZHAPower sensor, and you can use it to change the mode "on the fly".

I can't help you for jeedom but for the rest I m on discord with the same name, and I can speak french.