dresden-elektronik/deconz-rest-plugin #3042: _TYZB02_keyjqthh TS0041

bsfaxi, 2020-07-14:
Hi,
I have bought the following Smart Switch and Remote/Wireless Switch that normally work with a Xiaomi GW.
Smart switch : https://fr.aliexpress.com/item/10000008539597.html?spm=a2g0s.9042311.0.0.76326c37upJGCV
Wireless switch : https://fr.aliexpress.com/item/10000065424854.html?spm=a2g0s.9042311.0.0.27426c37Xn8EwX
I have succeded to pair them with Phoscon using a Raspbee. The Smart Switch works perfectly and is recognized as a light. So, it's okay to turn on/off the light. However, the Wireless switch is also recognized as a light, and thus cannot be used as a scene button.
So, is it possible to add the support of this device please?
Below are the required information. Are there any further details that are needed?
[image]
[image]

Thanks in advance for your precious help.

Smanar, 2020-07-15:
I m sorry for you, but I think you have one of this boring tuya device.
I think too you miss a cluster in your capture, but not your fault, it from the device (there is another device with the same problem), can you check (on deconz) if on the wired one you haven't a cluster EF00 ?

bsfaxi, 2020-07-16:
> I m sorry for you, but I think you have one of this boring tuya device.
> I think too you miss a cluster in your capture, but not your fault, it from the device (there is another device with the same problem), can you check (on deconz) if on the wired one you haven't a cluster EF00 ?
> 
> BTW, you haven't something to sniff zigbee device ? I m searching why it s the second device with this problem.

Hi Smanar,
Thanks for time to get an eye on this. I'm quite newbee to this. :)
Indeed it's a boring Tuya device, but with cool design! :)
I just need to recognize this wireless switch as a scene button (1 press, double press or a long press and release), or as a 2 ways switch with an automation. So, I can't if it's recognized as a light.
Please find below other screenshots including the smart switch (the wired one) and the wireless one.
What other info do you need?

[image]
[image]

Thanks.

Smanar, 2020-07-16:
For the moment I need help, but a solution ^^.
To make it clear, I m sure you miss a cluster, first because I have other device with same bug, and second because you only have a on/off cluster, so only with this cluster you can't make simple/double/long click.

I can add your device in whitelist for you try the tuya branch, but I already have some problem on other tuya device.

BTW I need the manufacture code, you can find it on "Node info" panel.

bsfaxi, 2020-07-16:
Here you go.

For the Wireless :
[image]

For the Wired :
[image]

Smanar, 2020-07-16:
Ok so, you try the tuya code.

On the compilation procedure, just chnage step 1 by

`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

Don't be optimistic, nothing magic, just you device will appear in API, and you will see some debug line when you use it.

Smanar, 2020-07-19:
Hello, have you tried your device with last code ?
If you can include it, I need you run deconz in debug mode and debug line starting with "Tuya debug 4:" that will apper when you use it.

Smanar, 2020-07-22:
We have finished the first tuya device https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3024
So if you can try the same code and check for "Tuya debug 4" in log when using the switch ?

bsfaxi, 2020-07-22:
Hi Smanar,
Sorry for the late response.
I'll give a try this weekend, and get back to you with feedback.
Thanks a lot again. :)

Smanar, 2020-07-31:
Have you got some time to make tries ?

cobirnm, 2020-09-30:
I have one of these switches. How can I help?
At the moment I can add it but it is added as a light. This should be a switch.

Smanar, 2020-09-30:
Sure, it s possible if your device have exactly the same manufacture name ?

But ATM the tuya code is in waiting list, so not a lot change ATM.

Wich one version are you using ? Do you know how to compile the code to make tries on your side ? have you a real Unix machine ?

cobirnm, 2020-09-30:
> Sure, it s possible if your device have exactly the same manufacture name ?
> 
> But ATM the tuya code is in waiting list, so not a lot change ATM.
> 
> Wich one version are you using ? Do you know how to compile the code to make tries on your side ? have you a real Unix machine ?

Well to answer you I have a raspberry pi, and probably no i can't copile the code.
It's a pity because tuya devices have great quality/price!

cobirnm, 2020-09-30:
> Sure, it s possible if your device have exactly the same manufacture name ?
> 
> But ATM the tuya code is in waiting list, so not a lot change ATM.
> 
> Wich one version are you using ? Do you know how to compile the code to make tries on your side ? have you a real Unix machine ?

Well to answer you I have a raspberry pi, and probably no i can't copile the code.
It's a pity because tuya devices have great quality/price!

Regarding the device yes it's exacly the same manufacture name. Regarding deconz I'm using version 2.05.82

Smanar, 2020-10-01:
A raspberry is good, but I suppose you have Hassio ?

bsfaxi, 2020-10-02:
Hello again,
I finally got a CC2531, which I patched with the ZBoss sniffer firmware.
In Wireshark, I can see the following command "0xfd" :
[image]
I hope it helps.

Smanar, 2020-10-02:
We already have this command now > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/button_maps.json#L573

The code have moved since the last time, and the tuya PR is validated.

Give me time to update with last deconz version, I will add the switch again for test.

bsfaxi, 2020-10-02:
Cool. :-)
I can see Tuya3gangMap.
Is it OK to handle also the 2 other Tuya Wireless Switches with 1 and 2 gangs?

Smanar, 2020-10-02:
Yep I think the same code will work for all gangs, but there is a problem.
The vendor manu number is on white list so the device is reconized as light instead of sensor ...

First try
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

This code will prevent the device will be visible as light.

cobirnm, 2020-10-03:
> A raspberry is good, but I suppose you have Hassio ?
> 
> And yes there is so much tuya device every week, realy active manufacture.

No I'm on Raspbian

cobirnm, 2020-10-03:
@Smanar , my model is TS0012 (2 gang) do you think your code will work for me?

Smanar, 2020-10-03:
@cobirnm 
I need to add it in the whitelist, I need
- manufacture number
- manufacture name
- the mac adresss (the 6 firsts number)
- if the device is reconized as router or not.

But yes I think it will use the same code

bsfaxi, 2020-10-03:
I compiled the latest code successfully.
The wireless switch appears in Deconz, but doesn't show up in the Phoscon App, neither in lights and in switches.

bsfaxi, 2020-10-03:
I have also the Wireless switch with 3 gangs.
I'll test it, and provide details.

Smanar, 2020-10-03:
Yep can be normal it will be invisible on phoscon.
Need to check in the api or in your third app.

bsfaxi, 2020-10-03:
In Home Assistant, the device appears in the integration, but not the entity switch which can be controlled.

[image]

[image]

bsfaxi, 2020-10-03:
For the Wireless 2 gangs, I have the following:

[image]

[image]

bsfaxi, 2020-10-03:
For the Wireless 2 gangs, it doesn't even appear in Home Assistant. It only appears in Deconz.

bsfaxi, 2020-10-03:
For the Wireless 3 gangs, I have the following:

[image]

[image]

bsfaxi, 2020-10-03:
Same thing for this Wireless switch: it doesn't even appear in Home Assistant. It only appears in Deconz.

Smanar, 2020-10-03:
They are all wireless ?

I have added the 2 other devices too > https://github.com/Smanar/deconz-rest-plugin/commit/ae77bcacaa413bbefdf5e8899f4d4a10d6e08e62

But you need to delete the one you can see, he is in "light" section, the good one will be in "sensor" section, you need to re - include them too.

bsfaxi, 2020-10-03:
It's recognized by the Phoscon App as a Sensor, but doesn't appear in the list of Sensors.
In Home Assistant, the device appears but not the sensor entity.

cobirnm, 2020-10-04:
> yes I think it will
Please find the details below:

[image]

[image]
[image]
[image]
[image]
[image]

Smanar, 2020-10-04:
Have added the @cobirnm switch, if you can try too ?

>It's recognized by the Phoscon App as a Sensor, but doesn't appear in the list of Sensors.
In Home Assistant, the device appears but not the sensor entity.

He is still in "light" section ?
Can you show the device JSON ? you can find it in your third app or using an url like this `http://IP:PORT/api/API_KEY/sensors`

IP and PORT > same than for phoscon.

bsfaxi, 2020-10-04:
> He is still in "light" section ?
>>
Not in lights either. Nowhere in Phoscon app. Only in Deconz.

> It mean you have a success message during pairing ?
>>
Yes, the Phoscon App show success paring with the device as a sensor. But then, it doesn't show up in the list.

> Can you show the device JSON ? you can find it in your third app or using an url like this `http://IP:PORT/api/API_KEY/sensors`
>>
{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"42d221e47b288d20c94bc52f251bb1ef","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2020-10-04T10:24:38.042","status":160,"sunrise":"2020-10-04T05:56:01","sunset":"2020-10-04T17:24:54"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:05:31:b7-01"},
"2":{"config":{"battery":null,"on":true,"reachable":false},"ep":1,"etag":"42d221e47b288d20c94bc52f251bb1ef","lastseen":null,"manufacturername":"_TYZB02_keyjqthh","mode":1,"modelid":"TS0041","name":"TS0041","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"}}

Smanar, 2020-10-04:
He is here ^^

`"2":{"config":{"battery":null,"on":true,"reachable":false},"ep":1,"etag":"42d221e47b288d20c94bc52f251bb1ef","lastseen":null,"manufacturername":"_TYZB02_keyjqthh","mode":1,"modelid":"TS0041","name":"TS0041","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"}}`

Try to use it and look if the field "buttonevent":null change.

bsfaxi, 2020-10-04:
Yep, I saw it. :)

The field buttonevent doesn't change at all.
In HomeAssistant, the sensor entity for this device doesn't show up.

Smanar, 2020-10-04:
I have updated the code to make a bind/attribute reporting for this device, if you can make another try ?

But I don't understand for HA, you have checked in all section ? Don't need a "synchronisation" ?

bsfaxi, 2020-10-04:
I will give it another try.

In HA, the update is automatically done. I even reinstalled the integration to be sure.

bsfaxi, 2020-10-04:
A little better. :) 
Now, I have the battery level. But, I still have this in the API {"buttonevent":null,"lastupdated":"none"}.

[image]

Smanar, 2020-10-04:
Ha yep haven't added the battery yet.
But it s strange you haven't value ...

Do you know how to enable deconz log on HA ?
When you press the button you will have some debug line, for exemple

>[INFO] - No button handler for: ..............

bsfaxi, 2020-10-04:
Yes, I have no value for the battery.

I have added this to the configuration.yaml of HA:
logger:
  default: info
  logs:
    pydeconz: debug
    homeassistant.components.deconz: debug

This is an extract from the logs concerning the device TS0041:
...
             '2': {'config': {'battery': 0, 'on': True, 'reachable': True},
                   'ep': 1,
                   'etag': '4b38ddc6e1fe3ef55b99aacbfbc54196',
                   'lastseen': '2020-10-04T13:24Z',
                   'manufacturername': '_TYZB02_keyjqthh',
                   'mode': 1,
                   'modelid': 'TS0041',
                   'name': 'TS0041',
                   'state': {'buttonevent': None, 'lastupdated': 'none'},
                   'type': 'ZHASwitch',
                   'uniqueid': 'ec:1b:bd:ff:fe:76:d7:64-01-0006'}}}

2020-10-04 17:33:54 DEBUG (MainThread) [pydeconz.deconzdevice] TS0041 created as
{'config': {'battery': 0, 'on': True, 'reachable': True},
 'ep': 1,
 'etag': '4b38ddc6e1fe3ef55b99aacbfbc54196',
 'lastseen': '2020-10-04T13:24Z',
 'manufacturername': '_TYZB02_keyjqthh',
 'mode': 1,
 'modelid': 'TS0041',
 'name': 'TS0041',
 'state': {'buttonevent': None, 'lastupdated': 'none'},
 'type': 'ZHASwitch',
 'uniqueid': 'ec:1b:bd:ff:fe:76:d7:64-01-0006'}

No events are logged when I press the button.

Smanar, 2020-10-04:
Don't make mistake beetween deconz log and the HA plugin log.
A deconz log look like 

> 9:43:24:977 06 unsupported index (data type 0x20)
> 19:43:32:720 0x00158D000171E9E0 error APSDE-DATA.confirm: 0xF0 on task
> 19:43:32:720 APS-DATA.confirm id: 76 status: transaction expired (8 seconds)
> 19:45:48:271 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xE9 on task
> 19:45:56:693 button 1002 Normal release
> 19:45:56:695 ZCL attribute report 0x00158D00018B4A84 for cluster 0x0006, ep 0x01
> 19:45:57:207 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
> 19:45:59:300 0x00178801021D61FE force poll (2)
> 19:46:00:357 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xA7 on task

bsfaxi, 2020-10-05:
Here you go:

In the beginning of the file:
11:43:48:885 verify 0xec1bbdfffe76d764 is child node after 809128 s
11:43:48:885 Node data 0xec1bbdfffe76d764 profileId: 0x0104, clusterId: 0x0000
11:43:48:885 SensorNode id: 2 (TS0041) available
11:43:48:885 Update Sensor 0xEC1BBDFFFE76D764 Basic Cluster
**11:43:48:885 [INFO] - No button map for: TS0041 endpoint: 0x01 cluster: 0x0000 command: 0x0A payload[0]: 001**
11:43:48:885 ZCL attribute report 0xEC1BBDFFFE76D764 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000
11:43:48:885    payload: 01002044
11:43:48:887 Websocket 192.168.0.29:51827 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"2","r":"sensors","t":"event","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"} (ret = 5695152)
11:43:48:888 Websocket 192.168.0.79:44786 send message: {"config":{"on":true,"reachable":true},"e":"changed","id":"2","r":"sensors","t":"event","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"} (ret = 5695152)

A little further, when clicked on the button:
11:44:03:818 Node 0xEC1BBDFFFE79B4C8 is known by 0 neighbors, last seen 0 s
11:44:04:154 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
11:44:04:155 Wait 5s till query finished
11:44:04:298 Node 0xBC33ACFFFE17054E is known by 1 neighbors, last seen 6 s
11:44:04:417 MAC Poll 0x02 0x1A8F
11:44:04:444 APS-DATA.indication from child 0x1A8F
**11:44:04:444 [INFO] - No button map for: TS0041 endpoint: 0x01 cluster: 0x0006 command: 0xFD payload[0]: 000**
11:44:04:557 HTTP client hdr detected:
GET /api/406/config/wifi/ HTTP/1.1^M
Host: 127.0.0.1^M
User-Agent: curl/7.64.0^M
Accept: */*^M

Smanar, 2020-10-05:
Ok I think I have understand, on last (and future version), you need to have a file called button_maps.json.
This file is editable with a text editor (json format) and contain all values for switchs.

So to test last versions you need to copy this file https://github.com/Smanar/deconz-rest-plugin/blob/tuya/button_maps.json in a deconz folder (/usr/share/deCONZ/devices/button_maps.json)

> 
> Outsource button maps to external JSON file @SwoopX (#3127):
> All available button maps have been outsourced to an external JSON file named button_maps.json, which must reside in the same directory as the deCONZ database zll.db. The purpose is to speed up integration of new switches as users can now create it theirselves (given that some appropriate guidance is provided).

bsfaxi, 2020-10-06:
Hi @Smanar ,
I think there still a problem with that file.
Here what I have in the log in my last test 10mn ago (I don't know why, in the log, the timestamp starts with my local time CEST and then switches to GMT):
...
08:47:38:275 [INFO] - Found file containing button maps. Parsing data...
08:47:38:280 [ERROR] - Error: illegal value at offset: 6 (in characters)
08:47:38:280 [ERROR] - No object named 'buttons' found in JSON file. Skip to load button maps.
...

bsfaxi, 2020-10-06:
it's better, Now I have this:
12:57:40:038 [INFO] - Found file containing button maps. Parsing data...
12:57:40:041 [ERROR] - Button map 'instaRemoteMap' in JSON file has no assigned ModelIDs. Skip loading button map...
12:57:40:041 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
**12:57:40:047 [INFO] - Button maps loaded.**
...
**12:59:29:080 {"2":{"name":"TS0041"},"lastscan":"active"}**
12:59:29:267 Node 0xBC33ACFFFE17054E is known by 1 neighbors, last seen 38 s
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0000
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0001
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0002
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0003
12:59:29:430 ZCL cmd-req nwk: 0xEED7, profile: 0x0104, cluster: 0x0000 cmd: 0x00
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0004
12:59:29:430 ZCL cmd-req nwk: 0xEED7, profile: 0x0104, cluster: 0x0000 cmd: 0x00
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0005
12:59:29:430 ZCL cmd-req nwk: 0xEED7, profile: 0x0104, cluster: 0x0000 cmd: 0x00
12:59:29:430 ZCL read cluster: 0x0000, attribute: 0x0006
12:59:29:430 ZCL cmd-req nwk: 0xEED7, profile: 0x0104, cluster: 0x0000 cmd: 0x00
12:59:29:431 ZCL read cluster: 0x0000, attribute: 0x0007
12:59:29:431 ZCL read cluster: 0x0000, attribute: 0x0008
12:59:29:431 ZCL read cluster: 0x0000, attribute: 0x0009
12:59:29:431 ZCL read cluster: 0x0000, attribute: 0x000A
12:59:29:431 ZCL cmd-req nwk: 0xEED7, profile: 0x0104, cluster: 0x0000 cmd: 0x00
...
**12:59:34:703 [INFO] - Button 1002 B1 short**
12:59:34:758 Websocket 192.168.0.29:63775 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-10-06T12:59:34.703"},"t":"event","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"} (ret = -1091027600)
12:59:34:760 discard sensor state push for 2: state/lastupdated (already pushed)
...
**12:59:41:776 [INFO] - Button 1004 B1 double**
12:59:41:778 Websocket 192.168.0.29:63775 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2020-10-06T12:59:41.775"},"t":"event","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"} (ret = -1091027600)
12:59:41:779 discard sensor state push for 2: state/lastupdated (already pushed)
...
12:59:43:210 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('2', 'TS0041', 'ZHASwitch', 'TS0041', '_TYZB02_keyjqthh', 'ec:1b:bd:ff:fe:76:d7:64-01-0006', '', '{"buttonevent":1004,"lastupdated":"2020-10-06T12:59:41.775"}', '{"battery":null,"on":true,"reachable":true}', '{"d":0,"ep":1,"in":[0,1,6],"p":260}', 'normal', '1')
...

So, it seems to be taken into account.
However, only the device info is present in Home Assistant not the entity, even with doing the deconz integration from scratch.
[image]

bsfaxi, 2020-10-06:
when I do curl http://192.168.0.79/api/79971660D3/sensors, I can see the 3 button events 1002 (single click), 1004 (double click) and 1003 (long press).
{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"98b56b3ab1212f881fbdfef20ec3c66d","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2020-10-06T13:16:41.530","status":170,"sunrise":"2020-10-06T05:58:58","sunset":"2020-10-06T17:20:44"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:05:31:b7-01"},"2":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"709e55c52a891b0f14c67c45d012366b","lastseen":"2020-10-06T13:36Z","manufacturername":"_TYZB02_keyjqthh","mode":1,"modelid":"TS0041","name":"TS0041","state":{"buttonevent":1003,"lastupdated":"2020-10-06T13:36:21.943"},"type":"ZHASwitch","uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"}}

cobirnm, 2020-10-06:
on quick question. I have passed the last 2 days trying to compile the branch in to a docker image. (whitout success). Are you using docker to test the image or testing it on the "normal" system?

I'm trying not to mess my configuration.

Smanar, 2020-10-06:
Hello

@bsfaxi  what you have done to solve
```
08:47:38:275 [INFO] - Found file containing button maps. Parsing data...
08:47:38:280 [ERROR] - Error: illegal value at offset: 6 (in characters)
08:47:38:280 [ERROR] - No object named 'buttons' found in JSON file. Skip to load button maps.
```

ATM I m locked, the compilation don't work on my machine, so I can make modification , but impossible for me to test them.

For the device that is badly displayed in HA, I think it s better to ask in HA plugin forum/github with showing the JSON

```
"2":{
   "config":{
      "battery":null,
      "on":true,
      "reachable":true
   },
   "ep":1,
   "etag":"709e55c52a891b0f14c67c45d012366b",
   "lastseen":"2020-10-06T13:36Z",
   "manufacturername":"_TYZB02_keyjqthh",
   "mode":1,
   "modelid":"TS0041",
   "name":"TS0041",
   "state":{
      "buttonevent":1003,
      "lastupdated":"2020-10-06T13:36:21.943"
   },
   "type":"ZHASwitch",
   "uniqueid":"ec:1b:bd:ff:fe:76:d7:64-01-0006"
}
```

@cobirnm you need a normal OS, It s possible on a VM with a full OS, but I realy don't know if it s possible on docker.

Next step the battery.

bsfaxi, 2020-10-06:
> One quick question. I have passed the last 2 days trying to compile the branch in to a docker image. (whitout success). Are you using docker to test the image or testing it on the "normal" system?
> 
> I'm trying not to mess my configuration.

Hi,
For my production, I'm using HassOS on Raspberry pie 4. It's using Docker underneath...
I searched a while, but it's not obvious to do all the dev staff with this configuration.

But, for the DEV (the same Raspberry with another SD card), I installed Raspbian and then the Deconz and Home Assistant packages to be able to build and test, and also to patch the Raspbee ii with the latest firmware.

bsfaxi, 2020-10-06:
> what you have done to solve
I simply was downloading the json with wget. :)
And then, I saw that the file exists already in the deconz-rest-plugin directory which was downloaded with the "git clone" command.

> ATM I m locked, the compilation don't work on my machine, so I can make modification , but impossible for me to test them.
Let me know when the code is modified to make the required tests. Let's go for the Battery... :)

bsfaxi, 2020-10-06:
> For the device that is badly displayed in HA, I think it s better to ask in HA plugin forum/github with showing the JSON

I will open an issue for that.
But, do you know why it's not displayed also in the Phoscon App as a sensor?

Smanar, 2020-10-06:
Lot of devices are not displayed in phoscon, this app use white/blacklist, for phoscon Im  not surprised, but for HA I don't understand `"type":"ZHASwitch",` most of third app just use that, and the value is correct.

BTW for battery I have forget something can you take a look on deconz cluster 0x0001 "power configuration" wich one attribute is used for battery ? 0x0020 or 0x0021 ?

bsfaxi, 2020-10-07:
> Ok , i have found the capture on previous posts. But "battery percent remain" = 200 ??? WTF

Maybe it's normal, since the device has 2 batteries.

> Hu, I have some problem with battery ^^, it seem all the code is already in place, you don't have something about cluster 0x0001 in your log ?

For @cobirnm the % is at 200%. For me, it's always at 0% !
I will try with another one.

I can see these lines in the code. No errors.
08:25:28:842 0xEC1BBDFFFE76D764: added ZCL value 0x01/0x0001/0x0021
...
08:26:54:078 ZCL got data for node=0xC1E4, cl=0x0001, at=0x0021, status=0x00, type=0x20
08:26:54:079 Node data 0xec1bbdfffe76d764 profileId: 0x0104, clusterId: 0x0001
08:26:54:079 0xEC1BBDFFFE76D764: update ZCL value 0x01/0x0001/0x0021 after 0 s
08:26:54:120 discard sensor config push for config/battery (already pushed)
...
In Home Assistant too:
[image]

bsfaxi, 2020-10-07:
I have tested with another wireless switch which has the same manufacturer name as @cobirnm's one: "_TZ3400_keyjqthh".

In deCONZ, the value is 200%.
[image]

In Home Assistant and APIs, the value is correct: 100%.
"3":{"config":{"**battery":100**,"on":true,"reachable":true},"ep":1,"etag":"1785727fefac4f06861bce9f00d40f4a","lastseen":"2020-10-07T09:29Z","manufacturername":"_TZ3400_keyjqthh","mode":1,"modelid":"TS0041","name":"TS0041 (2)","state":{"buttonevent":1004,"lastupdated":"2020-10-07T09:29:40.286"},"type":"ZHASwitch","uniqueid":"ec:1b:bd:ff:fe:87:1e:04-01-0006"}}
[image]

bsfaxi, 2020-10-07:
FYI, I have created an issue for the "Button Event" entity not been created in HA here https://github.com/home-assistant/core/issues/41405.

Smanar, 2020-10-07:
Nice, I m curious for HA, I don't know at all if they use white/blacklist (I think not)

For the battery still a mystery, I have compared the working device and the not working, exactly the same code, I don't see why 1 is working and not the other ....

And the code receive value
>08:26:54:078 ZCL got data for node=0xC1E4, cl=0x0001, at=0x0021, status=0x00, type=0x20

For the 200 value I think now it s normal, the code make a division by 2 with the value.

bsfaxi, 2020-10-08:
@Smanar 
I've done the test again. Not found the line with "Battery debug X"!
With the new code, I don't see anymore the button events...

Concerning HA, the integration works perfectly. I was not searching in the right place. The button generates an event and don't create an entity! :)

Smanar, 2020-10-08:
You have only compiled the code (no change on json file ?)
Because the only change there is was some debug line added > https://github.com/Smanar/deconz-rest-plugin/commit/61ad1374c8e8bdfa4321e53dfb50d3b1ba97da12
Can't change the working mode.

Check to see if you are on the good branch

```
cd deconz-rest-plugin
git branch
```

bsfaxi, 2020-10-09:
No change on json file.
I only recompiled the your previous commit:
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
git checkout -b bilel f0fc8afe54a8863ef62f07332c2a1cfbfcc3950d
With this plugin version, the button events worked, not with your last change.
I will retry later if you want.

> cd deconz-rest-plugin
> git branch
* bilel
  tuya

[64 more replies]

closed 2020-11-15 by manup