dresden-elektronik/deconz-rest-plugin #3044: IKEA Blind Stopped Working - Now Firmware Update Killed ConBeeII

Puntoboy, 2020-07-14:
## Describe the bug

In the last couple of days, by IKEA blind has stopped responding.

I have factory reset the blind, re-paired it to Deconz, and recharged the battery. Nothing has worked. I then tried updating Deconz to the latest but that hasn't helped either.

I can see the device in VNC and on the web app. but if I try to adjust it from the web app or from home assistant nothing happens. On Home Assistant I get a "Failed to call service cover/close_cover. /lights/2 resource, /lights/2, not available" and on Node-Red I get an API error.

I'm running my system in Docker. It has been working fine for a while.

If I manually call the "cover" service in HA, I see this error in the logs:

[image]

I also see this in the logs constantly.

[image]

Also, in VNC, the device is sitting with a green flashing indicator. All other devices flash blue.

[image]

How can I troubleshoot please? The logs for the add-on don't really show much..

I have tried to update the firmware on the ConbeeII but now Deconz won't connect to it.

## Steps to reproduce the behaviour

Adjust brightness of IKEA blind in Phoscon App - nothing happens

## Expected behavior

Blind should open or close.

## Screenshots

[image]

[image]

[image]

## Environment

 - Host system:  NAS
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 
 - deCONZ version: 2.05.78
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

`12:53:37:454 wait reconnect 10 seconds
12:53:38:454 wait reconnect 9 seconds
12:53:39:454 wait reconnect 8 seconds
12:53:40:454 failed to reconnect to network try=2
12:53:40:454 wait reconnect 7 seconds
12:53:41:454 wait reconnect 6 seconds
12:53:42:454 wait reconnect 5 seconds
12:53:43:454 wait reconnect 4 seconds
12:53:44:454 wait reconnect 3 seconds
12:53:45:454 failed to reconnect to network try=3
12:53:45:454 wait reconnect 2 seconds
12:53:46:454 wait reconnect 1 seconds
12:53:46:651 COM: /dev/ttyACM0 / serialno: DE2129434
12:53:46:651 COM: --dev: /dev/ttyACM0 (ConBee II)
12:53:47:282 device state timeout ignored in state 2
12:53:48:333 device state timeout ignored in state 2
12:53:49:383 device state timeout ignored in state 2
12:53:50:433 device state timeout ignored in state 2
12:53:50:454 try to reconnect to network try=4
12:53:51:954 device state timeout ignored in state 2
12:53:52:954 device state timeout (handled)
12:53:52:993 device disconnected reason: 1, index: 0
12:53:53:454 wait reconnect 15 seconds
12:53:53:454 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
12:53:53:454 device state timeout ignored in state 1
12:53:54:454 wait reconnect 14 seconds
12:53:55:454 failed to reconnect to network try=5
12:53:55:454 wait reconnect 13 seconds
12:53:56:454 wait reconnect 12 seconds
12:53:57:453 wait reconnect 11 seconds
12:53:58:454 wait reconnect 10 seconds
12:53:59:454 wait reconnect 9 seconds
12:54:00:454 failed to reconnect to network try=6
12:54:00:454 wait reconnect 8 seconds
12:54:01:454 wait reconnect 7 seconds
12:54:02:453 wait reconnect 6 seconds
12:54:03:453 wait reconnect 5 seconds
12:54:04:454 wait reconnect 4 seconds
12:54:05:454 failed to reconnect to network try=7
12:54:05:454 wait reconnect 3 seconds
12:54:06:453 wait reconnect 2 seconds
12:54:07:454 wait reconnect 1 seconds
12:54:07:650 COM: /dev/ttyACM0 / serialno: DE2129434
12:54:07:650 COM: --dev: /dev/ttyACM0 (ConBee II)
12:54:08:229 device state timeout ignored in state 2
12:54:09:279 device state timeout ignored in state 2
12:54:10:331 device state timeout ignored in state 2
12:54:10:454 try to reconnect to network try=8
12:54:11:954 device state timeout ignored in state 2
12:54:12:954 device state timeout ignored in state 2
12:54:13:954 device state timeout (handled)
12:54:13:994 device disconnected reason: 1, index: 0
12:54:14:454 wait reconnect 15 seconds
12:54:14:454 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
12:54:14:454 device state timeout ignored in state 1
12:54:15:453 failed to reconnect to network try=9
12:54:15:453 wait reconnect 14 seconds`

## Additional context

Puntoboy, 2020-07-14:
Seems like the issue I am having now is the same as this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2702

Puntoboy, 2020-07-14:
The firmware update wouldn't work on Hassio, so I created a Ubuntu VM on my PC and flashed it there. Downgrading the firmware doesn't resolve the issue but does complete successfully. Here are the logs.

`chris@chris-VirtualBox:/usr/share/applications$ deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1
libpng warning: iCCP: known incorrect sRGB profile
14:01:34:989 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
14:01:34:991 CTRL. 3.22.014:01:35:013 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
14:01:35:013 ZCLDB init file /home/chris/.local/share/dresden-elektronik/deCONZ/zcldb.txt
14:01:35:052 parent process bash
14:01:35:052 gw run mode: normal
14:01:35:052 GW sd-card image version file does not exist: /home/chris/.local/share/dresden-elektronik/deCONZ/gw-version
14:01:35:052 DB sqlite version 3.22.0
14:01:35:052 DB PRAGMA page_count: 30
14:01:35:052 DB PRAGMA page_size: 4096
14:01:35:052 DB PRAGMA freelist_count: 0
14:01:35:052 DB file size 122880 bytes, free pages 0
14:01:35:052 DB PRAGMA user_version: 6
14:01:35:052 DB cleanup
14:01:35:052 DB create temporary views
14:01:35:052 DB view [0] created
14:01:35:052 DB view [1] created
14:01:35:052 DB view [2] created
14:01:35:052 DB view [3] created
14:01:35:052 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
14:01:35:052 sql exec SELECT key FROM config2
14:01:35:052 sql exec SELECT key,value FROM config2
14:01:35:052 Load config UTC: 2020-07-14T11:07:46 from db.
14:01:35:052 Load config announceinterval: 10 from db.
14:01:35:053 Load config announceurl: http://dresden-light.appspot.com/discover from db.
14:01:35:053 Load config apiversion: 2.05.77 from db.
14:01:35:053 Load config bridgeid: 00212EFFFF04F8BA from db.
14:01:35:053 Load config datastoreversion: 60 from db.
14:01:35:053 Load config dhcp: true from db.
14:01:35:053 Load config discovery: true from db.
14:01:35:053 Load config factorynew: false from db.
14:01:35:053 Load config fwneedupdate: false from db.
14:01:35:053 Load config fwupdatestate: idle from db.
14:01:35:053 Load config fwversion: 0x264a0700 from db.
14:01:35:053 Load config gateway: 127.0.0.1 from db.
14:01:35:053 Load config group0: 65520 from db.
14:01:35:053 Load config groupdelay: 50 from db.
14:01:35:053 Load config gwpassword: $1$8282jdkm$moajcS3VXwDsvHFXOaAiQ0 from db.
14:01:35:053 Load config gwusername: delight from db.
14:01:35:053 Load config homebridge: not-managed from db.
14:01:35:053 Load config homebridge-pin:  from db.
14:01:35:053 Load config homebridgeupdate: false from db.
14:01:35:053 Load config homebridgeupdateversion:  from db.
14:01:35:053 Load config homebridgeversion:  from db.
14:01:35:053 Load config ipaddress: 10.0.2.15 from db.
14:01:35:053 Load config linkbutton: false from db.
14:01:35:053 Load config localtime: 2020-07-14T12:07:46 from db.
14:01:35:053 Load config mac: 38:60:77:7c:53:18 from db.
14:01:35:053 Load config modelid: deCONZ from db.
14:01:35:053 Load config name: Phoscon-GW from db.
14:01:35:053 Load config netmask: 255.0.0.0 from db.
14:01:35:053 Load config networkopenduration: 60 from db.
14:01:35:053 Load config otauactive: false from db.
14:01:35:053 Load config otaustate: off from db.
14:01:35:053 Load config panid: 0 from db.
14:01:35:053 Load config permitjoin: 0 from db.
14:01:35:053 Load config permitjoinfull: 0 from db.
14:01:35:053 Load config port: 8080 from db.
14:01:35:053 Load config portalservices: false from db.
14:01:35:053 Load config proxyaddress: none from db.
14:01:35:053 Load config proxyport: 0 from db.
14:01:35:053 Load config rfconnect: 1 from db.
14:01:35:053 Load config rfconnected: false from db.
14:01:35:053 Load config rgbwdisplay: 1 from db.
14:01:35:053 Load config runmode: normal from db.
14:01:35:053 Load config starterkitid:  from db.
14:01:35:053 Load config swcommit: 2a26b78f4139a179968e61aa0e0caf3fd1917162 from db.
[... 226 lines cut ...]
14:01:59:796 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
14:01:59:796 GW firmware update select USB device
14:02:00:539 failed to reconnect to network try=1
14:02:00:788 wait reconnect 5 seconds
14:02:01:789 wait reconnect 4 seconds
14:02:02:788 wait reconnect 3 seconds
14:02:03:789 wait reconnect 2 seconds
14:02:04:791 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:02:04:791 wait reconnect 1 seconds
14:02:04:798 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
14:02:04:798 auto connect com /dev/ttyACM0
14:02:04:825 Serial com connected
14:02:05:302 device state timeout ignored in state 2
14:02:05:789 try to reconnect to network try=2
14:02:07:288 device state timeout ignored in state 2
14:02:08:289 device state timeout ignored in state 2
14:02:09:296 device state timeout ignored in state 2
14:02:09:813 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
14:02:10:346 device state timeout ignored in state 2
14:02:10:793 try to reconnect to network try=3
14:02:12:290 device state timeout (handled)
14:02:12:290 MASTER kill cmd 0x08 (ERROR)
14:02:12:290 MASTER kill cmd 0x08 (ERROR)
14:02:12:343 Serial com disconnected, reason: 1
14:02:12:343 device disconnected reason: 1, index: 0`

SwoopX, 2020-07-14:
Have you used version .78? I see .77 in the log...

Puntoboy, 2020-07-14:
> 
> 
> Have you used version .78? I see .77 in the log...

I downloaded it this morning and that was the latest... There's no update for the hassio addon.

Mimiix, 2020-07-14:
@Puntoboy What version is your HA addon :)?

Puntoboy, 2020-07-14:
> 
> 
> @Puntoboy What version is your HA addon :)?

5.3.6. It was 5.3.5 (I think) but as part of my troubleshooting for the IKEA blind I upgraded. Which has only made things worse.

Mimiix, 2020-07-14:
Okay.

Can you please provide me:
- The output of : https://www.home-assistant.io/integrations/deconz/#debugging-integration that logging
- A screenshot of your configuration page of the HA addon.

Puntoboy, 2020-07-14:
> 
> 
> Okay.
> 
> Can you please provide me:
> 
>     * The output of : https://www.home-assistant.io/integrations/deconz/#debugging-integration that logging
> 
>     * A screenshot of your configuration page of the HA addon.

At the moment I have uninstalled the deconz addon for Hassio as part of the troubleshooting I have been trying. 

I have Deconz installed on a Windows PC and a Ubuntu VM, I used the Ubunutu VM as the firmware upgrade wouldn't complete on hassio. Neither the UB VM or Windows PC can connect to the ConBeeII since upgrading the firmware.

Mimiix, 2020-07-14:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Did you follow that guide and read trough it properly?

Puntoboy, 2020-07-14:
Yes, the update does complete successfully, so it says, but the Deconz application will no longer connect.

Mimiix, 2020-07-14:
The page says:
**Warning: It is strongly advised to use a native installation of one of the update methods. Virtual Machines _may_ work but are not supported, using a VM while trying to update firmware is on your own risk.**

I suggest you to use the deCONZ windows version to update.

Puntoboy, 2020-07-14:
OK so I have the Hassio addon installed, I can VNC to Deconz and I can see the devices. Great :)

[image]

However, when I got to the Phoscon webapp I'm missing all the lights (understandable) and when I go to Gateway it's blank... 

[image]

Smanar, 2020-07-14:
Have you tried with another browser ? or with deleting cache ?

On my side I have this bug when I use the wrong ip (the deconz on windows without conbee, instead of using the Pi ip)

Puntoboy, 2020-07-14:
> 
> 
> Have you tried with another browser ? or with deleting cache ?
> 
> On my side I have this bug when I use the wrong ip (the deconz on windows without conbee, instead of using the Pi ip)

I was just trying that when you replied :) Seems HA cached the page. I'm back in now so I can restore my backup.

ebaauw, 2020-07-14:
> I can VNC to Deconz and I can see the devices. Great :)

Looks like the REST API resources haven't been created (the nodes still show the NWK address instead of the resource name).  Also note the descriptors for the two end devices in the middle haven't been read.  Looks like you lost the deCONZ database.  Best restore that from backup.  Alternatively, you'll need to re-pair each device.  Best search for devices in Phoscon and read the _Basic_ cluster attributes in the GUI.  When the name of the node changes, the resource has been created and Phoson should show the device.  Make sure to wake the end devices when reading the _Basic_ cluster and read the _Simple Descriptor(s)_ from the left drop-down menu to see the clusters.

Puntoboy, 2020-07-14:
Thanks. I only understood some of that, but restoring the backup is exactly what I was just doing. I can see all the names in GUI now and I can see switches and sensors in Phoscon, but not lights... it's empty.

[image]
[image]

Puntoboy, 2020-07-14:
[image]
[image]

The greyed out devices are expected, they died a while ago and I haven't got round to fixing them.

Smanar, 2020-07-14:
> Best search for devices in Phoscon and read the Basic cluster attributes in the GUI. When the name of the node changes, the resource has been created and Phoson should show the device. Make sure to [...]

I think it can solve your "inivisible" light in phoscon.

For grayed Xiaomi device, I think you can re-include them ^^, for them I think it s dead.

Puntoboy, 2020-07-14:
The lights came back, I didn't change anything :)

[image]

Smanar, 2020-07-14:
joke application ^^

Puntoboy, 2020-07-14:
OK so things aren't stable. When I try to re-pair a switch (that's greyed out) Deconz crashes. The VNC server closes and I can't control any devices. 

Log file from docker.

[Copy of addon_core_deconz.xlsx](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4920199/Copy.of.addon_core_deconz.xlsx)

Mimiix, 2020-07-14:
Could you provide this in a txt file? This reads really bad.

I've asked you before: Can you show me a screenshot of your configuration page of the addon?

Mimiix, 2020-07-14:
Edited my last comment a bit.

Puntoboy, 2020-07-14:
> 
> 
> Could you provide this in a txt file? This reads really bad.
> 
> I've asked you before: Can you show me a screenshot of your configuration page of the addon?
> 
> You need to place the full serial string there, not just the dev/tty*

I could, but I'll have to convert the file. That's how it was outputted from Docker.

The documents say I can use dev/tty* 

[image]

Here's the config page.

[image]

Puntoboy, 2020-07-14:
Bare in mine my system was working perfectly fine, until the blind stopped working a couple of days ago. 

I used to use the full string but it would display my device as a RaspBee not ConBeeII.

Puntoboy, 2020-07-14:
OK changing it to the full device string seems to have fixed the crashing. Weird.

Mimiix, 2020-07-14:
Neither of this is a issue caused by the deCONZ application. We've got multiple reports and seen that the full string worked. This is something that needs to be fixed in the addon, which is not maintained by us.

>I could, but I'll have to convert the file. That's how it was outputted from Docker.

Docker outputting in a xlsx? 

>I used to use the full string but it would display my device as a RaspBee not ConBeeII.

This also is a HA Addon issue.

Puntoboy, 2020-07-14:
That's fine, It was the addon people that sent me here without fully understanding my issue. 

Everything seems to working now. Even the blind has started working, even though nothing had changed to stop it from working in the first place.

Mimiix, 2020-07-14:
@Puntoboy Where did they send you here?
Can you provide me a url/name?

Puntoboy, 2020-07-14:
I posted in their github first and they closed it and pointed me here. 

Tbh I don't know what the root cause was. In troubleshooting the blind not working I made things worse so additional steps where needed to recover. It's all working now :) thank you.

Mimiix, 2020-07-14:
Can you provide me the URL of that issue :)?

Puntoboy, 2020-07-14:
I opened two :) 

https://github.com/home-assistant/hassio-addons/issues/1460#event-3543256625

https://github.com/home-assistant/hassio-addons/issues/1461#issuecomment-658134820

Mimiix, 2020-07-14:
Found it already:)

Closed #3044

closed 2020-07-14 by Mimiix

Puntoboy, 2020-07-14:
> 
> Docker outputting in a xlsx?

Yes.

[image]

Mimiix, 2020-07-14:
This is a Synology  VM... 

You couldve just renamed to .txt.

Puntoboy, 2020-07-14:
> 
> 
> This is a Synology VM...
> 
> You couldve just renamed to .txt.

It's a Docker VM on Synology.

I could have, that's what I said about converting it. It wasn't needed anyway.