dresden-elektronik/deconz-rest-plugin #3045: Sengled Element Tunable White LED Bulb cannot change temperature

jon6fingrs, 2020-07-14:
Hi, I just installed a Conbee II and got it working through home assistant. So far I paired a samsung motion sensor which works fine. I just tried to pair a Sengled Element Tunable Bulb which was previously paired with SmartThings Hub and working fine to change temperature and brightness. I added it to the Conbee II without an issue. It was then added to home assistant. I can turn it on and off, and change the brightness from home assistant, but there is no option to change the temperature. If I try to manually specify the temperature, I get an error that the bulb does not support that command. Does Sengled use a nonstandard way of communicating? Is there anything I can try to change the temperature or should I go back to smartthings? Thanks!

Mimiix, 2020-07-14:
Hi! 
Related to: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3018

Should be fixed it tommorow's beta!

jon6fingrs, 2020-07-15:
Hah wow perfect timing! Thanks for the heads up!

Mimiix, 2020-07-15:
Your welcome!

I'll close this issue for now. If you have updated to the newest version and still encounter a problem, please open a [Bug report](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Bug+report&template=bug_report.md&title=). 

Thanks!

closed 2020-07-15 by Mimiix