dresden-elektronik/deconz-rest-plugin #3046: Xiaomi Smart Plug (Xiaomi Mija ZB3.0 ZNCZ04LM) doesn't show connection lines / mesh in deconz

fennsen64, 2020-07-15:
## Describe the bug
These plugs doesn't show up the existing connections / mesh to the end devices.
Only the connection to the coordinator is visible (see screenshots).

The connections from the (de-facto connected) devices to these plugs (here the Xiaomi Sensors: Multisensor-6 and Contactsensor-1 ) are also missing. With an other plug(Osram smart +), as well as connected directly to the conbee II stick there were shown up correctly

-I also tried the 2.05.78 beta: same behavior (and reverted because this versions crashes my raspbee gateway on pimatic)
-Tested two of them, same behavior

## Steps to reproduce the behavior
## Screenshots
[image]
[image]

## Environment

 - Host system: (Intel NUC, debian buster)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26580700)
 - deCONZ version: (2.05.77)
 - Device: (ConBee II)
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

SwoopX, 2020-07-15:
To be honest, I don't know what you want to say here. Do your sensors work?

Smanar, 2020-07-15:
The mesh is strange for me, why only 1 connection by router ?
They are so far away one from the other ?

1 of your 2 sensor have leave, and I have doubt for the second

fennsen64, 2020-07-16:
Thats exactly the point and my issue; as I said: ,,The Plugs doesn't show up the existing connections / mesh to the end devices"
Yes: sensors are working. But I like to see to which router they are connected.

ebaauw, 2020-07-16:
Zigbee doesn’t maintain connections.  The lines show neighbour table entries, not connections.  Not all routers support querying their neighbour table.

Smanar, 2020-07-16:
I have searched on other issue, I haven't found similar network (with the same device). But yes if all is working (even the grayed one), can be normal. I have never see this bug on my side, so it' s realy strange for me.

If someone else with a ZNCZ04LM can take a look on his own network .....

SwoopX, 2020-07-16:
Alright, tried it in my test network. Took pretty much the maximum diagonal distance at my place for the coordinator and a wireless switch, the plug in between.

[image]

The plug apparently routes appropriately. The lines didn't show up as well, but to be honest, I couldn't care less.

Maybe I can give it another try tomorrow with a more talkative end device.

Smanar, 2020-07-17:
Yep, but it s just by curiosity. honestly, if I had the same network I would have asked myself questions too.

Now we know what answer at next one with same problem ^^

closed 2020-08-15 by stale[bot]