dresden-elektronik/deconz-rest-plugin #3048: Issue with light reporting "lift" attribute

wvuyk, 2020-07-16:
Hello,

I have a user where a light with modelid "lumi.light.aqcn02" is reporting a "lift" attribute. This is not a curtain or blind, it is just a light. I think this attribute should be surpressed for real lights? Or how should one recognise a real blind device in this situation?

```
"type": "Color temperature light",
"name": "Studeerkamer",
"modelid": "lumi.light.aqcn02",
"swversion": "11-22-2018",
"uniqueid": "00:15:8d:00:04:5e:bf:8c-01",
"manufacturername": "LUMI",
"state": {
"on": false,
"bri": 167,
"ct": 359,
"alert": "none",
"colormode": "ct",
"reachable": true,
**"lift": 0,**
"open": false
},
"etag": "ea6e6f8d3f1bdbf0156025a877961c5a",
"hascolor": true,
"ctmax": 65279,
"ctmin": 0
}

```

SwoopX, 2020-07-16:
Can you please provide the screenshots mentioned in a request for new devices? This feels like another quirks for the Xiaomi's... Thanks!

wvuyk, 2020-07-17:
I do not have these lights myself, one of the users of my plugin is having these lights. They were working fine except that they are now reporting the lift attribute. The screenshots have been on the board before check issue #1083 Maybe this helps?
If not I can ask the user for new screenshots.

Thanks,

Wim

ebaauw, 2020-07-17:
According to the screenshot in that issue, the light advertises a _Windows Covering_ cluster.

SwoopX, 2020-07-17:
That was also my thought, because that's the only possible way. We could blacklist those devices for the next version.

wvuyk, 2020-07-17:
That would be helpful. Was hoping to have a common way of recognizing the blinds, but apparently there will always be exceptions. I will start here in my plugin to exclude these models for now.

Thanks

Mimiix, 2020-08-10:
@wvuyk I think we can close this then. If not, reach out to me ;)

closed 2020-08-10 by Mimiix