dresden-elektronik/deconz-rest-plugin #3051: Information for support of product byed from dresden elektronik ingenieurtechnik gmbh Enno-Heidebroek-Straße 12 01237 Dresden | Germany

MattWestb, 2020-07-17:
Most users have buying their products on amazon all over the world and have 2 years guarantee and support.
The only offishall contact i have finding is mail: support@dresden-elektronik.de
Then buying one product from the company I have one contract with them and ther CO its the contact person but i don't allowed writing his name here eve that is written as contact person on amazon.de and the same for the COO.

How do other users making then they having problem with the product they have byed and using it with the company's list of supported 3 parts products ?

This community issue tracker helping manny users but then its coming to real hard problems its very fue here that have the knowledge and possibility to helping.

I'm interested of other users experiences of how they getting help with ther technical problems with there products.

Mimiix, 2020-07-17:
**On a moderation note:** Thank you for opening your own issue :) I'm going to follow this with caution, to make sure the discussion stays civil.

**On my Personal note:** I think you are right that the harder questions should be answered by the company. At this moment, i do know that the support is overwhelmed with emails. Not just hard stuff, but also information or how to information: information that is already here but people CBA to search for. Because of that, the support is to busy to answer all. That's why we have git and discord: For the easy and medium questions about how to use the software. I don't see any need for messaging a CO or COO, as they probably don't know this either. 

However, this is a very hard product to maintain. It has so many components and 3rd party devices are all different due to differences in the zigbee spec. This product still manages to have a powerfull GUI and a Consumer experience for a relative low price. It feels like a Hue Bridge, but then one that can work for all kinds of devices.  They are doing a good job and the people here are all trying their best to make that happen. 

Yes there is flaws, but the product itself is in no way comparable.

morfei1, 2020-07-18:
I get help with a lot of reading and asking. Most if not all problems are well documented as issues in git. Most are sloved , some are not, because of some particular limitation of a specific device or it's FW and the way it's implement by it's developers/manufacturer.

If one searchs and like reading, learning and is not afraid to ask here or in dicord, he can avoid many problems using. The devs are very responsive and do their best to solved the problems.

To be universal, is not an easy job.

MattWestb, 2020-07-18:
I agree with @morfei1 for the community part it's helping many users that have problem with ther systems.
With my observings i see that for the last months the problem is that the real support form the company its nearly zero.
Back to the community support. I naming some users by nickname also then the moderator don't like it but I think it's needed taking the risk.
The largest source with deep knowledge its delivered by @manup but hi don't have so much time to spare and sadly also pretty hard restricted by time and knowledge from the company.
The second best without connecting to the company its @ebaauw that spending much time and deep penetrating problems with deCONZ and the HUE parts. Hi have in the last half year redone the OTA-plugin for DE and adding support for OTA for more device manufacturers and DE have not paying any cent or official credits him. ebaauw it's also very pedagogik and take his time explaining for the user way ,how and what to do.
Then it's some more (around 5 I have observing) with deeper knowledge and more or less connecting to DE they is also helping but they not at the same skill level as number 1 and 2 but less it's more the nothing.
And then latest large community contributor that have trying putting all the issues around the connection problem in the last release and betas together and trying finding a way for more or less desperate users getting their systems up and running again, its our stale king @Mimiix.
I'm interesting of more observations and experience from our users here for getting one overview how its working in real.

morfei1, 2020-07-18:
May I ask you something as a user... What's the point of this "issue"? Do you have any real problem with you deCONZ setup? I really can't get it. If you don't like DE in someway, you can become part of them and change that or you are free to chose another platform. There are alternatives.

MattWestb, 2020-07-18:
1. Yes pleas.
2. An interesting of wath other user have have experience of the support DE is giving ther customer.
3. Yest i have some old good bugs in the baggage that coming and going.
4 I like the product but but the service not from the community, from the company its for the moment in my eyes to low.
5. "you can become part of them" I dont understand.
6. I have more systems but for the moment its deCONZ going lower and my other 2 growing. and in the end some its doing out 100%.
7. I like the concept also the persons that from DE that helping solving the problems.
8. I knowing Z2M it's only getting better and better !!
9. Its taking my very hard see how users its left with 100% broke systems and not getting help for mounts. and if you reading the marketing its on premium product but for the moment its only the price you its paying.

MattWestb, 2020-07-18:
Your multi edit questions its not so easy 2 answer  !!!

morfei1, 2020-07-18:
Take it that way. DE is the company that makes great  hardware full of engineers and so on. The peaple you mentioned + a few more also very clever devs are the ones that make this great  hardware to come alive and become usefull for the end user as much as they can and make it getting constantly better and better.

Well my personal opinion is that deCONZ is more for the entusiasts user than to the regular ones right now, but I do believe that will change in the future.

Don't compare their support with another company for example. The approach is different and you need to accept it.

MattWestb, 2020-07-18:
I'm not allowed naming persons here because of your colleague and i only knowing one from DE that  helping with deep problem but hi have not so much time 2 spend her.
If you selling one product for around 40€ as one premium product with 2 years guarantee and included software.
1 hardware costs around 1-2€ to manufacturing and 5-10€ administration and seling.
DE have sold not some hundreds of the bricks and after written of the development its much money mande.
But if you don't handling the "aftermarket" = support with one premium product it's only time before you have killing yourself.

I can saying without problem that the product its mot mature then its have bugs that its long living and making very serious thing.  Paring not working after some paring and some runtime. Corrupting the database so the user must deleting it and starting from zero.
Its like my 25 years of experience with danich hard and software and that was much larger communicating systems then deCONZ are / going to being.

morfei1, 2020-07-18:
So, it's all about DE not including an extension cable when selling to you the conbee? Are you serious?

Just 2 examples from my own experience rpi and Odroid.

Wish you great time with your deCONZ setup. I'm sure we can find solution for most of your problems.

MattWestb, 2020-07-19:
Thank you very much for first your creative technical suggestion i have seen in here.
If DE have delivered one USB extension cable with ther CornBees both first and second gen and recommending it in the quickstart guide its being very much less problem for the users here.
Its something @Mimiix can feed back to DE as product enchantment and it's only cost the company some cents / produkt and mutch user dont getting the interference problem.
But i'm very sorry for you that cant helping my problem.
If you have reading little more you have understanding that i dot having any CornBee so its not working for my.
For the second i have enough education and experience with electronics from LF to very high band radar (also military and marine ones) and have the capability finding and eliminating  this kind of interference then needed.
But you looks having very deep technical knowledge so i like that you fixing the 2 problem i have pointed out before (paring and DB corruption).

If you like hearing some serious suggestion you can stopping being childish.

This github its one official part of DE and its made for helping users that have buying ther products.
The company should also have some professional technical person that working in the company and on working time helping the users with the technical problem.
I can being @manup that spending 2 or 3 day a week in the github or some other from the company with enough knowledge.
If DE its not making any serious its not helping that the products are good engineered and the community its trying getting it working then its not possible without the interest of the company.
Then im being very sorry but the RaspBee / CornBee dead project.
Only looking around here,  how many users have the last 2 months one compleet un usebal system ?
If i bying one premium / professional product i don't like having my house without working lights for more then 2 months and that its the facts.
This is not only damages the company but also complete community !
And good luck with your rpi and Odroid, if making some more serious you need something better as redundant storage (RAID 10) and so on (still talkin form some years professional experience of large professional communication systems).

ebaauw, 2020-07-19:
> This github [...] its made for helping users that have buying ther products.

No.  GitHub is a platform for developers to collaborate and share code.  From the GitHub main page:
> Built for developers
GitHub is a development platform inspired by the way you work. From open source to business, you can host and review code, manage projects, and build software alongside 50 million developers.

For product support, please see https://phoscon.de/en/support.

MattWestb, 2020-07-19:
You is right as alway but DE its using its as both developer questions and more or less one ticket system for support questions (or its accepted as).
And the only usable contact its the old well known support@dresden-elektronik.de

Mimiix, 2020-07-20:
>Only looking around here, how many users have the last 2 months one compleet un usebal system ?

I think you are over exaggerating here. In general, i haven't seen anyone who had a completely useless system. The ones who came close to that, went down to some user-error not related to the deCONZ software. However, i think it's best for you to join our Discord and have the discussion there :).

**On a moderation note: I notice that you are getting frustrated. Please keep that out. 
Comments as `If you like hearing some serious suggestion you can stopping being childish.` **

Smanar, 2020-07-20:
If you want to know some users return, you can too just look at amazon comment, I know it s not something that represent the reality but the product is rather appreciated.

MattWestb, 2020-11-26:
@ChrisPrefect You dont need opening one new issue i have already making it but Dennis D  was not liking it so much.

LG MW

MattWestb, 2020-12-10:
Latest status of to day:
The last days i have not seeing any "helping hand" from DE that taking care of the new and running issues in thies gitt.

I have trying helping some users that have problems but earlier today i was seeing one new post in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765#issue-752455629 ther one user pritty friendly was saying some things about the support and was recomending declering CornBee II end of life so DE can closing all support after the garatee time have expired (Its one OK way to do but not the best for the company image).
I was liking puting one emmo on the posing but it was not working so I was reloading the page and the posting  was deleted. Its one little chance that the user have deleting it self but not very likely. I dont remember his nick but his picture and it is not one newbee.

So my conclusion is its better deleting all things that the company dont like and taking all thru facebook and other medias and in the end also by the reviews from Amazon that is only posting verified buyers and cant being posted (but the buyer is returning the product to Amazon) is for the moment worse than some large chinese sellers of TI CC-253X (that is not easy getting under that if not starting refinished old CC-253X devices that was trached from one factory).

I finding the the situation more and more interesting but its very sad for the community (and some employee at DE) if its the company not doing any radically the situation is getting more then critical in the end and that is very fast.

One more thing: I finding it very strange implanting new devices only in the API and not in the close source Phoscon "app" and leaving one big mess after then not so experienced users can handling there devices in the GUI (more or less all new LIDL and tuya devices the last months).

Then the company have investing so much time and money  in the project i finding it very unprofessionally done !!

Mimiix, 2020-12-10:
This issue is getting locked as nothing constructive happens. Feel free to continue in the mentioned issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765