dresden-elektronik/deconz-rest-plugin #3052: IKEA Motion Sensors won't connect

jannisberry, 2020-07-17:
Hi, I have a fleet of IKEA motion sensors in my House and we wanted to get the Deconz app running on a newer Pi4 from the pi3B. That didn't work and so we wanted to get back to pi3 BUT now everything works except the Motion Sensors. We got one working but not sure how and with the others, there is no way connecting them. If you want to connect you choose them and follow the instructions but the Phoscon app freezes for the time it wants to connect to the Sensor and unfreezes if you move the mouse or when it is aborted from the sensor. We tried installing on 2 Pi's every Version of deCONZ from 2.05.60 and an older firmware on these but no success. The really strange thing is that the deCONZ UI shows pretty much all the Battery Powered Devices but with no name (so only a 0x56V3 as example) but it depends on if they are shown or not. I have no clue why but I have a feeling its inside the Phoscon app. Maybe someone an Idea? Maybe the Sensors Store the Information of the Host inside itself and won't connect to it again? Im really lost :D

Thanks in advance

edit: The one working is not really working it is more like turning the motion on and off and playing around with lights, but that may be more of a problem of the API

MattWestb, 2020-07-17:
Wten resetting one Ikea device its erasing all network info so its normally not problem.
Delete one sensor in phoscon-app and then if You have GUI access to deCONZ look if its ther.
Very oft its not complete deleted and is ther without name and only MWK adress as name = in the network but not compleet pared. Then marking the "node" and menu edit and "delete selected node".
Back to phoscon and try paring its as sensor. Try both as "ikea" and "other".
I have in the last month repairing both my old and new type old motion sensors in deCONZ and they its tricky to get working. 
Try it and if not its harder but not impossible.

MattWestb, 2020-07-17:
Back to your first "move" to Rpi 4:
I don't have experience of the RPi 4 only 3b+ and zero but the normale problem with the Rpi 4 its that USB 3 ports interferens with the CornBee dongle. Then if having USB2 ports free us them for the dongle and also an long USB cable so its coming away from the interfering source.
If you making one backup and the move it sud working OK but not 100% chance.
If not using one backup but using the same network key and pan id then its a big mess.
The reason its that all devices its rejoining the network and the network key and pan id that itst stored in the device its OK but deCONZ do not repairing them and you getting them as goustes in deCONZ.
The best way its to resetting all the "ghost" device and then taking out the battery so its clean and quiet in the zigbee network.
Then deleting all the ghosts in phoscon and manual in deCONZ.
Then starting paring one at the time as i have describe a bow.
It is also possible paring / reparing with deCONZ but its much more work 2 getting it working OK and little danger that the paring ending with ghost devices.

jannisberry, 2020-07-18:
> 12:24:57:866 add task 467 type 19 to 0x00158D00012F590E cluster 0x0006 req.id 197
12:24:57:866 Poll APS request 197 to 0x00158D00012F590E cluster: 0x0006
12:24:58:006 Poll APS confirm 197 status: 0x00
12:24:58:006 Erase task req-id: 197, type: 19 zcl seqno: 104 send time 0, profileId: 0x0104, clusterId: 0x0006
12:24:58:044 Node data 0x00158d00012f590e profileId: 0x0104, clusterId: 0x0006
12:24:58:044 0x00158D00012F590E: update ZCL value 0x01/0x0006/0x0000 after 0 s
12:24:58:046 Websocket 192.168.1.34:57429 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-18T12:24:58Z","manufacturername":"Paulmann lamp","modelid":"CCT Light","name":"Terrasse rechts","swversion":"19091613","type":"Color temperature light","uniqueid":"00:15:8d:00:01:2f:59:0e-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"00:15:8d:00:01:2f:59:0e-01"} (ret = 331)
12:24:58:046 Websocket 192.168.1.55:59922 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-18T12:24:58Z","manufacturername":"Paulmann lamp","modelid":"CCT Light","name":"Terrasse rechts","swversion":"19091613","type":"Color temperature light","uniqueid":"00:15:8d:00:01:2f:59:0e-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"00:15:8d:00:01:2f:59:0e-01"} (ret = 331)
12:24:58:056 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x4CFD
12:24:58:056 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x4CFD
12:24:58:056 new node - ext: 0x000d6ffffe0e12ae, nwk: 0x4CFD
12:24:58:293 Node 0x7CB03EAA00A9A374 is known by 3 neighbors, last seen 7 s
12:24:58:773 Node 0xCCCCCCFFFE0212B5 is known by 6 neighbors, last seen 21 s
12:24:59:008 poll node 68:0a:e2:ff:fe:72:48:20-01
12:24:59:008 Poll light node Wasserzirkulation
12:24:59:060 Poll APS request to 0x680AE2FFFE724820 cluster: 0x0006 dropped, values are fresh enough
12:24:59:253 Node 0x00158D00012F590E is known by 4 neighbors, last seen 1 s
12:24:59:734 Node 0x680AE2FFFE724820 is known by 1 neighbors, last seen 33 s
12:25:00:031 poll node 7c:b0:3e:aa:00:a9:98:91-03
12:25:00:031 Poll light node Carport links
12:25:00:079 read attributes of 0x7CB03EAA00A99891 cluster: 0x0006: [ 12:25:00:079 0x0000 12:25:00:079 ]
12:25:00:079 add task 478 type 19 to 0x7CB03EAA00A99891 cluster 0x0006 req.id 218
12:25:00:079 Poll APS request 218 to 0x7CB03EAA00A99891 cluster: 0x0006
12:25:00:213 ZDP skip fetch, node 0x7CB03EAA00A99891 has unconfirmed requests [1]
12:25:00:213 Node 0x000D6FFFFE388FAD is known by 1 neighbors, last seen 0 s
12:25:00:239 Poll APS confirm 218 status: 0xE9
12:25:00:239 	 drop item attr/modelid
12:25:00:239 	 drop item attr/swversion
12:25:00:239 	 drop item state/bri
12:25:00:239 	 drop item state/colormode
12:25:00:239 0x7CB03EAA00A99891 error APSDE-DATA.confirm: 0xE9 on task
12:25:00:239 Erase task req-id: 218, type: 19 zcl seqno: 105 send time 0, profileId: 0x0104, clusterId: 0x0006
12:25:00:453 Mgmt_Lqi_req zdpSeq: 88 to 0x680AE2FFFE6EC61E start index 0
12:25:00:495 Beacon src: 0x0CA4 ch: 25 updateId: 3
12:25:00:594 ZDP status = 0x00 -> SUCCESS
12:25:00:594 ZDP Mgmt_Lqi_rsp zdpSeq: 88 from 0x680AE2FFFE6EC61E total: 7, startIndex: 0, listCount: 3
12:25:00:594     * neighbor: 0x00212EFFFF0400B9 (0x0000), LQI: 129, relation: 0x02 rxOnWHenIdle: 1
12:25:00:594     * neighbor: 0x00158D00013BC4A7 (0x3107), LQI: 53, relation: 0x02 rxOnWHenIdle: 1
12:25:00:594     * neighbor: 0xCCCCCCFFFE0212B5 (0x48F6), LQI: 58, relation: 0x02 rxOnWHenIdle: 1
12:25:00:598 Websocket 192.168.1.34:57429 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-18T12:25:00Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI Driver 30W","name":"Chillicht","swversion":"1.2.245","type":"Dimmable light","uniqueid":"68:0a:e2:ff:fe:6e:c6:1e-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"68:0a:e2:ff:fe:6e:c6:1e-01"} (ret = 325)
12:25:00:598 Websocket 192.168.1.55:59922 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-18T12:25:00Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI Driver 30W","name":"Chillicht","swversion":"1.2.245","type":"Dimmable light","uniqueid":"68:0a:e2:ff:fe:6e:c6:1e-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"68:0a:e2:ff:fe:6e:c6:1e-01"} (ret = 325)
12:25:00:694 Node 0x7CB03EAA00A99891 is known by 2 neighbors, last seen 8 s
12:25:00:825 poll node 7c:b0:3e:aa:00:b0:55:79-03
12:25:00:825 Poll light node Bachlauf
12:25:00:873 Poll APS request to 0x7CB03EAA00B05579 cluster: 0x0006 dropped, values are fresh enough
12:25:01:174 Node 0x7CB03EAA00B05579 is known by 2 neighbors, last seen 27 s
12:25:01:654 Node 0x000D6FFFFE0E12AE is known by 1 neighbors, last seen 4 s
12:25:01:742 poll node 68:0a:e2:ff:fe:6e:c6:1e-01
12:25:01:742 Poll light node Chillicht
12:25:01:794 Poll APS request to 0x680AE2FFFE6EC61E cluster: 0x0006 dropped, values are fresh enough
12:25:02:134 Node 0x680AE2FFFE6EC61E is known by 5 neighbors, last seen 1 s
12:25:02:613 Node 0x00158D000470C698 is known by 5 neighbors, last seen 33 s
12:25:02:746 poll node 00:15:8d:00:04:70:c6:98-01
12:25:02:746 Poll light node Glaslampe
12:25:02:798 Poll APS request to 0x00158D000470C698 cluster: 0x0006 dropped, values are fresh enough
12:25:03:093 Node 0x00158D00013BC4A7 is known by 7 neighbors, last seen 9 s
12:25:03:161 DB save zll database items 0x00000801
[... cut ...]
What i saw was that the sensors have reconnected them selfs even when deleting them and paired again but it didnt show in Phoscon app. They show as 0x1F62, 0x5D92 or 0x4CFD in deCONZ GUI.
@Mimiix 

@MattWestb 
They Connect again and i see them without the name and only the id they even reconnect themselfs. The Sensors itself ahow with their light that they connected but they dont show in Phoscon and keep their ID only. Is there maybe a problem with the link between the Rest API and the Phoscon app? but only for these sensors

MattWestb, 2020-07-18:
The problem its that the sensors have being partly pared (its in the zigbee network) but deCONZ have not completed the paring.
I can being that the database its corrupt and you must redo all the paring but its come later if not getting it fixed.
First trying enabling paring in Phoscon.
Then going in deCONZ UI and marking the node you trying to getting pared.
Klick on the dit at the right hand (its normally 2 but you only have on). Then marking "read simply description then having the sensor awake (moving in front of it). If success you is getting a real name for the sensor and also the second dot at the right side.
Try multiple time and also resetting the sensor so t in paring mode.
Don't forget re enabling the paring in Phoscon.

MattWestb, 2020-07-18:
By the way in your logg i see only communication with Ikea drives and bulbs and no communication with "Ikea Tradfri motion sensor".
Its not so strange but not nice. Then the sensors was in your zigbee net before so the is knowing the network key and pan id. But deCONZ have not all information from the sensors = have not pairing (or not successful paring) them,

SwoopX, 2020-07-18:
This issue will be moved to the deconz rest api repository, as it apparently has nothing to do with Phoscon.

smulle48, 2020-07-19:
I had the same problem with some sensors when I was moving from an old defect rpi3 to a new rpi3. Unfortunately I did not have a backup from the webapp, and it gave me a lot of problems when I tried to include the sensors.

The only solution for me, was to reset the conbee from within the webapp, and then start to include everything again. Hard lesson learned :-/

After reset, the sensors included right away with no problem at all.

MattWestb, 2020-07-19:
I agree with you samulle48 its the safest way but can being a pain in the . . .  if having many devices or its not possible accessing them for making resett.
Then starting new its good changing the network key and pan id so the not resetted devices can't rejoining the network by them self with out manual paring and being ghosts in the deCONZ GUI and making it difficult / not possible paring them.

I wos rebuilding my deCONZ net in the beginning of the year because to mush crap in the database.
And very importing thing: MAKE BACKUP !!
Power outage and heavy writing its killing you SD card very fast and then you need a recent backup !!

jannisberry, 2020-07-20:
How can i reset the the Network completely? I tried this all now and it's not pairing.

MattWestb, 2020-07-20:
I think the easiest way its like smulle48 woas proposing: 
Open Phoscon.
Menu > Settings > Gateway.
Upper left: Reset Gateway.

Before start re adding your device adjust channel, pan id and network key.
If all the same as before all devices its trying rejoining the network.
I recommend changing network key and pan id and having one channel that is used for lighting link.

jannisberry, 2020-08-18:
Hi, I've been busy lately and tried somethings but with no success. I've reset the Gateway and tried to change the Pan ID and network key as well as Channel and  stuff like @MattWestb  said said. The interesting thing was that the network found itself again and everything was visible again and easy to get back up. Just  a question, how much do I have to change the pan id? I changed the last Digit of the 2nd hex value from e to f, is this enough or do I have to find a complete new id and network key and if yes which rules do they follow?

MattWestb, 2020-08-19:
Its depends how deCONZ is doing the operation. In ZB3 is possible the coordinator accounting change of channel. network key, pan-in and extended pan-id. If deCONZ is making that the devices getting the change with the old credential and start using the them. I was thinking deCONZ was not doing it this way, and only change the coordinator and starting the network. Then you first have doing one reset all network parameters is being factory mande and the database erase and new made. If its working well in thise mode all devices can "rejoining" the network and deCONZ can rebuilding the database, because all parameters was the same as before the reset (factory default) wen the coordinator was forming the network. I don't knowing if its working well or not then i have not testing it this way.
If you making one new reset of the gateway it sud not being possible for the device doing one "rejoining" then your parameters is not the factory one and need doing one repairing for all devices.
I was having the network parameters changed then i was doing the last reset of the gateway and must doing real repairing of all devices and getting one new clean database. Changes is absolute = 1 digit change is one new parameter. I was channing 2 or 3 positions in the keys / IDs. Then changing also changing extended pan-id and pan-id. Don't change the TC-Link key !!

jannisberry, 2020-08-19:
How does the factory reset work? is it over the cli? or do I have to do it via a gui

MattWestb, 2020-08-19:
Making one backup first so you can restoring your system if it not going well.
The easy way:
In the Phoscon app gui is the easter. Menu. Gateway. upper left reset gateway.
The "hard" way in Rpi:
First stopping deCONZ.
In the directory /home/pi/.local/share/dresden-elektronik/deCONZ
Deleting / renaming zll.db. session.default and config.ini.
Starting deCONZ.
The long way:
One safe way is deleting deCONZ rebooting and installing deCONZ (between deleting the directory mented abow Rpi).

jannisberry, 2020-08-19:
I don't really know because I tried to reinstall the whole system with no success. What maybe exactly that, a reset of everything. Is there a way to rewrite the internal flash or eeprom of the Raspbee itself? Or maybe use a custom Mac adress for it?

jannisberry, 2020-08-25:
So, to be fair with the old RaspBee there was no way of getting it to work. We bought a new ConBee II and plugged it in and everything worked fine (Not even reinstalling deCONZ). I still don't have a clue what it was, but only replacing the device worked for me in this case. Maybe a firmware update bricked it or smth. Thanks for the help u provided.

closed 2020-08-25 by jannisberry