dresden-elektronik/deconz-rest-plugin #3054: Deconz - Conbee2 disconnections after few minutes

king7222, 2020-07-18:
Hello everyone, I request your help for a problem I can't solve despite my researches. Sorry if my english is not perfect (I'm french), and thanks per advance for your help, I hope there is a solution :

I have a Conbee II stick plugged in a QNAP TS 251+ on a VM, with a Jeedom solution. Everything worked perfectly until yesterday (for no reason than I could understand...)

None of my Zigbee equipements answer anymore, and after multiple research and reboots (Jeedom, VM, NAS, Conbee stick...), I got to understand than the Conbee II stick seems to be the problem : The USB connection on my VM is lost after only few minutes. If I reboot the VM, everything works fine for 5-10 minutes (I can add new equipements and get them working)

I made the updates so my Conbee II is version 2.05.77 and Firmware 26580700

- My VM Jeedom start on my QNAP NAS
- The Conbee II is correctly detected
- After few minutes (sometimes 5, sometimes 20), the Conbee II is lot, and is not shown anymore when i do a "lsusb"
- On my QNAP VirtualizationStation, I still the the Conbee stick attached to my VM.

On the VM, when I lose the connexion with Conbee 2, I get this error : 

```
root@jeedom:~# [  384.855841] xhci_hcd 0000:00:04.0: xHCI host controller not responding, assume dead
[  384.855997] xhci_hcd 0000:00:04.0: HC died; cleaning up
[  384.856381] cdc_acm 1-3:1.0: failed to set dtr/rts
```

After that the only ways to get it back is either  : 
- Reboot the VM

- Do this : 
```
echo -n "0000:00:14.0" | tee /sys/bus/pci/drivers/xhci_hcd/unbind
sleep 5
echo -n "0000:00:14.0" | tee /sys/bus/pci/drivers/xhci_hcd/bind
```
[image]

Whatever I choose, the problem get back some minutes later...

As you can imagine, it's extremely annoying. 
I tried a full restore of my jeedom from a backup made 2 days ago, when it worked, but nothing changed.

Do you have an idea please ?
-->

## Screenshots
I saw on similar issue than create a file in rules.d could resolve this (you can see the cat of the file on the screenshot, but even after a reload-rules and a restart, no amelioration :( )

[image]

## Environment

 - Host system: (NAS)
 - Running method: (Debian on Virtual Machine)
 - Firmware version: (26580700)
 - deCONZ version: (2.05.77)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no)

king7222, 2020-07-18:
For information, after reading an other similar issue on a forum, I just disabled the autosuspend parameter on the bus/devices used by the conbee II.
It's working since 30 minutes now, so it seems to do the trick. I hope really so...

Smanar, 2020-07-18:
Hello, there is a know bug about device usb deconnexion loop, but this bug don't appear suddently (generally on new device).

You just need to make a firmware update (you can use the same you already have) but on a real machine, windows is fine because you listen the usb deconnexion sound, but It will work too on Ubuntu.

You need to use the command line, and use the parameter "-t 60" 

cf https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

king7222, 2020-07-18:
Thanks for your help, unfortunately it doesn't seems to resolve my issue :( BUT it was not useless : 
I did the firmware update on a Windows computer, no problem at all (with same version, 26580700), but you're right, I have multiple unplug sounds even without touching anything. There always are two unplug sounds at 1s interval (I guess one for Dresden Device and second for Adomax Device).
On Debian the disconnection seemed random (1 to 20mn). On windows it appears to be very precise : every 5 minutes

So the issue seems to be reproductible on a Windows computer. 
However, on Windows computer, the "GCFFlasher -l" command keep showing the ConbeeII, even after unplug sounds.

[image]

I hope it doesn't mean it's a Conbee II hardware problem ? Do you have an idea ?

Smanar, 2020-07-18:
Unfortunately, on windows, mostly bugs are hardware, no USB power problem, no software problem, no USB configuration problem (and you have the same bug on many machine) ...

Are you sure the firmware is updated (take another version to be sure) because I know some users where the update have failed, without error message.

Something I don't understand the disconnection timing is in second normally not in minuts for this bug.
But in your situation I think there is few hope, and you can start to contact the support.

king7222, 2020-07-18:
I'm pretty sure of the update because when I go to the Phoscon App page during the minutes the ConbeeII is detected, it confirms than Firmware is 26580700 (before saying "not connected" few minutes later...)

I tried a Downgrade to firmware deCONZ_ConBeeII_0x264a0700.bin.GCF, just in case, but same symptoms :(
Do you know if there is kind of "factory reset" possible ? I saw the Reset Gateway on Phoscon App but it just seems to reset the configuration

king7222, 2020-07-18:
Maybe it can help : When the application deconZ is launched on my Windows Computer (with USB Conbee II plugged in), I don't get unplug sounds anymore, the stick seems to be working correctly... very strange,

Smanar, 2020-07-18:
ModemManager is something usefull on your machine ?

And the "reset" feature is the one you have used, you already have reset the config and the firmware.

king7222, 2020-07-18:
> ModemManager is something usefull on your machine ? (there is some issue with it, for exemple, you need to disable it for firmware update)

I honestly don't know, I created the udev rule for ModemManager because i saw manup recommended it in this similar issue : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2419
but maybe I don't need it (sorry, I'm not very competent on the subject).

I'll try disable it,  make a little reboot and will tell you the result.

> And the "reset" feature is the one you have used, you already have reset the config and the firmware.

Ok, thanks
What do you think about the fact the Conbee stick doesn't produce unplug sounds on Windows when Deconz app is launched ? I had a hope that it could mean the problem is not hardware...

Smanar, 2020-07-18:
ha yep, sure, if it works fully on windows, it mean the device is good, but you need more test, it s fast to say if the device is bad, but need more time to say it is good ^^.

But if you haven't problem during some hours on windows, yes, you can try on another machine. And if you can without "ModemManager " to try.

SnaKKo, 2020-07-18:
Same bug here, after upgrade now the Conbee II disconnected himself.
When i try to execute "lsusb" (for example every 5 seconds) sometime i see the Conbee, sometimes not.

SwoopX, 2020-07-26:
@king7222 Although a little late, we really appreciate you sharing your experience in this. I know from my own experience that the Qnap OS is "not an easy client" to deal with.